visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #151: Feature Request - Power Monitoring

sjeffrey101, 2021-11-26:
Hi Visual Approach,

Just want to say again how thankful and lucky we are that you have created this project.  Really has changed the way we use the layz spa.  

Any way one of the things that im also paying alot of attention to lately is the power consumption of things around the home.  This been one of the biggest uses of power we have.  Certainly the biggest no essential luxury item.  Therefore i really want to monitor its power usage.

Im wondering if it would be possible to code something that monitors power usage through one of those CT clamps.   

Similar to how ESPHome does https://esphome.io/components/sensor/ct_clamp.html

Unfortunately i am unable to put a monitoring plug in as my outside socket doesnt have enough room.  I was wondering about using a shelly EM from inside the egg, but this seems like an expensive route and another (esp) device connected to the network.  The logical answer is that the device already in the egg can do this, and it would have easy access to the live feed to put the clamp on.

I believe the CT clamp returns an analogue voltage, depending on current consumed. - it probably not that simple.

What do you think.   I know its quite a big ask, and i really appreciate what you have done, so really wont be offended if its just too much work.  Especially since i can achieve this, with another device.

Thanks

Stuart.

chunkysteveo, 2021-11-26:
The CT clamp can be connected to the A0 pin of the ESP8266 and monitored there. 

You'd need some custom firmware with the added code to capture that data and send it "wherever". Assuming you have Home Assistant as you've mentioned it. You could add in some power monitoring code to the base and allow the Tub to send that over to Home Assistant as a power value.

You could also add in a separate ESP8266 device like an ESPHOME device with the clamp and treat that as a separate monitoring node - the code needed could be all managed via ESPHOME then and it would be easier.

I installed the "Home Assistant Glow" project using ESPHome and a photodiode to monitor my Electricity supply and it's my new obsession watching my money drain away!!

Back to the device though, you could tap off a 5V and GND, or even use the spare 3.3V and GND pins on the PCB to power the extra ESP8266 - it will only pull about 500-750mA tops.

sjeffrey101, 2021-11-26:
Hi,

yep it is cheap enough to show enough ESP device in the egg, just to monitor the power through esphome.  To be honest i can do this easy enough.  I was just wondering though if it would be quite a descent extra feature the project can do, as an all in 1, since its already in there an sending data.  I think alot of people might be interested in this.

However i dont want to make extra work, for VA if its not something he isnt interested in.

Yes i have recently dont the Glow project, and that is working well.  Im now trying to figure out the power usage on Ovens, Fridge, Dishwash, Washing machine, tumble as well.  So i can attibute any high usage to devices.  Also want to try and lower my house idle usage of 700 watts.  I think the pump on the hot tub is drawing 40watts (when not heating) but it would be good to be sure.  What fun.

chunkysteveo, 2021-11-26:
The A0 pin would be ideal use for a power clamp, and a revised PCB would be cool. (Or a bidge wire!). It's not being used for anything else.

For your power appliances I'd highly recommend the Sonoff POWR2 with Tasmota (or ESPHome). They monitor the devices power draw and report back to HA very well. I have one on my garage heater and its worked perfectly with Tasmota.

visualapproach, 2021-11-26:
> Hi Visual Approach,
>
> Just want to say again how thankful and lucky we are that you have created this project. Really has changed the way we use the layz spa.
>
> Any way one of the things that im also [...]

Hi!
Thank you for your kind words.
I assume you refer to the 4-wire version.
There is already a power calculator in the program. It's just multiplied by the price/kWh and sent as "cost" over websockets. In the 6-wire version both cost and kWh is explicitly sent over ws and MQTT. The parameters (pump, heater, idle and jets power) could be tweaked for more accuracy if wanted. I can update the 4-wire version to the same behaviour if you want. It's easy enough to do this weekend. I will not program a new solution for every special case like with CT-clamps. At least not in a near future. But it was ofc okay to ask :).

Example from my HA Energy page:
[image]

visualapproach, 2021-11-27:
KWH now sent over WS and MQTT topic /times
(development branch)

tempod, 2022-01-22:
you can use this

https://github.com/bramstroker/homeassistant-powercalc

the water is heated by a resistance that always consumes the same kw.
with that integration you can create a sensor that counts consumption every time the heating is switched on.

closed 2022-03-09 by visualapproach