visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #154: Palm Springs 4 wire problem

quisquater, 2021-12-06:
Hello,

I'm getting problems to get the temps and controls to work.
The esp is reachable and the display is burning but always 00 celsius.
I used your PCB and 8 bidirectional level converter.
I checked the connection on my cio and disp, so i know whats 5V, RX, TX, Ground.
Soldered cio rx to D3, cio TX to D2 and Dis TX to D6 , RX to D7.
Does the temps automaticly send to the display even if the esp not wright coded?
What do i need to change to get it workable? (coding beginner ;))
Grtz

visualapproach, 2021-12-06:
Hi, first try the development branch. Make sure to edit "model.h". You may have to try different models.
Then if it still doesn't work, check the console in the browser. What does it say about tx and rx?
Google how to bring up the console on your browser.
Good luck

quisquater, 2021-12-08:
> Hi, first try the development branch. Make sure to edit "model.h". You may have to try different models. Then if it still doesn't work, check the console in the browser. What does it say about tx and rx? Google how to bring up the console on your browser. Good luck

Hey, i tried different models and files but i'm not getting any connection to my cio.
on the web console it says cio-tx: 0 and dsp-tx: 1.
i can start the bubbles on the display but temp still giving me 00 celsius.

visualapproach, 2021-12-10:
> > Hi, first try the development branch. Make sure to edit "model.h". You may have to try different models. Then if it still doesn't work, check the console in the browser. What does it say about tx and rx? Google how to bring up the console on your browser. Good luck
> 
> Hey, i tried different models and files but i'm not getting any connection to my cio. on the web console it says cio-tx: 0 and dsp-tx: 1. i can start the bubbles on the display but temp still giving me 00 celsius. Now i even can start my pump and heater trough web if i take control, but temps stays 00 on display and web.

so it seems cio-tx line is not working here. Check for bad soldering. Measure connectivity between LLC and cio-tx. Check connector and so on.

quisquater, 2021-12-13:
> so it seems cio-tx line is not working here. Check for bad soldering. Measure connectivity between LLC and cio-tx. Check connector and so on.

Ive checked everything and al the connections are good, measurement ohm and voltage over al the connections.
there is even coming 1.6-3,3 voltage on my D2 and D3 pin. but still no cio-TX to 1. Maybe the problem is coding related?

closed 2021-12-13 by quisquater

visualapproach, 2021-12-13:
Yesss

pdecat, 2022-01-23:
Hi @quisquater, may I ask what model did you end up using in your `model.h` with your Palm Springs?