visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #157: Bestway Paris AirJet Hot Tub

chris153002, 2021-12-22:
Does anyone have any experience with the Bestway Paris with pump model 54149E?
It does not seem to work with the standard code. 

the filter unit is a 6 wire circuit unit with the following pinout:
1. 5v  
2. GND 
3. Data from Keys? (when this cable is removed between the pump and display you can see the display change but can not control)
4. Clock 
5. Data to display (when this cable is removed between the pump and display you can in fact control the pump but the display does not update)
6. Buzzer

The display board has a TM1640 led driver and a WTC6508NSI (can not find datasheet for NSI version but found BSI version)
[image]

I was able to grab a logic capture of pressing the "UP" temp button
[image]

Any chance that anyone has already been looking into a pump with a similar configuration?

visualapproach, 2021-12-23:
Interesting! 
Today on Lay-Z spa factory, episode 8: "100 ways to connect two things"... 😆 

Is the graph you posted truncated? Seems odd that it ends out of alignment with the appearant 10 bit boundary, leaving pin 3 hanging low. (Haven't read the datasheets) What happens the next time the clock ticks? Is it re-aligning or just continuing? 
I *think* this method is easier to handle than the "original". Might even be possible to use hardware SPI. But it needs reprogramming of the core library of course.

chris153002, 2021-12-23:
Yea someone in the engineering dept. likes to learn new things haha.

Pin 3 is from the buttons, it requires a start bit, 8 data bits ( the 8 input button) and a stop bit.

I am thinking that given there is no CS pin to reset the packet sequence, that a timer would need to be used , maybe 800 microseconds or  so ( minimum amount of time I noticed between packets was about 1 millisecond, and that was between the button read and the display write). the timer could be reloaded every interrupt of the clock pulse? I'm not sure of a way to do this without a timer, any thoughts?

I also noticed that the pump unit seems to set pin4 (data out to display) low about 24 microseconds before the first clock pulse of the packet, where as for the button read the pump unit keeps pin 4 high the entire time. (the TM1640 data sheet shows that as well but doesn't give a time requirement that i am seeing).
[image]

The attached capture is to show the general packet spacing.
1. the button read happens every 9 to 10 milliseconds ( not sure it needs to happen that often, it is definitely responsive at 100ms )
2. The pump appears to write to the display on demand, also when I unplug the display unit and take it inside to play with it powers up with the last valid display data displayed.
[image]

I think the combined clock may make it difficult to use standard SPI interface?

visualapproach, 2021-12-23:
> Yea someone in the engineering dept. likes to learn new things haha.
> 
> Pin 3 is from the buttons, it requires a start bit, 8 data bits ( the 8 input button) and a stop bit.
> 
> I am thinking that given there is no CS pin to reset the packet sequence, that a timer would need to be used

Yes, that's my thought too.

> I also noticed that the pump unit seems to set pin4 (data out to display) low about 24 microseconds before the first clock pulse of the packet, where as for the button read the pump unit keeps pin 4 [...]

Not necessarily. SPI is two way (MOSI/MISO) and it can transmit both ways at the same time with one clock. So the display (or ESP) just prepares the button code and sends it whenever the start bit is coming.

After X-mas I might play with it a bit if you need help. If you have the button codes and the LED bit to segment decoded it would be helpful. Please note if you read the bits as MSB first or LSB first.

chris153002, 2021-12-24:
Mappings for Buttons (LSB first) 
bit0 =  up button 
bit1 = down button 
bit2 = pump button 
bit3 = heat button
bit4 = f/c button
bit5 = bubbles button
bit6 = timer button
bit7 = lock/unlock button

the display mappings are ( all LSB first):
packetBit 0-7 are command1 lsb first, always seems to be command 0b01000000 (Normal mode)

packetBit 9 is the end/start sequence of the display NOTE: this is signified by the data rise and fall all while the clock is high

packetBit 10-17 are the command2 lsb first, always seems to be 0b11000000 (Display address 00H)

packetBit 18-25 are data1 - digit 1

packetBit 26-33 are data2 - digit 2

packetBit 34-41 are data3 - digit 3

packetBit 42-49 are data4 - b0 = NA, b1 = bubble button, b2 = heat button red, b3 = heat button green, b4 = timer button, b5 = lock indicator, b6 = timer timing indicator, b7 = timer done indicator

packetBit 50-57 are data5 - b0 = degree C indicator, b1 = degree f indicator, b2 = pump button, b3 - b7 = nothing

packetBit 58 is the end/start sequence of the display NOTE: this is signified by the data rise and fall all while the clock is high

packetBit 59-66 is command3 lsb first, always seems to be 0b10001111 (Pulse width 14/16) for display brightness

packetBit 67 is the bus return to idle state

[image]

visualapproach, 2021-12-24:
Awesome! Great work! I'll be back with some questions. Have a nice holiday now if you're in that part of the world

Cheers

visualapproach, 2021-12-27:
@chris153002 I'm working on a solution now and will post a test version here for you to try when I'm done.
In your pictures above, DATA5 bit3 is set (1). You wrote "b3 - b7 = nothing". I suspect that bit is used for something, but we will find out when we test the new version.
I looked into the hardware SPI specs, and decided not to go that route. Docs states that slave mode is not supported, and the receive buffer is max 32 bits. Other sources indicate that it should be possible, but it's easier for me to continue bit banging.

visualapproach, 2021-12-27:
It compiles but if it works is another thing...
The code assumes my version of PCB. If not you have to change pinouts in the library, or use the same pins.
Excited to hear the results.
[testversion.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/7780426/testversion.zip)

chris153002, 2021-12-27:
The code do not seem to work outright. No display, could not seem to control the unit with the buttons. i did notice that the mqtt topic for the button is toggling between NOBTN and UNIT every other second. I will connect up the serial line to a terminal and connect up the logic analyzer later on this evening. 

DATA5 bit3, when setting only this but in a sketch, nothing seems to light up (maybe an indicator on other models)

i did manage to get a pass thru working with the ESP8266 in between, however when writing this code I cobbled a lot together trying different things so it is not pretty, but in case it helps...
 
[hotTubPassThru.txt](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/7780862/hotTubPassThru.txt)

visualapproach, 2021-12-27:
Okay, let's hope the analyzer shows what's wrong. I'll study your code to see if I can spot the culprit. But we approached the task so differently that it is hard to compare :-)
Bit order is prone to mistakes. I tried to be consistent but it's easy to make errors. And I didn't use a timeout because the messages should be distinguishable by data fall/rise while sclk is high. Trying to make my code somewhat compatible with earlier versions is also complicating things which may lead to errors. My goal is to not have a third version of the program to maintain, but to use #defines. But it depends on how different I need to make the code.

visualapproach, 2021-12-27:
Ok, I think I got the charcodes completely wrong. I'll change it shortly

visualapproach, 2021-12-27:
Hi again. Only changed the charcodes and cpu speed here:
[TEST VERSION2.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/7781101/TEST.VERSION2.zip)

chris153002, 2021-12-28:
Still not working, however,
The ESP8266 reading the buttons seems ok, I can also see the button I am pressing over MQTT (they match up).

ESP8266  Read from display buttons
[image]

the ESP8266 sending the button data to the pump however keeps toggling on and off  bit 6 and keeping bit 7 high, also the button data line (PIN3 to the pump controller) does not return to idle state of high.

[image]
[image]

The button bit toggling is causing the ESP8266 to send 10+ MQTT messages a second which toggle between "NOBTN" and "UNIT", I assume "UNIT" means it is seeing a press on the C/F Button? in which case should be bit4 not bit 6 (TIMER).

[image]

The ESP8266 Display Write Packet to the Display Unit only has a total of 64 Low to High clock transitions ( where the display requires 67)

I Think the packet is missing:
CMD1 is missing the last clock pulse
Missing a clock pulse after the last data byte following CMD2
CMD3 is missing the last clock pulse.

[image]

When I try to send a command from the web page, it seems like the ESP8266 is attempting to unlock the unit.

[image]

I However never see a display packet come back from the pump controller ( I think the pump controller might ignore button presses when the data line doesn't return high after the read). 

These are my findings thus far.

visualapproach, 2021-12-28:
It's a pleasure coworking with you because you give clear and detailed information. Now I have something to work with. Probably later today. Thank you

visualapproach, 2021-12-28:
Mostly fixed now. Just have to check one more thing, but I will take a walk first in the 1 dm of snow that came just after I swept my yard...

visualapproach, 2021-12-28:
Okay, I think there is a chance this will work. I sprinkled some more clockcycles on it and cheered up the dataline to a higher idling level and some more.
[testversion3.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/7785762/testversion3.zip)

chris153002, 2021-12-28:
Making progress, 
the display is functioning now (displays IP on boot). I did have to change line 16 in BWC_8266_globals.h
from
`const uint8_t CMD3 = DSP_DIM_BASE + DSP_DIM_ON + 7;`
to
`const uint8_t CMD3 = DSP_DIM_BASE + DSP_DIM_ON;`
in order to get the display to show anything.

The button topic on MQTT is now oscillating between "NOBTN" and "DOWN" now.

There still doesn't seem to be  control from the ESP to the pump controller nor valid decoding of display data on the ESP.

I am looking into the button oscillation now.

chris153002, 2021-12-28:
Ok I took the function `getButton()` and copied it verbatim from the library code, same timing and everything, to the pass through code I have and it works flawlessly.
1. it does not have random noise when no buttons are pressed ( the 2 in the string of characters above)
2.  It also holds the state of the button if i hold it down ( in the library it would occasionally drop the button being pressed and return a 0 for a cycle.

I am thinking something else in the library is playing with the button pins or the button variables.

visualapproach, 2021-12-28:
Thanks, I'll look at it tomorrow.

visualapproach, 2021-12-29:
> the display is functioning now (displays IP on boot). I did have to change line 16 in BWC_8266_globals.h
from
const uint8_t CMD3 = DSP_DIM_BASE + DSP_DIM_ON + 7;
to
const uint8_t CMD3 = DSP_DIM_BASE + DSP_DIM_ON;
in order to get the display to show anything.

Good catch! Switched to OR-ing the brightness to avoid havoc by adding it several times.

> The button topic on MQTT is now oscillating between "NOBTN" and "DOWN" now.

hm. Strange. The program tries to establish the target temperature at start-up, by faking an UP button press. But it only affects the cio side and never messes with the DSP::getButton() variables.

> I did try changing the code to sample on the rising edge (which is how i read the data sheet)

I would say it already is. Depending on interpretation maybe. As I understand it, the display changes the data line when the clock is LOW. So the state is uncertain right after clock switches to LOW. (For 20 us or so according to the datasheet). Then immidiately after clock goes HIGH the ESP samples the data line:
```
    for(int i = 0; i < 8; i++){
      digitalWrite(_CLK_PIN, LOW);
      delayMicroseconds(CLKPW);
      digitalWrite(_CLK_PIN, HIGH);
      newButton |= digitalRead(_TD_PIN)<<i;
      delayMicroseconds(CLKPW);
    }
```
In practice it shouldn't matter if you read before or after the clock goes high as long as you've waited >20 us after the falling edge.

> There still doesn't seem to be control from the ESP to the pump controller nor valid decoding of display data on the ESP from the pump controller.

That is dissappointing. Any logic analyzer dumps to share?

visualapproach, 2021-12-29:
Oops! I found an embarrasing error in clkhandler. It turned out that I flipped rising/falling edges. And there were some other errors that I think I've sorted now. I'm not sure it will affect the button reads, but I hope it magically did.
[testversion4.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/7788590/testversion4.zip)

chris153002, 2021-12-29:
It would appear that everything is functioning!
the only problem I am seeing now is the phantom button presses ( it turned the setpoint all the way down on me lol). 
I will do some poking to see if I can pinpoint it .

visualapproach, 2021-12-29:
Wow that's good to hear! I'll see if I Can think of something as well. No ideas for the moment though.

visualapproach, 2021-12-29:
Hi again. What if you switch order of theese rows:
```
      digitalWrite(_DSP_CLK_PIN, HIGH);
      newButton |= digitalRead(_DSP_TD_PIN)<<i;  //TRY MOVING THIS ROW UP ONE LINE
```
in this function:
```
uint16_t DSP::getButton(void) {
  if(millis() - _dspLastGetButton > 20){
	  uint16_t newButton = 0;
	  _dspLastGetButton = millis();
    //startbit
    digitalWrite(_DSP_CLK_PIN, LOW);
    delayMicroseconds(CLKPW);
    digitalWrite(_DSP_CLK_PIN, HIGH);
    delayMicroseconds(CLKPW);
    //clock in 8 data bits
    for(int i = 0; i < 8; i++){
      digitalWrite(_DSP_CLK_PIN, LOW);
      delayMicroseconds(CLKPW);
      *digitalWrite(_DSP_CLK_PIN, HIGH);*
      *newButton |= digitalRead(_DSP_TD_PIN)<<i;*
      delayMicroseconds(CLKPW);
    }
    //stop bit
    digitalWrite(_DSP_CLK_PIN, LOW);
    delayMicroseconds(CLKPW);
    digitalWrite(_DSP_CLK_PIN, HIGH);
	  _oldButton = newButton;
  }
  return (_oldButton);
}
```
I'm thinking that maybe the digitalWrite is upsetting the angry pixies just before sampling.

visualapproach, 2021-12-29:
In case we don't find a solution to the problem - here is a workaround. Single misreads will be ignored.

[testversion5.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/7789366/testversion5.zip)

chris153002, 2021-12-30:
I did try that that previously to no avail. Sorry for the delayed response, i am in the middle of a laptop reinstall.

visualapproach, 2021-12-30:
No worries. Did you try the workaround in testv5 or switching the rows? Both maybe, I just wanted to be sure. Good luck with your laptop.

chris153002, 2022-01-03:
Hello again, sorry for the delay.
Swapping the lines of code did not  make a difference.
```
Hi again. What if you switch order of theese rows:

      digitalWrite(_DSP_CLK_PIN, HIGH);
      newButton
```
The test version 5 did mast the button issues, however I am back to no pump communication with the ESP, not sure why as of yet. ( did not work before I flashed version 5, so the previously working code/setup stopped working)

chris153002, 2022-01-03:
I noticed this comment
`//require two consecutive messages to be equal before registering`

Which led me to the code
```
if (checksum != prev_checksum) {
	prev_checksum = checksum;
	return;
}
```
I commented it out given that the CIO only updates the display on demand, instead of routinely ( I.E. the CIO doesn't send duplicated display packets in a row.
The pump is completely responsive again, I can control it from the webpage, Iwill let it run and check periodically and report back.

chris153002, 2022-01-05:
Been working well so far.

visualapproach, 2022-01-14:
lovely

closed 2022-03-09 by visualapproach