visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #177: 4 Wire Paris 54154 Intermittent flickering of Pump, Bubbles, Heater

chunkysteveo, 2022-01-11:
Hi all,

Finally got round to adding this project to my Paris 54154 4 Wire model. Hardware and firmware all setup and it connects, functions, MQTT, Take control etc. I had to define model 54123 to get the unit working and talking. But what I am seeing is intermittent changes in state for the pump, bubbles, heater in the MQTT messages, and a noticeable "reset" of the pump when this occurs. And irregularities in how the heater, pump and bubbles function.

I am currently allowing the display to take control and just have it sat at 37°C. Watching the web GUI, I see the Pump switch turn off/on, randomly, as does the Bubbles. Watching the MQTT messaged, RED, FLT and sometimes AIR switch between 1/0 for a split second. If FLT does it - I sometimes here the pump stop/start itself.

If I take control and set the temperature on the ESP - the intermittent switching seems to go away. Suggesting that the ESP is not passing the right information between the display and unit when not taking control?

On testing the Bubbles - this turns off Pump and Heater, and switching off Bubbles, the Pump and Heater remain OFF. This is NOT what occurs when the ESP unit was present - I could have all three on (pump, heater, bubbles), and turning off the bubbles would keep the heater and pump on afterwards (albeit they don't turn off, so remain on).

So I have two thoughts on the two issues (intermittent on/off when not in control, and bubbles turning off heater and pump).

Would the intermittent issue be a hardware issue such as poor solder join between display and ESP, or more of a communication issue with incorrect setup? I am guessing that the Paris 54154 4 wire model is slightly different to the 54123 model, and thus do I need to redefine the bitmasks to allow for differing setups/config to match my model?

It's so close I can almost taste it, but keen to squash all my bugs before I commit to "bolting down the egg"!

Cheers,

Steve

visualapproach, 2022-01-11:
Hi @chunkysteveo 
Thank you for your detailed description. I'm traveling Atm but I will answer more in a few days. I think it needs  low pass filters to get rid of the glitches. Regarding the states of bubbles/heat on at the same time, it's just a matter of changing some arrays in the program. I will help you with that when I get home. Stay tuned :)

chunkysteveo, 2022-01-11:
That's cool, thanks! 

Guessing when you say low pass filter, you mean a hardware based filter and not a "de-bounce" style filter in the code? I may see if creating a better solder joint fixes anything - as it can be quite a "jolt" on the pump sometimes and I'm sure it can't be good for it in the long run?! The were soldered in a rush, and hopefully it may just be the display connector as there's no "jolts" when I take control (i.e. ESP to CIO seems to be solid??)

Just tested again (I have my de-scaler attached, so can heat, pump, blow etc without needing to be hooked up to the tub). When not controlling, the unit can run both Bubbles and Pump and Heater (Assume heater as the heat light does not go out when blowing bubbles). When I take control and hit Bubbles - the pump and heater switch off. If we can get a custom array of what's on/off, that would be amazing! I did see the setup @dietmar-1 had with LED lights hooked up to heater elements and pump etc - I could do that to  help further and confirm what I see, maybe?!

I'll keep an eye out for when you return, much appreciated!

visualapproach, 2022-01-12:
Hi again @chunkysteveo 
Back home again. I attached an excel sheet for you to fill in and return to me, and I'll add model 54154 to the code. I tried to explain how to fill it in, on the spread sheet itself. But just ask if you don't understand something. Or anything.
[jumptable layzspa 4-wire template.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/7856584/jumptable.layzspa.4-wire.template.zip)

And yes, I meant a hardware filter. Now, I'm not an EE so I can't really say this is the correct way, but I'd try something like the picture below. As for the values of R and C... don't know. 100 Ohms and 500 pF? Or just resistors of 1K and hope the wires has enough capacitance? I don't even know it's the real cause. You can also try overclocking the ESP by adding this line to platformio.ini file:
`board_build.f_cpu = 160000000L`
to see if that has any effect.
[image]

chunkysteveo, 2022-01-12:
Thanks for the table and the confirmation on the LP filter. I will check the table out ASAP and report back.

chunkysteveo, 2022-01-13:
Hmm, OK @visualapproach  - I have followed the Excel instructions. Only thing I wasn't sure about was if I worked out only the first row on the "second" Jumptable. If so - I have the following:

Button->	
Airbubbles	Hydrojets		Filter pump	Heater
8			4			1			3

Is that right?

chunkysteveo, 2022-01-13:
note: The 54154 model doesn't have any Hydrojets, so i've filled them out like the airbubbbles.

chunkysteveo, 2022-01-13:
Or is the Jumptable for all combinations, and if there's not a combination from the Allowedstates table - it's a 0?

visualapproach, 2022-01-13:
Some clarification, maybe:
You need to find out what state the pump will be in after you press a button, and find that state in the left table. Then enter that state's row number in the button cell.
Example:
Row 12 says: State is [bubbles ON, Jets OFF, Pump OFF, Heater OFF]
Consider what will happen if you press Pump button. You will probably end up with the state [bubbles ON, Jets OFF, Pump ON, Heater OFF]. Or, if your model can't run bubbles and pump at the same time, you will get [OFF, OFF, ON, OFF]. Find the row that describes the resulting state. Lets say its at row 15. Enter "15" in jumptable, row 12, column "Filter pump".
Repeat until every cell in the jumptable is filled in. Post the .xls file here.
I'd do it for you, but you're the only one that knows what states your tub can have. And more important - what states it cannot have. 

To make your life easier, I've removed all lines where hydrojets is on, and all lines where heat is on without filter pump. Leaving 7 lines. I filled them in with the assumption that only one of two heater elements will run when the airbubbles is on. You can have a look and see if this is expected behaviour. Or just try out the temp branch, where I will upload the code shortly. Don't forget to change model.h to your model

visualapproach, 2022-01-13:
[jumptable layzspa 4-wire 54154.xlsx](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/7865253/jumptable.layzspa.4-wire.54154.xlsx)

chunkysteveo, 2022-01-13:
Ok thanks. I will have a look at the unit in action. 

The only puzzle for this was that I can have states on and off, the bubbles for example can be on by themselves, or on with the pump on, or on with the heater (and pump) on. How does that work with the allowed states etc? Does that table work that all out?

Steve

chunkysteveo, 2022-01-13:
Ok. Thanks. Will have a tinker with the unit over the weekend... I'm cleaning the tub, insulating it and hopefully fixing a small air leak. Perfect time to open the unit up and test!

chunkysteveo, 2022-01-14:
Hey @visualapproach  - did the spreadsheet after working out what you meant, and after testing the unit with a multimeter.... the 54154 can run the heater and pump when the blower is on, and the heater at 50%. Full heater kicks in after several seconds when heating, so similar to the 54123.

Did the spreadsheet and actually came up with the same results you did, which are also in the V3 branch, which I am now testing. A quick test showed when taking control that I could have the heat on, and the bubbles, great!

Just testing a long run now, and i removed my header pins that the connectors were on and soldered the cables direct to the PCB - lets see if that solves the intermittent flickering of states?!

I'm keen to get a list together of what the 4-wire can and can't do compared to the 6-wire. The "7-seg" display on the webpage for example does not show anything unless the unit errors, but then it shows [E00], not [E02] for example. Assume we can't get the character arrays? Also, the MQTT value always shows 0 - see image for example:
[image]

Assume these are 4-wire issues? I think I may look for some 2nd hand eggs on eBay!!?

chunkysteveo, 2022-01-14:
(just noticed that the web gui has not updated to the new one for release 3? I get a magic packet error on upload OTA for file system. I may try USB...

visualapproach, 2022-01-14:
Yeah, the GUI is only updated for 6-wire for the moment.
OTA - sometimes you need to reset the device after uploading by USB.
You are right about the 7 segments - unless you pull the display apart there is no way of knowing what it says. "E00" is just a way to say "the error bit is set". And the MQTT 0 thing, is also because the GUI for the 4 wire is behind... We'll see when it gets updated :-). The zero means "connected" btw.
Good luck with the tests / Thomas

chunkysteveo, 2022-01-14:
Ahhhhhh - good to know!!

If it's a case of porting over the 6-wire HTML to the 4-wire and testing - happy to make a fork and PR for this?

Testing is OK-ish so far... On first time heating, when it gets to temperature, it seems to turn off the heater, and then the pump cuts out? Clicking the pump and heater again seems to clear that and it regulates, but it has done it twice on two attempts. I'll see if I can replicate that with regularity before confirming any issue.

visualapproach, 2022-01-14:
Sure, make the PR to the dev branch, thanks. @torei FYI

chunkysteveo, 2022-01-15:
On a side note the flickering statuses seem to have been resolved with a solid solder connection to the board 👍🏻

visualapproach, 2022-01-15:
Awesome

torei, 2022-01-17:
@chunkysteveo i haven't done any testing on that so far.
but if you have the time and the willing, you can give it a try:
https://github.com/torei/WiFi-remote-for-Bestway-Lay-Z-SPA/tree/feature/4wireUpdate

chunkysteveo, 2022-01-17:
Let me take a look!

chunkysteveo, 2022-01-18:
Hey @torei - looks a lot nicer, thanks! Had an error trying to upload over OTA the file system, so uploaded them one by one via the old GUI. It works nice for a new GUI. Couple of things:

- The Set temperature slider defaults to 0 on page refresh no matter if the heater is on or not, or if it's 20°C or set to a high temp.
- Take control (God Mode) toggle does nothing and changing this does not trigger any update to the data in websockets or MQTT.
- The Heater toggle button also default to off on page refresh no matter the heating state.
- Temp Units toggle retains it's state on page refresh, but the change of this does nothing to the GUI (change values). Value changes in Websockets, but not MQTT.
- Appreciate the Hydrojet toggle has gone from my model (which does not have it) but the label "Hydrojets" is still present.

It's looking good, but I think some linkage to and from the backend are missing. I can keep digging - but great work so far!

Going to have to put it back to the master branch for now as rebooting the ESP defaults God Mode to False, which means I can't toggle anything back on.

Steve

torei, 2022-01-18:
haloho @chunkysteveo - thanks for the reply
hmm .. hmm.. did you delete your web cache before testing?

checking the hydrojet toggle now, i think i did some "always disable" to the html because i was curious about the BWC replies without any "HASJETS". @visualapproach 

so your device does not have the JET functionality but the button was always visible?

chunkysteveo, 2022-01-18:
**so your device does not have the JET functionality but the button was always visible?**
It was present in the old version, you advised to hide it in the HTML. Assumed the new version and yours had a check for "hasjets" and hid it for me - but the label was still there - just the toggle switch was hidden. 51545 does not have jets.

**hmm .. hmm.. did you delete your web cache before testing?**
Didn't clear my cache, just a Ctrl+F5 and a reboot. I assumed that with no testing that this was a quick and dirty upload which wasn't going to work most of the time.  I could try again with a fresh browser if the buttons should work? Am wondering the old GUI tied in the God Mode checks for all toggles.... If GM was false - the toggles wouldn't respond.

torei, 2022-01-18:
@chunkysteveo thanks again.
i think i did some fixes which should resolve bugs :]
still not sure how the godmode works but i did a fix on that too (the command was not recognised)

chunkysteveo, 2022-01-18:
Well well well... that solved all that!!

- The Set temperature slider defaults to 0 on page refresh - **solved**.
- Take control (God Mode) toggle does nothing and changing this does not trigger any update to the data in websockets or MQTT. - **Solved**
- The Heater toggle button also default to off on page refresh no matter the heating state. - **Solved**
- Temp Units toggle retains it's state on page refresh, but the change of this does nothing to the GUI (change values). Value changes in Websockets, but not MQTT. - **Hmm... The slider changes, values, but the top Target and Actual values remain at °C no matter the state of the toggle.**
- Appreciate the Hydrojet toggle has gone from my model (which does not have it) but the label "Hydrojets" is still present.  -**Solved**

This is new, but historic from the old version - the timers for chlorine and filter always show as 19010.62 days till I reset them. But the MQTT shows them as 0, which is what they should be? Assumed this was due to the ESP being reset, but thought those settings would/should be saved? meh? Just picking away at things now.

Oh, the Display 7-segment used to be [   ], then on your branch it said [DSP], but now it's back to [   ]. The 4 wire can't read/see the display so this is correct, and does show [E00] when error flag is raised. Not an issue more of a question/idea of the best approach here, guessing as it is with a blank screen as we can't read characters.

The updates have done wonders though, very impressed!! I need a 6-wire egg!!!

torei, 2022-01-18:
not sure if i can fix that in the web interface. seems like a problem in the BWC.
the web interface is stupid, its just grabbing the content from "BWC::getJSONStates()"
@visualapproach any ideas?

about the "19010.62 days" issue. you are right, that happens when you completely erase your esp because "settings.txt" are stored in the /data/ directory. you can save your settings with calling <yourWeb>:/dir/settings.txt and upload that after flashing a new version. or you just flash the program and use the web interface to upload the /data/ dir.

no idea on that display thingy

why do you need a 6-wire egg? i dont get the point :D

visualapproach, 2022-01-18:
What's the question again? How to get it to show degrees Frankenstein? In that case, yeah IIRC the CIO is hardcoded to just report degrees Science, so I think the best way would be to convert the numbers within the html and don't bother sending any message on the unit button. But I may be wrong

visualapproach, 2022-01-18:
Try swapping unit from the pump display and see if the temp on web gui changes. If not, I was right. Else wrong.

torei, 2022-01-18:
ok, if so - i will see if i can do some web stuff... wondering how it was handled before...

visualapproach, 2022-01-18:
It probably wasn't;)

visualapproach, 2022-01-18:
From the library:
```
  //Not used. Here for compatibility reasons
  states[LOCKEDSTATE] = false;
  states[POWERSTATE] = true;
  states[UNITSTATE] = true;   //Celsius

```
Meaning: those states is not reported by the CIO.

So maybe add js-code like 
`temp.innerhtml = unitbutton.checked ? value.toString() : ((value-32)/1.8).toString();`
and have the unit button update everything temp related somehow

visualapproach, 2022-01-18:
@chunkysteveo The 4-wire has one thing the 6-wire can't do. It can control the two heater elements individually. Not implemented as a user feature, but the code do that. I can think of a couple of use cases, for example if you have weak fuses in the house.

chunkysteveo, 2022-01-18:
Thanks for the info @torei about the settings.txt page, makes sense now after all my resets and changes!

**why do you need a 6-wire egg? i dont get the point :D** - the 6-wire versions of the unit (called the "egg" by alot as they look like eggs). Seems the 6-wire has more abilities to control back and forth. The 4-wire is a monitor or take over the panel situation.

Thanks @visualapproach about the heaters - I like them both on when possible. Why does the 6-wire version not have access to the two heating elements?

visualapproach, 2022-01-18:
> Why does the 6-wire version not have access to the two heating elements?

*It* (the CIO) uses both but we can't control them individually from the ESP.

torei, 2022-01-19:
fixed the temperature.
found the part in the old code:
```cpp
if(msgobj.UNT){
	document.getElementById('atlabel').innerHTML = msgobj.TMP.toString();
	document.getElementById('ttlabel').innerHTML = msgobj.TGT.toString();
} else {
	document.getElementById('atlabel').innerHTML = String(msgobj.TMP * 1.8 + 32);
	document.getElementById('ttlabel').innerHTML = String(msgobj.TGT * 1.8 + 32);
}
```
new code should fix it:
```cpp
// temperature
document.getElementById('atlabel').innerHTML = (msgobj.UNT ? msgobj.TMP.toString() : String(msgobj.TMP * 1.8 + 32));
document.getElementById('ttlabel').innerHTML = (msgobj.UNT ? msgobj.TGT.toString() : String(msgobj.TGT * 1.8 + 32));
document.getElementById('temp').min = (msgobj.UNT ? 20 : 68);
document.getElementById('temp').max = (msgobj.UNT ? 40 : 104);
```

i have updated the 4 wire feature branch. @chunkysteveo please test

chunkysteveo, 2022-01-19:
Ahhh awesome! Do you never take a break?!! Haha

I will update me local repo and test...

torei, 2022-01-19:
only for bananas :-)

chunkysteveo, 2022-01-19:
Oh, I also solved the issue uploading the filesystem and getting Error 0... the two tasks to upload filesystem - the "OTA" one errors. Works with the none OTA and platformio file pointing to the ip. Oh well?!

torei, 2022-01-19:
i'm new to platformio and all the ESP stuff.
once uploaded the filesystem, i cannot upload again. don't know why. never tried OTA on that.
i'm updating the /data/ stuff via web uploader

torei, 2022-01-19:
just because i want to keep my settings and stuff :]

chunkysteveo, 2022-01-19:
The toggle changes the values when flicked, looking good.

I don't want to keep raising issues, but the slider for target temperature seems to toggle to 40 if you change units and it's set to 20 target. Refreshing the page brings back the real value, but the unit toggle seems to mess it up?

It doesn't look like the UNT is passed to the MQTT message either on change? Monitoring it in MQTT it does not change from 1-0-1 etc when toggling? Websockets you can see the change, MQTT there's no change?

chunkysteveo, 2022-01-19:
> just because i want to keep my settings and stuff :]

Wise!

chunkysteveo, 2022-01-19:
Hmm, not sure on the MQTT stuff - I am testing with switches from Home Assistant and commands are sent and actioned, but the MQTT message topic is not always updated for UNT and GOD. 

AIR seems to update on every command whether it's from the web GUI or an MQTT command from Home Assistant. Toggling UNT and GOD does not seem to give updates to MQTT message topic.

Sorry to keep digging and digging. Everyone likes to squash bugs, but I need to stop!

torei, 2022-01-19:
keep digging, not a problem. personally i appreciate that

chunkysteveo, 2022-01-19:
haha, OK, I won't feel too bad then!

I've not got my head around the code, but the GODMODE certainly changes how things are dealt with. There's a note too about checking for changes to trigger an update. - BWC_8266_4w.cpp line 128...

This triggers dataAvailable = true, which in turn triggers the MQTT message to but updated. I am thinking/guessing that on UNT change and GODMODE change toggle, the dataAvailable is not being set to true??

torei, 2022-01-19:
i'm not too deep in the BWC code. but i guess, yes

visualapproach, 2022-01-19:
I'm starting to get my network to behave again. Will take a look later if you want.

torei, 2022-01-19:
created a draft PR #190

visualapproach, 2022-01-19:
> created a draft PR #190

What's the use case of a draft PR? (Not beeing snarky, I just want to learn)

visualapproach, 2022-01-19:
> haha, OK, I won't feel too bad then!
>
> I've not got my head around the code, but the GODMODE certainly changes how things are dealt with. There's a note too about checking for changes to trigger an [...]

Presently, the CIO code reports if anything changes. The UNT and GODMODE (I'm surprised the wording hasn't been questioned 😄 ) is not known by the CIO and can therefore not trigger the dataAvailable. However, I can add a trigger in BWC code, since it's a thing between the web GUI and the BWC. Same with target temperature. The pump doesn't know anything about it's purpose in life, it just follow orders from the DSP/ESP, like "keep the heater on and report the temperature and I'll tell you when to stop heating."

visualapproach, 2022-01-19:
ah, thanks.
Updated dev: unit, set target temp and "godmode" should trigger dataAvailable which should send MQTT and WS messages.

chunkysteveo, 2022-01-19:
> ah, thanks. Updated dev: unit, set target temp and "godmode" should trigger dataAvailable which should send MQTT and WS messages.

Yes please. If anything that also keeps any external inputs such as Home Assistant switches keeping in track. Currently when I flick the switch in HA it goes on then off, never sticks, as MQTT doesn't report the change.

Coolio

torei, 2022-01-19:
merged into the 4wire branch

chunkysteveo, 2022-01-20:
Hmm, quick merge and test and the MQTT is not updating with GOD and UNT still? I need time to dig but at work now.

Steve

visualapproach, 2022-01-22:
I found a bug. It's smashed now. MQTT will update when a state is changed. You can test by switching unit for instance.

chunkysteveo, 2022-01-23:
Awesome. Will test!

chunkysteveo, 2022-01-26:
I'm going to close this issue off now as it's been more than long enough, and the work done @visualapproach  @torei  has far surpassed the original issue! Well done all-round - the latest stable branch is a huge leap forward for the 4-wire crew! Well done on your efforts!

closed 2022-01-26 by chunkysteveo

visualapproach, 2022-01-26:
Yes, big shout out to @torei for his work