visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #191: Target Temperature Slider

mbdawson, 2022-01-20:
I've noticed the slider used to set the target temperature will always on page load set to the default position of 68°F and not the actual target temp. I've attached a recording of the issue. In this example, the hot tub is already at a target temp of 95 degrees with the heater running. I drag the slider to 104°F. As I drag the slider, all of those set points are sent to the hot tub. You can see the target temp drop all the way to 68°F and turn the heater off, then slowly ramps all the way up to 104°F and then turns the heater back on.

This seems like it could easily be fixed by either on page load, read the correct target temp and set the slider to be at that temperature. Or when you drag the slider, it doesn't set the target temperature until you let go of it.

https://streamable.com/j1ep4v

torei, 2022-01-20:
hello @mbdawson 
thanks for the details. do you have the 4-, or the 6-wire version installed?

mbdawson, 2022-01-20:
Newest 6-wire build on a Coleman egg pump.

torei, 2022-01-21:
just created a PR #193 
this may fixes the problem. really not sure about.
mind trying the source branch to have a test?

mbdawson, 2022-01-21:
Forgive me, I'm new to the Github life, so if you'd rather me reply elsewhere, I can.  I wasn't sure what you meant by source branch. Do you mean download the "developer" branch build and upload OTA to it?

Instead I was able to upload the new index.html and WebSocket file using the file uploader. With that combo, no slider appears now. I can still click and change the temperature, but it seems to still have the same behavior.

https://imgur.com/c9wGLv9

When I uploaded the current Master index.html, with your new WebSocket.js, the slider appears and on page load correctly at the current Target Temp, but the slider will not move. So it seems you solved one problem, but created another.

torei, 2022-01-21:
oh ok, not a problem.
try this branch:
https://github.com/torei/WiFi-remote-for-Bestway-Lay-Z-SPA/tree/develop

[image]

mbdawson, 2022-01-21:
Ok. Working better. 

https://streamable.com/axzx6p
In this video I load the page. The slider is in the correct spot. Target temp is already at 96. I drag slider to 104 and let go. Everything works perfectly.

https://streamable.com/dvgh4a
In this video, I filmed right after changing to 104. I reloaded the page fresh and then dragged slider with my finger until I found 94, and then lifted my finger. You can see the target temp dropped all the way to 68 and then back up to 94. 

So from what I can tell, you solved the slider loading on pageload to the correct target temp. Raising the temperature up operates normally. Lowering seems to still have the issue I had previously.

visualapproach, 2022-01-21:
That's a property of the code, not the web interface. Sometimes it misses the target and bottoms out before going up again.

mbdawson, 2022-01-21:
Got it. Thanks for your work on this. 
If the code can't fix that issue, would it make more sense to have the option of not using the slider on the web interface? Such as manually typing in the setpoint I want or the option of up and down arrows to basically mirror the buttons on the hot tub? I honestly love the idea of tapping a box and typing in a setpoint, or a dropdown box that has all the temperatures. On a mobile browser that would look pretty slick of scrolling through to the temperature and selecting OK.

mbdawson, 2022-01-21:
Crudely done but a concept at least.
https://streamable.com/0j6iy1

visualapproach, 2022-01-21:
Thanks for your suggestions. The way of selecting the temperature will not change the behaviour. The ESP is "pressing" UP or DOWN buttons until target temp flashes on the display, but sometimes it misses and has to change "search direction". Exposing the UP/DOWN buttons would not suffer from that issue but personally I don't fancy that solution. It's like manual labor to me, instead of just setting the temp and be done with it. Even if it looks funny for a few seconds. Your opinion is appreciated though and I'll keep it in mind.

mbdawson, 2022-01-21:
Understood on the interface not having any impact on that behaviour. Thank you for addressing my issues. If you need me to test anything else on my 6-pin, let me know.

visualapproach, 2022-01-21:
Will do thank you!

cyberfly79, 2022-02-01:
I also noticed this problem. 
I came across this project last summer in June
first. It still works well with this code, there is no overrun of the set value. Could it not be that the scrolling is too fast and the response is coming more slowly from cio, is that why esp is moving forward? There were 3 "laps" between 20 and 40, by which time we managed to set ...

What has changed from last year's solution?

visualapproach, 2022-02-03:
I have an idea on how to fix it properly.
> What has changed from last year's solution?

I don't remember what I had for breakfast... You could look through the commit history though or compare to the backup branch.

visualapproach, 2022-02-07:
New method implemented to change target temperature. Dev branch. Slower but better IMHO. I also moved the model setting into file "model.h".

mbdawson, 2022-02-07:
So far so good for me on my 6-wire. I don't think it being slower is much of a big deal for anyone. If anything buggy happens, I'll report back.

closed 2022-03-09 by visualapproach