visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #192: Not remembering Wifi Credentials / Not staying connected

Lukey1979, 2022-01-21:
Hi @visualapproach , I have uploaded the newest Master release to my ESP module (success) and connected to AutoPortal, however, once I enter my wifi credentials and click save, all looks as though it is working correctly, I see my ESP module pop up in my connected devices list on my router, but cannot connect to it. After a few minutes it drops from my network and goes back to showing the autoportal network. It did however pop up once and requested a username and password of which i used those listed in your changelog.txt code. this worked and took me to the WebUI however after each reboot, it shows autoportal again. I noticed in your screenshots of the WebUI, there is a Network Config page, however my WebUI, when i can get it to work, does not have anywhere to configure a network. Any ideas on what's happening here?

torei, 2022-01-21:
hello @Lukey1979 
it seems that you did not update your data folder. that's the "upload file system" part.
you can do this by using the web uploader. but i would prefer "erase flash" and do a clean install
**REMEMBER:** to save your settings.txt and stuff to not get your stats lost while updating the data folder

once done that, please come back here and report, thanks! <3 :)

Lukey1979, 2022-01-21:
Hi @torei i absolutely did upload file system. I attempted this multiple times and on 3 different ESP modules. I always build file system data and then upload file system data. Something else is causing it. I tried 3 seperate times using the Dev version on 3 seperate devices and then 3 seperate times using the master version on 3 seperate devices. 

The steps I followed are:
1. Erase Flash
2. Open the 4wire version folder
3. Edit the model.h file to un-comment my version
4. Build
5. Upload
6. Build file system data 
7. Upload file system data

Lukey1979, 2022-01-21:
@torei i just read over what I sent you and realised I'm sounding rude... I apologise and thank you for your help on this 😃

torei, 2022-01-21:
oh well, the 4-wire version is one step behind the 6-wire. you would have to tune into my feature branch.
maybe you can just wait for the PR to be done: #190
you could also just check out my feature directly here:
https://github.com/torei/WiFi-remote-for-Bestway-Lay-Z-SPA/tree/feature/4wireUpdate

closed 2022-03-09 by visualapproach