visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #194: MQTT Enabled, but down, blocks unit control

chunkysteveo, 2022-01-21:
OK, I had a major catastrophe with the Hot Tub after my Home Assistant Mosquito MQTT broker died on me this evening... only a backup from the day before (and subsequent updates from today did not replicate the issue??!.. go figure?!)

But in doing so, my Hot Tob pump started to stop and start, and the temp drop. Numerous reboots of the ESP failed to allow me to connect. I was having a hot dip this evening so I disconnected the ESP. Hooking it up to my PC and watching the terminal showed it was looking at MQTT connection 2 failure loops? I could intermittently connect to the GUI and ping it, but any request to a page was very slow in the serial terminal.

Crisis over and HA is back and running, Hot Tub manual, but need to raise as a question or bug that the ESP needs an MQTT connection before proceeding? The pump (4 wire) was unresponsive to the display buttons etc when this occured. Ideally I would think the MQTT connection is not a blocking event on boot? DSP to CIO must come up no matter the state of the ESP?

Runningn on @torei 4 wire branch, but I think this has been hidden on master till such event that your MQTT connection fails (error 2??)

Thoughts?!! 🤔🤟🏻

visualapproach, 2022-01-21:
Turning on MQTT without a running broker makes the ESP trying to connect every loop(). This slows the program execution vastly and loading pages gets very slow. Communication between DSP-ESP-CIO is most likely suffering too. Especially in the 4 wire version. I recently tried to mitigate this but ran into some strange problems. I hope it will be fixed in a later version.

chunkysteveo, 2022-01-21:
Yeah.... what he said @visualapproach 🤣🤣. 

From a user perspective and real life case - this took hours to diagnose. My whole house hangs off my MQTT broker, wasn't just the Hot Tub (old tasmota, zigbee etc) but there were workarounds. This appeared as a dead unit sadly.

It's good to know this is how it works per say.. and my swtup wasn't something unusual. I'd be keen on looking at the hierarchy of the setup/loop and making sure the display can always override/pass through messages to the CIO no matter the state of the ESP.

MQTT info/control is secondary to hitting buttons on the display. As long as there is breath in the ESP it should relay message no matter 🤣.

Appreciate the work involved and that it's been looked at, would be good to bump this up in pecking order. You've got quite the beast on your hands now @visualapproach @torei 😈.

P.s. compliment sandwich... I've had a whole week of tinkering, OTA, MQTT, @torei  branches, messing with the project without a single hiccup. Its a beast!! Sods law states when I go to use the Tub it goes tits up... (HA to blame!!) kids went to bed in absolute terror in darkness too... so there's that to laugh at, at least.... 😬 still sat here in 40°C water and breathing so life is good 👍🏻

visualapproach, 2022-01-22:
I can think of a worse case: the "bring me beer" button [mentioned here](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/discussions/188#discussioncomment-2014564) out of order 😄 
Anyhow, I updated both 4w and 6w in development version. It *should* work like this now: If MQTT server is down, the ESP only tries to reconnect every 60 seconds, instead of doing it every loop(). Leaving breathing room for everything else to work.

Thank you for you fine words!

chunkysteveo, 2022-01-23:
Hahahaha that would be a worse case for sure!!! I will check out the new updates. Donated too 👍🏻🤟🏻😇

visualapproach, 2022-01-23:
Very kind of you sir! Thanks a lot 🙇

chunkysteveo, 2022-01-26:
Hey - this update certainly helped. I was able to see how it now functions after doing some work on my HA setup and rebooting etc. The pump was on, all good. I turned off the MQTT broker and after a minute - the pump would stop while the MQTT check was being done. After a failed login the pump would resume for another minute, and so on.

I am assuming the MQTT check is a blocking routine in the loop - it would be great to push this out to a non-blocking task without a delay or pause in comms with the CIO, somehow? Sounds like a large change to how the loop works though!

I'm happy to close off this issue though for now.

closed 2022-01-26 by chunkysteveo

visualapproach, 2022-01-26:
OKay, thanks for reporting. I didn't realize that the pump stopped when comms were out for a little while. Now I know.

chunkysteveo, 2022-01-26:
Yeah, that was how it occurred (purely as I could hear it stop, then start up again, and then stop again after a minute, so I knew it was the new MQTT  code).

Didn't test it further to see what happened with the bubbles on, or the heat on, and if that reset those too? I could, but not right now.

visualapproach, 2022-01-26:
There may be a timeout to adjust in the mqtt library. But I'll play with it when I have the time.

chunkysteveo, 2022-01-27:
Ahhh... 30 down to 3? Will test that out when I'm not frozen!!! Haha thanks!

visualapproach, 2022-01-27:
Yes, I don't know how impatient the pump is. Maybe it can tolerate 3 seconds of silence, maybe not. You have to experiment a little to see what value will go unnoticed for the pump and still be enough for mqtt server to respond. Hopefully there is a value that satisfies both.

chunkysteveo, 2022-01-27:
Leave it with me, no problem!