visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #197: esphome this project

tempod, 2022-01-22:
is it possible to convert this project to be used with esphome?

visualapproach, 2022-01-23:
You are welcome to try. I don't think it's worth the effort but don't let that stop you 😸.

chunkysteveo, 2022-01-23:
Look up ximon on gitub.. he had a Lay Z Spa projectand someone forked it to work in ESPHome... think it was the 6 wire models...

tempod, 2022-01-23:
> Look up ximon on gitub.. he had a Lay Z Spa projectand someone forked it to work in ESPHome... think it was the 6 wire models...
i searched but it seems they are all 3 wire and none with esphome. where did you see him?

> You are welcome to try. I don't think it's worth the effort but don't let that stop you 😸.
I don't have the knowledge to do it: P
however i think it is a good idea for maintenance and updating.

closed 2023-11-23 by visualapproach