visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #199: 404 Error - idiots guide required

theandydove, 2022-01-23:
I must admit I am extremely out of my depth on this one, Im ok(ish) with home assistant but im really struggling with this. I just get the 404 error trying to connect now that I have adopted it to my wifi. I read the bit about littlefs data, but I have zero idea how to use VS code, i just clicked the "build platform and filesystem image" but I still get the 404, I have tried clicking the tick and right arrow but still no luck.
Thanks in advance.

visualapproach, 2022-01-23:
Maybe upload filesystem image also?

theandydove, 2022-01-29:
Thank you, that worked. I only just had a chance to attempt it

bubble07, 2022-02-18:
I also fell into this trap, however my issue was that I couldn't see an option for build filesystem image or upload filesystem image in PLATFORMIO-->nodemcuv2-->Platform.   It took me a while to work out what I was doing wrong, but the menu option appears after I installed python and restarted Visual Studio Code. Probably obvious for someone who knows what they are doing, but for a noob like me, it was a pitfall I fell into!

It's a great project, I've ordered the PCB and am looking forward to getting on with the hardware installation.

closed 2023-03-13 by visualapproach