visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #200: Palm Spring 4-wire 54129 not responding and incorrect temperature on display

pdecat, 2022-01-23:
Hi @visualapproach,

I've being eyeing at your impressive work on this project for a year or so and finally decided to give it a try with my Bestway Palm Spring 4-wire 54129 (egg-shape, just bubbles, no hydrojets).

For starters, he are a few pictures of this model:

[image]

[image]

[image]

[image]

[image]

[image]

So far, my pump is not responding to any actions from the display or the web application.
And the temperature shown on the display is incorrect, 04 when the selected temperature unit on the display is Celsius and 39 when it is Fahrenheit:

[image]

[image]

[image]

Also, the temperature is not read by the web application:

[image]

This is using [current](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/commit/af1206be90ab85c29fced4aec0c1a4e71ad3fabc) `development`, and I've tested each of the 4 models available in [`model.h`](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/blob/af1206be90ab85c29fced4aec0c1a4e71ad3fabc/Code/4-wire-version/lib/BWC4W/model.h) and they all behave the same.

Any idea what may be wrong with my setup?

visualapproach, 2022-01-23:
Did you "take control" in the web GUI before evaluating the web functionality?
I'd say check your soldering and wires. Convenient for me to blame hardware, yes, but I actually think it's the most probable cause. 
For starters you can check this:
Is the blue LED blinking? It means there is traffic between CIO and ESP. Check browser console for any clues (you can paste here), in particular the dsp_tx value. If it says false or 0 you can suspect that line.

pdecat, 2022-01-24:
It was the hardware: my level converter was poorly seated because I soldered the pins the wrong way (long side up) :facepalm:

I fixed that and now the LED is blinking and I can take control, but toggling heater powers-on the bubble.

Now I need to identify the proper model to use...

pdecat, 2022-01-24:
It seems to be working better with `NO54154`.

pdecat, 2022-01-24:
Thanks again @visualapproach!

Let's close this issue.

visualapproach, 2022-01-24:
Your welcome! Happy it worked out

pdecat, 2022-01-25:
Hmm, my ESP is rebooting every few hours (once at 2 AM, 30 minutes ago) and looses its state and settings, so heating stops unexpectedly.

Any idea what could that be caused by?

visualapproach, 2022-01-25:
May be brown out resets (low power). What's the reboot cause? Look at the config page at the bottom. 4-wire systems seem to have weaker power supply to the ESP. You can see if it works longer by disconnecting the display completely from the ESP, and disconnect the 5V from the CIO. Keep the GND and data lines. Then run the ESP from a powerbank. It's important to disconnect as I described or you may release the smoke.
Typically the brownout would appear when the heater element is turning on.

pdecat, 2022-01-25:
```
Last boot: 25/01/2022, 10:09:56 Hardware Watchdog
``|

visualapproach, 2022-01-25:
okay, strange. I'll let a device run here for a while and see if I can replicate. No pump to connect to though.

visualapproach, 2022-01-26:
Mine has been running now for 22h without a hickup, but as I said, without a pump connected to it. You could try erasing the flash completely and recompile, upload, upload files, and make a powercycle on the ESP.

pdecat, 2022-01-26:
Apparently, that reboot issue only happens when `Take control` is checked in UI.

I left it unchecked since yesterday at noon and my ESP had not rebooted by itself.

In the meantime, I faced a second issue: even with `Take control` off, my unit got into `E02` error twice.
I've already replaced the pump's impeller in October 2021 so I doubt it's happening again already as it previous impeller lasted several years since I originally bought the spa.

For now, I've removed the ESP to check if it reproduces without it.

chunkysteveo, 2022-01-26:
Have you cleaned/de-scaled your unit? Here in the UK we need to de-scale our lay y spa pump units as we do get a build up of calcium deposits due to the water type.

I was plagued with a lot of E02 errors last year till I bought a "E02 blaster" kit online (basically a tube that connects inlet and outlet and allows you to run de-scaler in the unit for 30 mins or so). Since doing these de-scalers actions my pump unit is always pushing out small chunks of white (calcium) deposit as it's all becoming loose. In turn I spotted that the outlet nozzle was becoming clogged over time with these deposits.  A good brush with a stiff brush so all the nozzle "holes" are poked and cleared would dislodge the deposits that got clogged.

I was sure my unit was becoming defective and the impeller going, but another year in and a de-scaler every 6-9 months (if in daily use) and my E02 errors have all but disappeared - it amazed me!

visualapproach, 2022-01-26:
Thanks @chunkysteveo for chiming in. Good suggestion. Unstable communication between ESP and the pump usually summons an E13. Some people seem to have that problem with their 4-wire pumps.
@pdecat I'm not aware of anyone that was helped by this tip but for the sake of it - try running @160 Mhz (edit platformio.ini) in addition to my previous suggestions.

pdecat, 2022-01-27:
20h of uptime without any issue since I've redeployed in 160 Mhz. That's the only thing I changed as I prefer to check one thing at a time.

Note: my ESP8266 is a 12F  designed to run at that speed.

Let me try with `Take control` checked now...

pdecat, 2022-01-27:
And a new `E02` after having taken control for two hours.

I'll leave it without `Take control` checked for a longer period...

visualapproach, 2022-01-27:
I'll see if I can do something

pdecat, 2022-01-27:
Well, got a `E02` with `Take control` unchecked, probably unrelated to that setting then...

chunkysteveo, 2022-01-27:
Have you checked the quality of the connections on the PCB? I had loads of intermittent flickering of the pump and air bubbles that I could see in the websockets and MQTT. You could see the Web GUI toggles also sometimes flick on and off... I had it setup with headers and pins so I could easily remove the cables. The pin headers has some "wiggle play" to them, so after soldering the wires direct to the PCB it's been solid as a rock - hardware issue.

I see from the pics that the wires are soldered through the holes direct to the PCB, which is good. But worth checking each solder joint for any possible dry or "sketchy" join, and re-seating the LLC and ESP as a test too?

pdecat, 2022-01-27:
I do not see any flickering in websocket frames and MQTT messages. Web GUI seems pretty stable too.

I checked every point to point connections on the PCB with a multi-meter, and also ensure there were no short circuit between neighbor joins. I'll try re-seating the LLC and ESP next time an error happen.

Thanks for all the tips!

PS: I've been looking at E02 blasters but found none in France where I leave, and those from UK have prohibitive shipping fees (twice the cost of the product).

chunkysteveo, 2022-01-27:
Ahh yes. You can make your own (the ones on eBay are all hand made!) With some PVC pipe and some plumbing fittings. It's just a vertical large pipe to hold about 1.5-2 L of water, and two inlets drilled into it. I believe the Layz Spa connectors are standard sizes that fit metric thread. YouTube has a few DIY videos using a bucket also... a weekend project??!! Good luck!

Steve

visualapproach, 2022-01-27:
A real E02 is only triggered when the pump is on. I suggest this test to find out what's causing it:
Either "take control" and leave the filter pump OFF a couple of hours, OR
remove the ESP device and set the pump ON.
Wait.
1. Got E02 error with ESP running and pump OFF -> ESP HW or SW is the likely cause.
2. No E02 error with ESP running and pump OFF -> descaling is likely needed.
3. Got E02 error without ESP and pump ON -> descaling is likely needed.
4. No E02 error without ESP and pump ON -> ESP HW or SW is the likely cause.
Makes sense?

pdecat, 2022-01-27:
The `E02` error meaning water flow is not detected through the pump, I don't believe it would ever trigger if pump is off.

Just got a new `E02` error, I will re-seat ESP and LLC, but will run without ESP for a few days before reinstalling it.

visualapproach, 2022-01-27:
> The E02 error meaning water flow is not detected through the pump, I don't believe it would ever trigger if pump is off.

Exactly! So if you get it anyway it's probably the ESP's fault. Or it's peripherals like cables.
Likewise, if running without the ESP and you get the error it is because of low water flow.

visualapproach, 2022-01-27:
I don't mistrust you at all and it would be one heck of a coincidence, but it's best to be methodical and narrow down the possible cause as much as possible. The more things we can eliminate the less impossible it will be to remote debug the error.

pdecat, 2022-01-31:
So, my spa pump has been running flawlessly without ESP for more than 4 days, and still is.

Would it be possible to add verbose debug logging to troubleshoot what is going on? Maybe through an MQTT topic to make it easier. And with an option in configuration to avoid impairing performance when not needed.

visualapproach, 2022-01-31:
I've been thinking about logging too. Need some free time first to work on it. I'll let you know here when I've looked at it

visualapproach, 2022-02-03:
@pdecat In the meantime I think you should run the `test-hardware\4-wire-serial-bypass` program also. It just copy the data between the CIO anad DSP. No modification at all except for a couple of microseconds delay which shouldn't be an issue. If it works, the code is probably causing the issues. If not, it's probably hardware related.

visualapproach, 2022-02-03:
Log feature in dev branch. Edit config.h and set DEBUGSERIAL = true;
In browser - open console and see if the debugging data is showing. If so you can let it save output to file until you get an error.

pdecat, 2022-02-04:
Thanks @visualapproach, I have installed the `test-hardware\4-wire-serial-bypass` program, it's been working fine for half an hour, will report how it goes...

pdecat, 2022-02-05:
Hi, `test-hardware\4-wire-serial-bypass` program has been running for more than 24 hours without issues. I'll leave it a few more days as is...

visualapproach, 2022-02-05:
Okay so cables and HW is probably fine then. Good to know. Next step would be to log the messages I appended in the dev branch. Specifically the ciotx ciorx dsptx and dsprx payloads.

pdecat, 2022-02-06:
Hi, I've installed the latest development branch with `DEBUGSERIAL = true;`, and websocket communication is failing completely for `DEBUG` messages when ESP is installed on Spa:

```
WebSocket.js:42 WebSocket connection to 'ws://192.168.1.40:81/' failed: Could not decode a text frame as UTF-8.
```

[image]

When ESP is not plugged not plugged to Spa, I can see `DEBUG` messages with empty payload.

Maybe a empty encoding of the payload?

pdecat, 2022-02-06:
Also, it triggered an E02 error almost as soon as it was plugged to the spa.

visualapproach, 2022-02-06:
Okay understood

pdecat, 2022-02-06:
I added the following encoding code:

```
diff --git a/Code/4-wire-version/lib/BWC4W/BWC_8266_4w.cpp b/Code/4-wire-version/lib/BWC4W/BWC_8266_4w.cpp
index 068e3ab..331101e 100644
--- a/Code/4-wire-version/lib/BWC4W/BWC_8266_4w.cpp
+++ b/Code/4-wire-version/lib/BWC4W/BWC_8266_4w.cpp
@@ -561,6 +561,14 @@ String BWC::getJSONCommandQueue(){
   return jsonmsg;
 }

+String BWC::encodeBufferToString(uint8_t buf[7]){
+  String str = String();
+  for (unsigned long i = 0; i < 7; i++) {
+    str += String(buf[i], HEX);
+  }
+  return str;
+}
+
 String BWC::getSerialBuffers(){
   ESP.wdtFeed();
     DynamicJsonDocument doc(512);
@@ -568,10 +576,10 @@ String BWC::getSerialBuffers(){
     // Set the values in the document
     doc["CONTENT"] = "DEBUG";
     doc["TIME"] = _timestamp;
-    doc["TODSP"] = _cio.to_DSP_buf;
-    doc["TOCIO"] = _cio.to_CIO_buf;
-    doc["FROMDSP"] = _cio.from_DSP_buf;
-    doc["FROMCIO"] = _cio.from_DSP_buf;
+    doc["TODSP"] = encodeBufferToString(_cio.to_DSP_buf);
+    doc["TOCIO"] = encodeBufferToString(_cio.to_CIO_buf);
+    doc["FROMDSP"] = encodeBufferToString(_cio.from_DSP_buf);
+    doc["FROMCIO"] = encodeBufferToString(_cio.from_DSP_buf);

     // Serialize JSON to string
     String json;
```

pdecat, 2022-02-06:
Seems to be working better:

[image]

I did not enable `Take control`, so I'm not sure why `TODSP` value is not the same as the others:
```
{
CONTENT: "DEBUG"
FROMCIO: "bb3100114fd"
FROMDSP: "bb3100114fd"
TIME: 1644141145
TOCIO: "bb3100114fd"
TODSP: "bc328012cfd"
}
```

pdecat, 2022-02-06:
Ok, `FROMCIO` was showing value of `FROMDSP`

pdecat, 2022-02-06:
Submitted https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/pull/210

pdecat, 2022-02-06:
Debug content is now:

```
{
CONTENT: "DEBUG"
FROMCIO: "bc328012cfd"
FROMDSP: "bb3100114fd"
TIME: 1644141938
TOCIO: "bb3100114fd"
TODSP: "bc328012cfd"
}
```

visualapproach, 2022-02-06:
Great, I'll look at this later.

pdecat, 2022-02-06:
Something I've noticed is that performing an OTA update of the firmware triggers an E02 error.
Maybe it's caused by the fact that the ESP is unresponsive for a few seconds, and maybe it is the same issue that happens from time to time.

visualapproach, 2022-02-06:
Display will think CIO is dead and throws an error. Are you getting E02 on the actual display or just in the web gui?

pdecat, 2022-02-06:
I get E02 on the actual display, web GUI only shows E00.

visualapproach, 2022-02-06:
I adjusted your string formatting a little. 0x0a will show as 'a' which makes it difficult to see where one byte ends and the next begins. Just added spaces between bytes. Thanks for your PR btw. Can you recompile and upload the results before and after an error?

pdecat, 2022-02-07:
I've managed to capture the following log happening before and after an error:

```
{"CONTENT":"STATES","TIME":1644236353,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":0,"FLT":1,"TGT":20,"TMP":40,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644236353,"CLTIME":1644216105,"FTIME":1644216106,"UPTIME":21138,"PUMPTIME":21129,"HEATINGTIME":13027,"AIRTIME":0,"JETTIME":0,"KWH":7109.01123,"COST":7109.01123,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644236353,"TODSP":"bc 3 28 0 1 2c fd ","TOCIO":"bb 3 10 0 1 14 fd ","FROMDSP":"bb 3 10 0 1 14 fd ","FROMCIO":"bc 3 28 0 1 2c fd "}
{"CONTENT":"STATES","TIME":1644236354,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":0,"FLT":1,"TGT":20,"TMP":40,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644236354,"CLTIME":1644216105,"FTIME":1644216106,"UPTIME":21139,"PUMPTIME":21130,"HEATINGTIME":13027,"AIRTIME":0,"JETTIME":0,"KWH":7109.01123,"COST":7109.01123,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644236354,"TODSP":"bc 3 28 0 1 2c fd ","TOCIO":"bb 3 10 0 1 14 fd ","FROMDSP":"bb 3 10 0 1 14 fd ","FROMCIO":"bc 3 28 0 1 2c fd "}
{"CONTENT":"STATES","TIME":1644236355,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":0,"FLT":1,"TGT":20,"TMP":40,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644236355,"CLTIME":1644216105,"FTIME":1644216106,"UPTIME":21140,"PUMPTIME":21131,"HEATINGTIME":13027,"AIRTIME":0,"JETTIME":0,"KWH":7109.01123,"COST":7109.01123,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644236355,"TODSP":"bc 3 28 0 1 2c fd ","TOCIO":"bb 3 10 0 1 14 fd ","FROMDSP":"bb 3 10 0 1 14 fd ","FROMCIO":"bc 3 28 0 1 2c fd "}
{"CONTENT":"STATES","TIME":1644236359,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":0,"FLT":1,"TGT":20,"TMP":40,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644236359,"CLTIME":1644216105,"FTIME":1644216106,"UPTIME":21144,"PUMPTIME":21134,"HEATINGTIME":13027,"AIRTIME":0,"JETTIME":0,"KWH":7109.01123,"COST":7109.01123,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644236359,"TODSP":"bc 3 28 0 1 2c fd ","TOCIO":"bb 3 10 0 1 14 fd ","FROMDSP":"bb 3 10 0 1 14 fd ","FROMCIO":"bc 3 28 0 1 2c fd "}
{"CONTENT":"STATES","TIME":1644236359,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":0,"FLT":1,"TGT":20,"TMP":40,"CH1":69,"CH2":48,"CH3":48,"JET":0,"ERR":2,"GOD":false}
{"CONTENT":"TIMES","TIME":1644236359,"CLTIME":1644216105,"FTIME":1644216106,"UPTIME":21144,"PUMPTIME":21134,"HEATINGTIME":13027,"AIRTIME":0,"JETTIME":0,"KWH":7109.01123,"COST":7109.01123,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644236359,"TODSP":"bc 3 28 2 1 2e fd ","TOCIO":"bb 3 10 0 1 14 fd ","FROMDSP":"bb 3 10 0 1 14 fd ","FROMCIO":"bc 3 28 2 1 2e fd "}
{"CONTENT":"STATES","TIME":1644236359,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":0,"TGT":20,"TMP":40,"CH1":69,"CH2":48,"CH3":48,"JET":0,"ERR":2,"GOD":false}
{"CONTENT":"TIMES","TIME":1644236359,"CLTIME":1644216105,"FTIME":1644216106,"UPTIME":21144,"PUMPTIME":21134,"HEATINGTIME":13027,"AIRTIME":0,"JETTIME":0,"KWH":7109.01123,"COST":7109.01123,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644236359,"TODSP":"bc 3 28 2 1 2e fd ","TOCIO":"bb 3 0 0 1 14 fd ","FROMDSP":"fd bb 3 0 0 1 4 ","FROMCIO":"bc 3 28 2 1 2e fd "}
{"CONTENT":"STATES","TIME":1644236359,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":0,"FLT":0,"TGT":20,"TMP":40,"CH1":69,"CH2":48,"CH3":48,"JET":0,"ERR":2,"GOD":false}
{"CONTENT":"TIMES","TIME":1644236359,"CLTIME":1644216105,"FTIME":1644216106,"UPTIME":21144,"PUMPTIME":21134,"HEATINGTIME":13027,"AIRTIME":0,"JETTIME":0,"KWH":7109.01123,"COST":7109.01123,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644236359,"TODSP":"bc 3 28 2 1 2e fd ","TOCIO":"bb 3 0 0 1 4 fd ","FROMDSP":"bb 3 0 0 1 4 fd ","FROMCIO":"bc 3 28 2 1 2e fd "}
{"CONTENT":"STATES","TIME":1644236359,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":0,"FLT":0,"TGT":20,"TMP":40,"CH1":69,"CH2":48,"CH3":48,"JET":0,"ERR":2,"GOD":false}
{"CONTENT":"TIMES","TIME":1644236359,"CLTIME":1644216105,"FTIME":1644216106,"UPTIME":21144,"PUMPTIME":21134,"HEATINGTIME":13027,"AIRTIME":0,"JETTIME":0,"KWH":7109.01123,"COST":7109.01123,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644236359,"TODSP":"bc 3 28 2 1 2e fd ","TOCIO":"bb 3 0 0 1 4 fd ","FROMDSP":"bb 3 0 0 1 4 fd ","FROMCIO":"bc 3 28 2 1 2e fd "}
```

visualapproach, 2022-02-07:
Very good work on the logging! I was thinking it might have been a single bit error due to noise on the wires, but it doesn't make sense. Firstly because it's consistent over many rows. (fourth byte FROMCIO is a 2 instead of 0) and secondly and more important - the sixth byte is a checksum and it is also consistent with the error byte. Meaning the ESP has received the correct data from the CIO. There is an anomaly in FROMDSP and TOCIO I have to look into. And there is a chance that the CIO read the data incorrect but in that case it would throw E13 error instead. I think.

visualapproach, 2022-02-07:
Hello again @pdecat 
New dev version uploaded: Updated the software serial library, and the espressif8266 platform.
Made some experimental tweaks in my own library regarding the communication.
Not sure it will help but I'd like you to retry with new code.

visualapproach, 2022-02-07:
You may need to click the platformio icon (the alien head) and "Clean all" before compiling.

pdecat, 2022-02-08:
I've installed the new version. One thing I've noticed is that it takes way longer to initialize, something like 1 or 2 minutes. And during that time, physical display shows `04`, physical buttons react to touches, but do not control the pump, and wifi is inaccessible.

pdecat, 2022-02-08:
Got a new error after a few hours:

```
{"CONTENT":"DEBUG","TIME":1644338691,"TODSP":"bc 3 29 0 1 2d fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd "}
{"CONTENT":"STATES","TIME":1644338692,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":41,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338692,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20497,"PUMPTIME":20415,"HEATINGTIME":17177,"AIRTIME":1810,"JETTIME":0,"KWH":9693.010742,"COST":9693.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338692,"TODSP":"bc 3 29 0 1 2d fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd "}
{"CONTENT":"STATES","TIME":1644338693,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":41,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338693,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20498,"PUMPTIME":20416,"HEATINGTIME":17178,"AIRTIME":1810,"JETTIME":0,"KWH":9694.010742,"COST":9694.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338693,"TODSP":"bc 3 29 0 1 2d fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd "}
{"CONTENT":"STATES","TIME":1644338714,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":41,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338714,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20519,"PUMPTIME":20437,"HEATINGTIME":17199,"AIRTIME":1810,"JETTIME":0,"KWH":9706.010742,"COST":9706.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338714,"TODSP":"bc 3 29 0 1 2d fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd "}
{"CONTENT":"STATES","TIME":1644338735,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":42,"CH1":69,"CH2":48,"CH3":48,"JET":0,"ERR":2,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338735,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20540,"PUMPTIME":20458,"HEATINGTIME":17220,"AIRTIME":1810,"JETTIME":0,"KWH":9717.010742,"COST":9717.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338735,"TODSP":"bc 3 2a 2 1 30 fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 2a 2 1 30 fd "}
{"CONTENT":"STATES","TIME":1644338756,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":44,"CH1":69,"CH2":48,"CH3":48,"JET":0,"ERR":2,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338756,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20561,"PUMPTIME":20479,"HEATINGTIME":17241,"AIRTIME":1810,"JETTIME":0,"KWH":9728.010742,"COST":9728.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338756,"TODSP":"bc 3 2c 2 1 32 fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 2c 2 1 32 fd "}
{"CONTENT":"STATES","TIME":1644338776,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":46,"CH1":69,"CH2":48,"CH3":48,"JET":0,"ERR":2,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338776,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20581,"PUMPTIME":20499,"HEATINGTIME":17261,"AIRTIME":1810,"JETTIME":0,"KWH":9738.010742,"COST":9738.010742,"FINT":30,"CLINT":14,"TTTT":0}
```

pdecat, 2022-02-09:
When monitoring ESP boot phase via USB, it takes a minute before displaying the first `WiFi > using WiFi configuration with SSID "mywifi"` message:

```
> Executing task: platformio device monitor --environment nodemcuv2 <

--- Available filters and text transformations: colorize, debug, default, direct, esp8266_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
WiFi > using WiFi configuration with SSID "mywifi"
WiFi > Trying to connect ....
.
.
WiFi > Connected.
 SSID: "mywifi"
 IP: "192.168.1.40"
OTA > ready
HTTP > server started
WebSocket > server started
MQTT > Buffer size successfully increased
MQTT > Connecting ... success!
End of setup()
```

visualapproach, 2022-02-09:
I wouldn't worry too much about that. If startup takes too long it would give errors at start, not in a couple of hours. I'll think of the next step, I have some ideas. but having a little shortage of free time right now. Posting here when I have something.

visualapproach, 2022-02-09:
Couple of questions:
Your logs shows 42C. Do the regulator not work or is the native pump keeping it +/- 2C?
Does the ERR2 come up without heater on?
Latest log shows 20 second between updates. Did you cut out the rest or is it that slow now?

visualapproach, 2022-02-09:
OK I guess the stock thermometer is a degree off. Good, we don't want a runaway situation on the heater.
I'll be back about the rest.

pdecat, 2022-02-09:
I still have a larger part of logs available from that same error:

```
{"CONTENT":"DEBUG","TIME":1644338675,"TODSP":"bc 3 29 0 1 2d fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd "}
{"CONTENT":"STATES","TIME":1644338676,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":41,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338676,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20481,"PUMPTIME":20399,"HEATINGTIME":17161,"AIRTIME":1810,"JETTIME":0,"KWH":9685.010742,"COST":9685.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338676,"TODSP":"bc 3 29 0 1 2d fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd "}
{"CONTENT":"STATES","TIME":1644338677,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":41,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338677,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20482,"PUMPTIME":20400,"HEATINGTIME":17162,"AIRTIME":1810,"JETTIME":0,"KWH":9685.010742,"COST":9685.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338677,"TODSP":"bc 3 29 0 1 2d fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd "}
{"CONTENT":"STATES","TIME":1644338680,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":41,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338680,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20485,"PUMPTIME":20403,"HEATINGTIME":17165,"AIRTIME":1810,"JETTIME":0,"KWH":9687.010742,"COST":9687.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338680,"TODSP":"bc 3 29 0 1 2d fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd "}
{"CONTENT":"STATES","TIME":1644338681,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":41,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338681,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20486,"PUMPTIME":20404,"HEATINGTIME":17166,"AIRTIME":1810,"JETTIME":0,"KWH":9687.010742,"COST":9687.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338681,"TODSP":"bc 3 29 0 1 2d fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd "}
{"CONTENT":"STATES","TIME":1644338682,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":41,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338682,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20487,"PUMPTIME":20405,"HEATINGTIME":17167,"AIRTIME":1810,"JETTIME":0,"KWH":9688.010742,"COST":9688.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338682,"TODSP":"bc 3 29 0 1 2d fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd "}
{"CONTENT":"STATES","TIME":1644338683,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":41,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338683,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20488,"PUMPTIME":20406,"HEATINGTIME":17168,"AIRTIME":1810,"JETTIME":0,"KWH":9688.010742,"COST":9688.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338683,"TODSP":"bc 3 29 0 1 2d fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd "}
{"CONTENT":"STATES","TIME":1644338684,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":41,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338684,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20489,"PUMPTIME":20407,"HEATINGTIME":17169,"AIRTIME":1810,"JETTIME":0,"KWH":9689.010742,"COST":9689.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338684,"TODSP":"bc 3 29 0 1 2d fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd "}
{"CONTENT":"STATES","TIME":1644338685,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":41,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338685,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20490,"PUMPTIME":20408,"HEATINGTIME":17170,"AIRTIME":1810,"JETTIME":0,"KWH":9689.010742,"COST":9689.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338685,"TODSP":"bc 3 29 0 1 2d fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd "}
{"CONTENT":"STATES","TIME":1644338688,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":41,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338688,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20493,"PUMPTIME":20411,"HEATINGTIME":17173,"AIRTIME":1810,"JETTIME":0,"KWH":9691.010742,"COST":9691.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338688,"TODSP":"bc 3 29 0 1 2d fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd "}
{"CONTENT":"STATES","TIME":1644338689,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":41,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
[... cut ...]
{"CONTENT":"STATES","TIME":1644338838,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":47,"CH1":69,"CH2":48,"CH3":48,"JET":0,"ERR":2,"GOD":false}
{"CONTENT":"TIMES","TIME":1644338838,"CLTIME":1644318297,"FTIME":1644318298,"UPTIME":20643,"PUMPTIME":20561,"HEATINGTIME":17323,"AIRTIME":1810,"JETTIME":0,"KWH":9772.010742,"COST":9772.010742,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644338838,"TODSP":"bc 3 2f 2 1 35 fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMDSP":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 2f 2 1 35 fd "}
```

visualapproach, 2022-02-09:
Lot of work for you, but just for fun, can you see if it makes any difference if you set the baud rate to 
  cio_serial.begin(11000, SWSERIAL_8N1, D2, D3, false, 64);
  cio_serial.setTimeout(100);
  dsp_serial.begin(11000, SWSERIAL_8N1, D6, D7, false, 64);
  dsp_serial.setTimeout(100);
in BWC_8266_4W.cpp

visualapproach, 2022-02-09:
actually, I uploaded another version now, that sends even the dismissed messages. I didn't change baud rate from 9600 but you can play with it. Change one thing at a time.

pdecat, 2022-02-10:
Updated to latest development version and it did not work at all, showing only zeros in the debug traces:

```
{"CONTENT":"STATES","TIME":1644478714,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":20,"TMP":0,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644478714,"CLTIME":0,"FTIME":0,"UPTIME":81,"PUMPTIME":0,"HEATINGTIME":0,"AIRTIME":0,"JETTIME":0,"KWH":0,"COST":0,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644478714,"FROMDSP":"fd bb fd bb bb fd bb ","TOCIO":"0 0 0 0 0 0 0 ","FROMCIO":"0 0 0 0 0 0 0 ","TODSP":"0 0 0 0 0 0 0 ","FROMDSPFAIL":"fd bb fd bb bb fd 0 ","LENDSP":6,"FROMCIOFAIL":"0 0 0 0 0 0 0 ","LENCIO":0}
{"CONTENT":"STATES","TIME":1644478714,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":20,"TMP":0,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644478714,"CLTIME":0,"FTIME":0,"UPTIME":82,"PUMPTIME":0,"HEATINGTIME":0,"AIRTIME":0,"JETTIME":0,"KWH":0,"COST":0,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644478714,"FROMDSP":"bb fd fd bb bb fd bb ","TOCIO":"0 0 0 0 0 0 0 ","FROMCIO":"0 0 0 0 0 0 0 ","TODSP":"0 0 0 0 0 0 0 ","FROMDSPFAIL":"bb fd fd bb bb fd 0 ","LENDSP":2,"FROMCIOFAIL":"0 0 0 0 0 0 0 ","LENCIO":0}
{"CONTENT":"STATES","TIME":1644478714,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":20,"TMP":0,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644478714,"CLTIME":0,"FTIME":0,"UPTIME":82,"PUMPTIME":0,"HEATINGTIME":0,"AIRTIME":0,"JETTIME":0,"KWH":0,"COST":0,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644478714,"FROMDSP":"bb fd fd bb bb fd bb ","TOCIO":"0 0 0 0 0 0 0 ","FROMCIO":"0 0 0 0 0 0 0 ","TODSP":"0 0 0 0 0 0 0 ","FROMDSPFAIL":"bb fd fd bb bb fd 0 ","LENDSP":2,"FROMCIOFAIL":"0 0 0 0 0 0 0 ","LENCIO":0}
{"CONTENT":"STATES","TIME":1644478714,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":20,"TMP":0,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644478714,"CLTIME":0,"FTIME":0,"UPTIME":82,"PUMPTIME":0,"HEATINGTIME":0,"AIRTIME":0,"JETTIME":0,"KWH":0,"COST":0,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644478714,"FROMDSP":"bb fd fd bb bb fd bb ","TOCIO":"0 0 0 0 0 0 0 ","FROMCIO":"0 0 0 0 0 0 0 ","TODSP":"0 0 0 0 0 0 0 ","FROMDSPFAIL":"bb fd fd bb bb fd 0 ","LENDSP":2,"FROMCIOFAIL":"0 0 0 0 0 0 0 ","LENCIO":0}
{"CONTENT":"STATES","TIME":1644478714,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":20,"TMP":0,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644478714,"CLTIME":0,"FTIME":0,"UPTIME":82,"PUMPTIME":0,"HEATINGTIME":0,"AIRTIME":0,"JETTIME":0,"KWH":0,"COST":0,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644478714,"FROMDSP":"bb fd fd bb bb fd bb ","TOCIO":"0 0 0 0 0 0 0 ","FROMCIO":"0 0 0 0 0 0 0 ","TODSP":"0 0 0 0 0 0 0 ","FROMDSPFAIL":"bb fd fd bb bb fd 0 ","LENDSP":2,"FROMCIOFAIL":"0 0 0 0 0 0 0 ","LENCIO":0}
{"CONTENT":"STATES","TIME":1644478714,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":20,"TMP":0,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644478714,"CLTIME":0,"FTIME":0,"UPTIME":82,"PUMPTIME":0,"HEATINGTIME":0,"AIRTIME":0,"JETTIME":0,"KWH":0,"COST":0,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644478714,"FROMDSP":"bb fd fd bb bb fd bb ","TOCIO":"0 0 0 0 0 0 0 ","FROMCIO":"0 0 0 0 0 0 0 ","TODSP":"0 0 0 0 0 0 0 ","FROMDSPFAIL":"bb fd fd bb bb fd 0 ","LENDSP":2,"FROMCIOFAIL":"0 0 0 0 0 0 0 ","LENCIO":0}
{"CONTENT":"STATES","TIME":1644478714,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":20,"TMP":0,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644478714,"CLTIME":0,"FTIME":0,"UPTIME":82,"PUMPTIME":0,"HEATINGTIME":0,"AIRTIME":0,"JETTIME":0,"KWH":0,"COST":0,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644478714,"FROMDSP":"bb fd fd bb bb fd bb ","TOCIO":"0 0 0 0 0 0 0 ","FROMCIO":"0 0 0 0 0 0 0 ","TODSP":"0 0 0 0 0 0 0 ","FROMDSPFAIL":"bb fd fd bb bb fd 0 ","LENDSP":2,"FROMCIOFAIL":"0 0 0 0 0 0 0 ","LENCIO":0}
{"CONTENT":"STATES","TIME":1644478715,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":20,"TMP":0,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644478715,"CLTIME":0,"FTIME":0,"UPTIME":82,"PUMPTIME":0,"HEATINGTIME":0,"AIRTIME":0,"JETTIME":0,"KWH":0,"COST":0,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644478715,"FROMDSP":"bb fd fd bb bb fd bb ","TOCIO":"0 0 0 0 0 0 0 ","FROMCIO":"0 0 0 0 0 0 0 ","TODSP":"0 0 0 0 0 0 0 ","FROMDSPFAIL":"bb fd fd bb bb fd 0 ","LENDSP":2,"FROMCIOFAIL":"0 0 0 0 0 0 0 ","LENCIO":0}
{"CONTENT":"STATES","TIME":1644478715,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":20,"TMP":0,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644478715,"CLTIME":0,"FTIME":0,"UPTIME":82,"PUMPTIME":0,"HEATINGTIME":0,"AIRTIME":0,"JETTIME":0,"KWH":0,"COST":0,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644478715,"FROMDSP":"bb fd fd bb bb fd bb ","TOCIO":"0 0 0 0 0 0 0 ","FROMCIO":"0 0 0 0 0 0 0 ","TODSP":"0 0 0 0 0 0 0 ","FROMDSPFAIL":"bb fd fd bb bb fd 0 ","LENDSP":2,"FROMCIOFAIL":"0 0 0 0 0 0 0 ","LENCIO":0}
{"CONTENT":"STATES","TIME":1644478715,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":20,"TMP":0,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
[... cut ...]
{"CONTENT":"STATES","TIME":1644478970,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":1,"FLT":1,"TGT":20,"TMP":41,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}
{"CONTENT":"TIMES","TIME":1644478970,"CLTIME":0,"FTIME":0,"UPTIME":95,"PUMPTIME":7,"HEATINGTIME":7,"AIRTIME":0,"JETTIME":0,"KWH":3,"COST":3,"FINT":30,"CLINT":14,"TTTT":0}
{"CONTENT":"DEBUG","TIME":1644478970,"FROMDSP":"bb 3 1b 0 1 1f fd ","TOCIO":"bb 3 1b 0 1 1f fd ","FROMCIO":"bc 3 29 0 1 2d fd ","TODSP":"bc 3 29 0 1 2d fd ","FROMDSPFAIL":"fd 0 0 0 0 0 0 ","LENDSP":1,"FROMCIOFAIL":"0 0 0 0 0 0 0 ","LENCIO":0}
...
```

visualapproach, 2022-02-10:
OK, I made some more changes to avoid waking up the watch dog. Just in case it's the filehandling that takes too much time due to an error in the filsystem, I would like you to erase flash on your ESP. You can do it from platformio "project tasts" menu. Don't forget to build and upload filesystem again, after uploading the new code.

visualapproach, 2022-02-10:
You can also check your WiFi strength in this update. RSSI is transmitted to the web page console. I'm no expert on this but here is a guideline to what values are good/bad:
[https://www.metageek.com/training/resources/understanding-rssi/](url)

[41 more replies]

closed 2023-03-13 by visualapproach