visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #202: Strange behaviour with poor WiFi network

NitramH, 2022-01-24:
Hi,

my ESPD1 mini was running the first time and since two days without any problems.
Then something strange happened. The connection was lost, recovered again, lost again and so on.
The up-time periods were smaller and smaller and then it died finally.

The ESP ist connected to my network but not talking. No reaction regarding MQTT, no answer
if I try to open the Webpage. But the spa is working (display available).

Does the ESP has problems with noisy environments?
Does it try to recover automatically? (I turned off the Access Point mode)

Would be great if someone could help. I`m so short reaching to the target to commission it finally :-(

visualapproach, 2022-01-24:
Hi

I made an update in dev branch yesterday where I tried to make sure everything WiFi related restarts after a drop out. Tested OK here. The ESP tries to reconnect all the time to WiFi.
Not sure it helps in your case but maybe worth a try.

visualapproach, 2022-01-24:
I noticed a weird behavior the other day though. I couldn't reach the ESP from chrome on my computer after I fiddled with blacklisting the device. But I could connect from safari on my phone...
Restart of computer solved it.

NitramH, 2022-01-24:
I tested your dev branch but I do not have a chance to enable an stable accesspoint connection.
Our Wifi environment is heavily overloaded in the neighbourhood and I have to install a repeater in my garden.

I will answer later, when the repeater is working.

Other question, but important. When the ESP stays unconnected, is the spa still self protected against low temp.???
We have minus degrees in the next nights and I fear to destroy my Miami spa.

visualapproach, 2022-01-24:
There is no warranty on this device 😃 
But the 6 wire version will not change the factory protection. Possibly some crazy corner cases may interfere but the chance is small.
The 4-wire will keep the temp at I think 10 degrees C when ESP is in charge. If it crashes this may fail. When display is in charge and ESP crashes it is uncertain what happens. If the protection is located in the dsp it can fail. If in the CIO odds are better.

NitramH, 2022-01-25:
I have the 6-wire version so there is no hurry for me :-)
Anyhow, the big time constant of the pool helps me to brigde the 3 cold days, now it is at 11°C.

I implemented the dev branch and had some trouble:
- you need to change the legacy name in the homeassistant config from "BW_2.0.0" to "layzspa" for the MQTT messages
- additionally the buttons do not work anymore, was there any change regarding Miami Spa interface?

NitramH, 2022-01-25:
> Legacy name, good point Buttons, uh no change that I'm aware of. Display buttons I presume?
I didn't checked at the device. But homeassistant as well as web interface did not accept changes to ON state.

visualapproach, 2022-01-25:
updated the yaml files. Did you choose the right model before compiling?

NitramH, 2022-01-25:
Sorry to ask you again, is maybe here a wrong setting? I use the d1mini

[platformio]
src_dir = src
data_dir = data

[env:nodemcuv2]
platform = espressif8266@^3
;board = nodemcuv2
board = d1_mini
framework = arduino
lib_deps = 
	bblanchon/ArduinoJson
	mcxiaoke/ESPDateTime
	links2004/WebSockets@^2.3.3
	knolleary/PubSubClient@^2.8
	tzapu/WiFiManager@^0.16.0
board_build.filesystem = littlefs
monitor_speed = 115200
;Set upload speed to 115200 if you get transfer errors
upload_speed = 921600
; set frequency to 160MHz
board_build.f_cpu = 160000000L

visualapproach, 2022-01-25:
You shouldn't need 160 MHz.
What pump model do you have? NOxxxxxx
What model did you define in config.h?
What is the problem? Compiling, uploading or "buttons not working"? Please be as detailed as possible.

NitramH, 2022-01-26:
> You shouldn't need 160 MHz.
Please confirm that I can go down to 80 MHz, there was another topic with problems using the 6-wire Miami model -> https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/54#issuecomment-864589824
where the 160MHz was proposed.

Do I need to change the "[env:nodemcuv2]" to "[env:d1_mini]" as well or does this not matter?

The rest is solved now, caused by my Wifi network. Now I have an second access point for the garden. And I had a static IP problem, the Fitzbox (maybe you know this product from Germany) is causing sometimes trouble by mixing up the IPs in the network. I changed to a new IP and it works now. But this is an off-topic issue.

visualapproach, 2022-01-26:
If it works @160MHz you should try 80MHz. It's better not to overclock if it ain't necessary. I often suggest overclocking when people have problems with new pump models because I don't know if their communication protocol is faster. If it works @80 that's what you should stick to.

[env:] is just a label as far as I know. Does not matter.

NitramH, 2022-01-26:
Perfect, will check with 80 MHz

closed 2022-01-26 by NitramH