visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #204: MQTT Telemetry Interval

Scoobler, 2022-01-25:
Just got MQTT up and streaming data. It appears if nothing changes, current temperature or heater turning on and off etc, the MQTT data seems to update every 10 minutes, it would be good to be able to adjust the telemetry so uptime, kWh etc updates more frequently - most of my IoT is set to 60s updates.
Also, is there a definition of the MQTT time TTTT (guessing Time To Target Temp?).

Thanks in advance!

visualapproach, 2022-01-26:
Hi @Scoobler 
You're right, and you're able to adjust the update interval to whatever you want. It's the beauty of open source.
"main.cpp":
```
  // update MQTT every 10 minutes. (will also be updated on state changes)
  updateMqttTimer.attach(600, []{ sendMQTTFlag = true; });
```
Your guess is completely correct.

cheers

Scoobler, 2022-01-26:
@visualapproach thank-you for the pointer that's great to hear! I am just waiting for some boards to arrive from JLC PCB so have a breadboard installed and working on a St Moritz 2021 6-wire at the moment.

I may have a bash at adding the MQTT Telemetry Interval to a setting on the Web UI while I wait for the new board.

The TTTT value is sometimes negative, but sometimes positive - what is the difference?

visualapproach, 2022-01-26:
Negative TTTT meaning the two curves crossed in the past.
Your welcome to make a PR to the dev branch.

Scoobler, 2022-01-27:
#205 closes this issue, thanks!

closed 2022-01-27 by Scoobler