visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #208: E13 Error - 4-Wire - Hawaii Hydrojet Pro #54138

DeanozUK, 2022-01-30:
Hi, 

First things first @visualapproach, Well done on this excellent project, I've been looking for this type thing for ages. Looking through your code amazes me. the fact you are making this adaptable to different version/models is a hard thing. Pat on the back m8.

I've a Hawaii Hydrojet pro (2019) #54138 and am getting same issue as described in other threads but no answers found, tried 2 different ESP's, Converters, PCB and leads.

Background - Im a technical person with lots of electronics hobbies and been trying this for days. Additional Wifi AP added moved near to rule out, So not signal related and confirmed. will stay connected allday long with DSP disconnected. Also have 47uF cap on PCB 5v line, tried the 100uF too, but think this is for random brown outs some people report not this issue.

Plug in module, powers up DSP, 10sec later E13. Blue light is goes out but looks like its trying to flash (bootup)
Unplug DSP and blue light on (and flashes to indicate CIO data?). (botom of web page CIO Active - Dev branch) but no control (even with 'take control' on) looks good so far.
Plug in DSP and 10secs later E13 and light goes out, starts to flicker evey now and then.

Tried disconnecting 5v from CIO (dont want that magic smoke to escape) to PCB and powered PCB, ESP and DSP from powerbank. Get same results however the ESP is either rebooting as wifi keeps going down on ESP so web page might come up briefly then timesout, is ESP crashing or Interference causing disconnect? I have no idea now and am loosing my mind.

Im going to see if i can workout how to flash this test passthrough code to confirm hardware. Just not sure how atm.

Will send pics if needed.

visualapproach, 2022-01-30:
> Hi,
> 
> First things first @visualapproach, Well done on this excellent project, I've been looking for this type thing for ages. Looking through your code amazes me. the fact you are making this adaptable to different version/models is a hard thing. Pat on the back m8.

Thank you. Yes it has grown over time. Would be nice if it worked for you though.

> Also have 47uF cap on PCB 5v line, tried the 100uF too, but think this is for random brown outs some people report not this issue.

Yes, that's a proposed solution for brown outs, but you're having communication problems. With the DSP it seems. Does the webpage say "dead" on the DSPTX?

> Im going to see if i can workout how to flash this test passthrough code to confirm hardware. Just not sure how atm.

Just open the folder "test-hardware" in VSC and upload. It's a super simple program that just retransmits everything. No web gui or wifi. If you get E13 with this running you're most likely to have hardware issues.

You probably did, but just to be clear - have the DSP and CIO connected when powering up. Disconnecting the DSP may trigger error in the CIO and vice versa. (No powerbank connected now)

DeanozUK, 2022-01-30:
Thanks for your response,

> Does the webpage say "dead" on the DSPTX?
I can't get onto the web page when DSP is connected as the wifi connection keeps looping the minute i connect it.

> You probably did, but just to be clear - have the DSP and CIO connected when powering up. Disconnecting the DSP may trigger error in the CIO and vice versa. (No powerbank connected now)
You are correct. no powerbank and original wiring, power unit it E13's and wifi wont connect for longer than few secs. tried with power bank by powering both at same time and same results.

> Just open the folder "test-hardware" in VSC and upload.
I've opened VSC and opened just the folder as you said 'test-hardware' inside 4-wire folder but can't see a upload icon at the bottom. is there a guide to follow to do this?

just a note when the wifi is up for few secs i managed to take control and start and stop pump before wifi timed out. think this proves CIO is communicating ok?

visualapproach, 2022-01-30:
Oh, you have to open it from platformio (the ant-head to the left)

DeanozUK, 2022-01-30:
Ah, looks like its inside another folder, see below. sorry for the noob question but I was a PIC MC kinda guy ;)

\test-hardware\4-wire-serial-bypass

DeanozUK, 2022-02-03:
Hi @visualapproach 

Sorry for the delay, I've been doing lots of testing trying to work this out. I did the hardware test as you suggested with mixed results. first test lasted 30secs then E13, after fiddling with connectors i got it to last several hours with just pump running all good then E13. I thought this is very odd. Thinking it was the connections I tried several tests using pins, different connectors etc and still mixed results. I even tried different type ESP still same. 

Finally think i may now be onto something. I originally thought interference of some kind. So I shortened the fly leads (See pics) and soldered really close to board. and bingo, it works, albeit until I turn bubbles on or heater (may be pushing noise over the edge). But it worked far better than before. still need to do more testing. I'm thinking I may have to re-cable from CIO to DSP with maybe Beldon cable which is screened and don't make it too long.

Whats your thoughts on these findings? I doubt there is anything in coding that can help here? Have you had similar results before? Have you done a live test with water yet and not just bench test? 

Thanks
Deano

visualapproach, 2022-02-07:
Hm, I thought I answered this but apparently not... Sorry. I have personally only run it in a dry condition. I can't recommend making changes to the existing cables. It will void warranty, and what if it still doesn't work. But it's your stuff and I wont stop you of course. But it seems that the connections are shaky. I can only repeat what other people has reported - that the GND may be poor. And maybe connecting 100 Ohms resistors in series on the serial wires will help. What happens when you start bubbles or heater? Does  the ESP reset? You can check last reboot reason on the config page. If it says brown out it is probably the power. But I see you have a cap already...

DeanozUK, 2022-12-14:
Hi all,
I just wanted to post good news and an update on this problem which I put on hold for awhile hopping it will help others out.

Now with the massively increased energy costs I needed to look back into it so I can control heating during Economy 7 tariff. I also wanted to look around the other new issues etc to see if there has been any changed that may helped.

I upgraded my PCB to the v2, installed filters and new release flash from scratch and found the same issue of the dreaded E13. damn. so I looked harder at all the comments and noticed several times (cant remember the issue numbers sorry) that there was a suggestions to not use the Blue IC LLC but use the Red transistor base one. So I did and hey presto it worked

So below is what my setup is as per build instructions. (This is for the 4-wire 54138)
- New PCBv2 with headers
- 100mm fly leads as before (not cut down)
- installed all 4 filter caps and resistors (just in case)
- new RED LLC (removed the 10k resistor for the blue LLC which i was removing)
- Flashed the latest Dev code as of today (release build was coming up with compile error in code, cant remember now but dev fixed this)
- before flash pre-programmed the OTA and AP / IP details

Been working for 2 days now. I've had 2x E01 errors but believe this may be related to me playing with MQTT/HA or the cold temps -1C at the mo. I will keep an eye on this. 

@visualapproach If you are getting more info on using the Red LLC then maybe good idea to put in build doc as a suggestion to try first especially if using 4-wire or getting E13. 

Again, Thanks for your work on this and continued updating/support. Anything I can do to help let me know. Donation coming your way.

visualapproach, 2022-12-17:
Lovely @DeanozUK ! Thank you for sharing this info. I'll update the docs. And also the master branch. Wasn't aware it was failing at compilation. Thanks again!

closed 2023-03-13 by visualapproach