visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #209: 6 wire Palm Springs

Chamberlainpac, 2022-02-04:
Thanks for this amazing site . I’m a complete noob but still  managed to build my first ever wifi unit thanks to your amazing installation instructions. And learnt how to code and set up the AP. My issue is I have a 6 wire palm spring unit. I’ve plug it in and no response - won’t even power on. I suspect it’s the wiring.  How do I figure out which of the 6 wires is which ?  Which one is 5+ , ground ? Etc ?  Any Help much appreciate. 

[image]
[image]
[image]
[image]

visualapproach, 2022-02-04:
Thank you @Chamberlainpac  for your kind words!
What's your model number? It's on the sticker on the egg.

Chamberlainpac, 2022-02-05:
Hi vis. It’s the #54129 
[image]

Chamberlainpac, 2022-02-05:
[image]

Chamberlainpac, 2022-02-05:
[image]

Chamberlainpac, 2022-02-05:
[image]

visualapproach, 2022-02-05:
Great, look's alright to me. To check which pin is which, you need a multimeter but you can only see +5V and GND with those. The other pins will switch rapidly between 0 and 5 V and will show something in between on a dmm.
But if the blue LED turns on or blinks you at least have the power pins right. Do you get to the esp WiFi web page?

Chamberlainpac, 2022-02-05:
No power . No web page. I checked the volts. It’s wired backwards . The awg is ground .. the - - - - is 5v .  I’ll re solder the connections and give it a try.

Chamberlainpac, 2022-02-05:
Ok we now have Power on the esp. we have blinking blue light and I can see the esp webpage . But no display on the egg. I can hear the display clicking ?

visualapproach, 2022-02-05:
did you also rewire the cable to display?

visualapproach, 2022-02-05:
If you rewired the ESP to CIO by rotating it 180 degrees. I mean from 123456 to 654321 you should do the same on the display cable. Otherwise you feed 5V into the speaker which I guess is what you hear.

Chamberlainpac, 2022-02-06:
Ok… I’ve reversed wires on output side too . Have flashing blue light . Melody plays and I can see the esp page . Display on the egg is still off and unresponsive.  
[image].

visualapproach, 2022-02-06:
Okay, maybe you have one of those pumps that needs resistors on the data outputs. On the DSP Clk, data and cs lines try putting a 500-ish Ohms resistor in series. 
LLC-R1----wire to dsp
LLC-R2----wire to dsp
LLC-R3----wire to dsp

Chamberlainpac, 2022-02-06:
Ok.. I’ll order some and let you know.

Chamberlainpac, 2022-02-13:
Hi vis, unfortunately that didn’t work.   Still nothing on the display. 
[image]

visualapproach, 2022-02-13:
Hi, are you able to control the pump from web gui? Like turning on the bubbles.

Chamberlainpac, 2022-02-13:
Yes, the blue light flashes, music is played.  The gui is fully working . Correct temp, bubbles work. Just no display at all ? 
[image]
[image]

Chamberlainpac, 2022-02-20:
I changed the LLC plus added the 10k resistor just to be surely  Still no display and now I’m not seeing the temp on the web page or able to control the pump.    Still got blue light and music and webpage opens.   Before changing the llc I checked all solders and wires and all ok.

visualapproach, 2022-02-22:
I'm busy a while but will call you back. Isn't it strange that the webpage temp stopped working just because you switched LLC?

Chamberlainpac, 2022-02-22:
It wasn’t easy to unsolder and take out  the llc so may have damaged or dislodge something else in the process?  I will check continuity to be sure. I’m happy to test it using the multimeter to let you know what it’s doing when live ?  

Regards 
Paul

Chamberlainpac, 2022-03-13:
I decided to build a new one.. and it’s now working perfectly. Thanks again for all your help , code and time.  I will now close this thread.

closed 2022-03-13 by Chamberlainpac