visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #215: Setpoint for temperature running away ->40°C

NitramH, 2022-03-03:
Hi,

I had a strange reaction of my device, after selecting the new setpoint at the web interface (e.g. from 20°C to 32°C)
the ESP was moving it towards infinity and only limited by the max. allowed spa temp of 40°C.
I repeated it several times until it accepted the correct value.

For manual access this is not a matter, but for home automation I need to improve this.
Workaround could be to control the temperature by home assistant and to work always with 40°C setpoint in the spa
to disable the spa control, but I do not like this approach.

How ist the control done? Is it a simulation of pressing the "up" button until the target is reached?
Where can I find it in the code?

Martin

visualapproach, 2022-03-03:
Dev or master branch? Dev has new code for that.

NitramH, 2022-03-03:
It was Dev but older version, downloaded 23.01.2022

visualapproach, 2022-03-08:
> How ist the control done? Is it a simulation of pressing the "up" button until the target is reached?
Previously yes. New version simulates a number of presses.

closed 2022-03-09 by visualapproach