visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #232: Problem Temp. Button v3.1.1

Laudings, 2022-03-12:
Hello,

first of all thank you for your work and the new update. It hits the target temp always by the first time. This is great!
Also the startup with the showing IP Adress is now correct.
But i have one trange thing. If I change the temp by pressing the button on the pump, in the web interface the actual temp is changing and blinking and not the value of the target temp. 
But only when i change the temp with the button. If i change it with the web interface or MQTT everthing is how it should be. 
 
Can somebody confirm this? 

Thank you in advance!

visualapproach, 2022-03-12:
I can confirm that it happens occasionally when you fiddle a lot with those buttons. I looked at it before but nothing obvious stood out and since I never use those buttons I decided it was too much work to fix it. At least for the moment.

Laudings, 2022-03-13:
Yes i can understand. I tried to check the BWC_8266.cpp file but i am definitely not a js expert. 
I tried the older verison 3.1.0 there i can't determine this behavior. Maybe this can help.