visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #235: MQTT command doesn't work

e-i-k-e, 2022-03-23:
Hi there,
Unfortunately I can't manage to send MQTT commands to the .. or the device doesn't receive anything.

MQTT runs in my ioBroker (without access data) I also receive data.
[image]

However, the orders do not arrive or go out.
What can i do?

Command: {"CMD":"4","VALUE":"true","XTIME":"0","INTERVAL":"0" or {"CMD":"4","VALUE":"1","XTIME":"0","INTERVAL":"0"

[image]
[image]
[image]
[image]

visualapproach, 2022-03-23:
Do you miss the closing curly bracket?
{"CMD":"4","VALUE":"true","XTIME":"0","INTERVAL":"0"
should be
{"CMD":"4","VALUE":"true","XTIME":"0","INTERVAL":"0"}

e-i-k-e, 2022-03-23:
> Do you miss the closing curly bracket? {"CMD":"4","VALUE":"true","XTIME":"0","INTERVAL":"0" should be {"CMD":"4","VALUE":"true","XTIME":"0","INTERVAL":"0"}

Excuse me. This was now a copy error. The bracket is of course otherwise present.

visualapproach, 2022-03-24:
try without citationmarks around the values:
{"CMD":12,"VALUE":3,"XTIME":1648149567,"INTERVAL":0}

e-i-k-e, 2022-03-24:
> try without citationmarks around the values: {"CMD":12,"VALUE":3,"XTIME":1648149567,"INTERVAL":0}

Unfortunately it doesn't work either. :/

visualapproach, 2022-03-25:
In your first picture the connect count says 105. It should ideally only read 1. Check serial output from the ESP in a monitor and you'll probably see it has to reconnect all the time. The MQTT code is working, so it must be something in your setup.
What's the RSSI?

e-i-k-e, 2022-04-07:
I deleted the data points in ioBroker, installed a WiFi repeater nearby and reestablished the connection.
Now it works. Thanks!

closed 2022-04-07 by e-i-k-e