visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #236: Obviously wrong temperature readings

Skeletorjus, 2022-03-24:
This might be more of a feature request, but I think there should be some filtering/discarding of temperature readings that are way off. In my case I sometimes get readings of -8 and -9 degrees, I'm unsure what exactly triggers these values.
[image]

Since the real temperature rarely, if ever, changes more than 1 degree at a time, I think a safe way to do it would be to discard values that deviates by more than 2 compared to the previous value.

visualapproach, 2022-03-24:
@Skeletorjus  that can ofc be done, but it would only hide the symptoms instead of finding the underlying problem. What pump model are you using, and what software model is defined?

Skeletorjus, 2022-03-24:
Pump model S100101. It's a Miami 2021, so that's what I have defined. Running latest dev version.

No resistors connected to the board. 

I haven't verified if the same error is present in the web interface, only via MQTT.

I actually right now made an interesting discovery. At the exact time the readings are negative, I get "phantom" statuses as well. C/F, bubbles, and lock all toggle on/off within the same second. Nothing happens physically with the spa itself.

visualapproach, 2022-03-25:
Okay, I'm pretty sure that's caused by unstable signals in the wires. The 2021 models use a different driver chip as I understand it. (Don't have one myself) That's the reason it needs resistors in some cases. Actually I think the specs calls for a low pass filter consisting of a series resistor (~500 Ohms) and a small capacitor (a few pF???) to GND on each com line. Most people only puts resistors on the data/cs/clk wires to the display to avoid flickering. If they put it on the pump side it stops working strangely enough. I'd say you would want the filters on the pump side, but you'll have to test for yourself I guess. Maybe it helps just changing the length of the wires.

Skeletorjus, 2022-03-25:
Thank you for the reply.

This is interesting.. My wires _are_ long (~15cm), and according to https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/discussions/188#discussioncomment-2176634, what I'm experiencing could very well be symptoms of imperfect signals.
Since we're entering a territory with electronics where I'm not very knowledgeable, I think my first step is to go the easy route and assemble another board with shorter wires and try swapping it in the next time we're changing water or doing some other maintenance to the spa.

I haven't seen any real world consequences yet, both the ESP and display are fully functional, so unless there's something I haven't considered, I guess it's OK for now until I try the other board. Just have to remember that all Home Assistant automations that triggers on states from the spa should always have a `for`-condition with a time of 3+ seconds 😄

visualapproach, 2022-03-25:
Since so many bits are simultaneously wrong, I suspect that the clock signal is noisy. Try latest DEV version (firmware version 2022-03-25) and look at the MINCLK values shown below. It should say around 18 us. Check several reports.
[image]

Skeletorjus, 2022-03-25:
Thanks. I took ~25 samples, and they were all between 5 and 8 us.

visualapproach, 2022-03-25:
Ah so that means either the baudrate is significantly higher or there is unwanted level changes on the clock. I bet it's the latter.

Skeletorjus, 2022-07-18:
Except from the zero-values during reboot, which we've discussed elsewhere, my readings have been rock solid for the last few days with the 2022-07-15 Dev version. No hardware changes.

visualapproach, 2022-07-18:
Todays version has an MQTT startup delay and should not send zeroes