visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #240: Uploaded program doesen't work.

70hnKuky, 2022-03-26:
After I sucessfuly upload to nodemcu, nothing happens.

serial monitor shows:
{l␀d���␂�c␃�n␌␄␌�␌l�␌c|��␂�p␃�{��␟�␄b��no�loN���␌c␜p��#$`␂rlp�n�␐␂␌␄�␌$␌��␌␄␌c␄n�|␂�␌���␄b��no�␂l��l`␂�␒␓oN␌�{l␏␂N{���o␄␌rr� ␂p�o�␐␂␄␌r�ܜ�␜��␃␌B␌o�|␃␄��␌B�|␡�n�␂␄�␀d`␃�␒␛oN␌�␏l␏␂n{���o␄␌r␓l ␂␎r��o␄␌r␓l ␂�␂␄␌쏏�d�
l ␂��o�␎$

if I change baudrate for serial monitor to 74880 it returns something more readable:
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000748e0
~ld

I have tryed with 3 diferent nodemcu-s but no diference. i have tried with both vscode and atom.

Can somebody help me.
tnx

visualapproach, 2022-03-26:
Hi, did you follow the build instructions? You need platformio extension and the serial baudrate is set automatically to 115200.

70hnKuky, 2022-03-26:
Thank you for quick response.
Yes, I have followed build instructions and I have platformio installed. only difference in my case is that I use nodemcu V2 (still ESP8266)

70hnKuky, 2022-03-26:
let me add, that I am trying 6-pin version and I also get warning when trying to upload:

Compiling .pio\build\nodemcuv2\FrameworkArduino\Esp-version.cpp.o
.pio\libdeps\nodemcuv2\PubSubClient\src\PubSubClient.cpp: In member function 'boolean PubSubClient::publish_P(const char*, const uint8_t*, unsigned int, boolean)':
.pio\libdeps\nodemcuv2\PubSubClient\src\PubSubClient.cpp:523:16: warning: comparison of integer expressions of different signedness: 'unsigned int' and 'int' [-Wsign-compare]
  523 |     return (rc == expectedLength);
      |             ~~~^~~~~~~~~~~~~~~~~
Archiving .pio\build\nodemcuv2\libf4c\libESP8266WiFi.a

visualapproach, 2022-03-26:
That's just a warning in the mqtt library. Don't worry about it.

h2owasser, 2022-04-01:
Hi, I just returned to the project...tried to install the newest version. (master branch)
I got similar Issues as stated above.

(I got the Web-Interface working last year but not the controls and i wanted to try again)

Followed the Build instructions completely.

Flashing works fine, but the access point is not showing up. (Also the LED is only flashing during the flash-process)
With the older versions it works just fine. I also erased the flash before trying.

Also set bool enableAp = true; in config.h for testing, but no access point. (I mean the new one with lay-z spa module as name)

---
After some erasing and flashing again i got the 4 wire version to setup an access point.
Also reinstalled visual studio code just to be safe and having all updates...

With the 6-Pin version still no chance even after waiting for ~5 minutes after flashing.

I also went to 3.1.1 and it works as intended (with the Web-Interface)

visualapproach, 2022-04-03:
ESP will only launch an AP if it cannot connect to WiFi. To know what is happening you should run the ESP "on the bench" with a USB cable. Look what it says in the terminal window.
The code works, I tried it very recently. Try "clean build" before compiling.

h2owasser, 2022-04-03:
I did a 2 Minute video of the upload process. (I did pause the recording while on longer sections of compiling or uploading)
...In the end i press the reset button on the nodemcu a few times.
No terminal output except the reset happening as show.
-> Build Cleaned (Erased Flash beforehand)
-> Uploaded/Compiled the firmware (no Changes. Freshly downloaded)
-> HW Reset ESP -> Build Filesystem
-> Uploaded Filesystem
-> HW Reset ESP 
-> Opened Terminal and Changed Baudrate to 74480
-> HW Reset ESP (X-Times)
-> ESP seems dead. No Life-Signs.   (Tested on 2 different ones one of which i built with 3.1.1 and am running now)
 
https://user-images.githubusercontent.com/1337441/161446155-2f082e64-0f9c-41fe-81bf-3175a2e38102.mp4

I happily help with any debugging i can do.

---
Offtopic:
With the 3.1.1 flashing and filesystem it worked flawlessly on the spot and it's running by now (Problem of last year was a fried Voltage converter ...and for the flickering Display i used some 470 Ohm resistors which worked flawlessly on dry testing)

visualapproach, 2022-04-03:
> I did a 2 Minute video of the upload process. (I did pause the recording while on longer sections of compiling or uploading) ...In the end i press the reset button on the nodemcu a few times. No terminal output except the reset happening as show. -> Build Cleaned (Erased Flash beforehand) -> Uploaded/Compiled the firmware (no Changes. Freshly downloaded) -> HW Reset ESP -> Build Filesystem -> Uploaded Filesystem -> HW Reset ESP -> Opened Terminal and Changed Baudrate to 74480 -> HW Reset ESP (X-Times) -> ESP seems dead. No Life-Signs. (Tested on 2 different ones one of which i built with 3.1.1 and am running now)
> 
>  2022-04-03_21-47-41.mp4 
> I happily help with any debugging i can do.
> 
> Offtopic: With the 3.1.1 flashing and filesystem it worked flawlessly on the spot and it's running by now (Problem of last year was a fried Voltage converter ...and for the flickering Display i used some 470 Ohm resistors which worked flawlessly on dry testing)

Video is not playable here. A few comments though:
- "clean build" and "erased flash" is not the same thing, but maybe you knew that. "Clean build" is clearing the "cache" on the PC so you don't use old files.
- You should not need HW resets after every step, except possibly before you want to upload over the air one first time. But it's difficult not to do a HW reset, when unplugging the USB.
- Why do you change baudrate? The code sets it to 115200 and platformio.ini is setting the terminal speed to that. Just hit "upload and monitor" and you're done and should see the serial output.
-

h2owasser, 2022-04-04:
Oh that with the video is sad... tried to record it 5 times
I uploaded it again on googledrive:
https://drive.google.com/file/d/196pjzpT0FvJjf5UezxyjlVchiqNlOdKe/view?usp=sharing

-For Point 1, yes I am aware of the difference. I just wanted to point out what i did exactly.

-Basically i use my "good" nodemcu in the pump already. The one i use right now is soldered to 5v, rx, tx and gnd to a programmer (which i bought for flashing tasmota to esp8266mini) ...a bit complicated but it works. I tested it with the USB connected one beforehand and had the same behaviour.

-With standard baudrate i get the same output as @70hnKuky. With it set to 74480 i get a readable output. Same as in OP.

I'm quite new to microcontrollers but wouldn't it mean if i only get something from the terminal at 74480 the clockspeed is set to 74480 ...which would be against the code as it is set to 115200?

If i switch to 3.1.1 it works in an instant, so i do not think it is a hardware fault

Really strange.

visualapproach, 2022-04-06:
Okay, Something is broken! It worked. Now it don't. Only thing I can recall is an update to VSCode. I need to investigate this and hope to have a fix in a couple of days.

visualapproach, 2022-04-06:
It wasn't a bug per se, I had added some code to initialize some variables, but it was blocking the program when not connected to the pump. I removed it again which will make it work without the pump connected, but it may send wrong sensor units to home assistant during start-up. I'll fix that in a better way at a later date. Sorry for confusing you.

closed 2023-03-13 by visualapproach