visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #244: Compiling/Running on a Wemos D1 mini

creativtransfer, 2022-04-08:
For whom it may concern:
I am not able to make the last release v3.2.0 (configured miami2021) work on a Wemos D1 mini. 
It compiles and uploads but there is no AP.
The previous release v3.1.1 works just fine.

visualapproach, 2022-04-08:
Dev or master branch? Try dev if you haven't

creativtransfer, 2022-04-08:
I used the (master) release from the right sidebar. 
I just downloaded and compiled the development Branch and it works fine.
Thank you!
PS: Can you add the Version information to the homepage - maybe at the bottom below or above the mqtt status?

visualapproach, 2022-04-08:
Good. Actually the master is also working but it needs to be connected to the pump.
Ps I hear you

creativtransfer, 2022-04-08:
Ok, thank you!

closed 2022-04-08 by creativtransfer