visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #251: Display not working

eried, 2022-04-15:
I assembled it for my S100101 pump, and the display does not react (it gets power and shows 22 but nothing else). Is it something stupid I am not doing?

[image]

The wifi part works, shows 22C, but for if I switch from C to F for example the slider goes back to C after few secs.

visualapproach, 2022-04-15:
It should work:
https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/92#issue-942265708

Have you edited the file "model.h" to tell the ESP what model you have?

If so, it seems you have a hardware issue. Bad soldering or needing resistors, but first things first.

visualapproach, 2022-04-15:
Very happy to see you used female header pins. 👍

eried, 2022-04-15:
Thanks! I will continue experimenting. I tried another level converter board, same thing. Another esp, same. So I will test another pcb now

eried, 2022-04-15:
So, I tried with everything different, and still no go (new JST, esp, etc). What I noticed is this network activity:

[image]

Pressed buttons sometimes say feff instead of ffff so I imagine there is some noise. I am not sure what else to try

visualapproach, 2022-04-15:
You didn't answer my question though

eried, 2022-04-15:
Oh sorry, yes I tried all versions in that file 🤓

visualapproach, 2022-04-16:
Okay, the button code seem wrong. No version has that as no btn. May be a bad connection. Can you control the pump, like turning on airbubbles from the web UI?

eried, 2022-04-16:
No, I cant control anything. You think I need to add pull down resistors? first test I will do is just to write a small pass-thru code to test the esp and connections. I have a digital analyzer but that will take more effort.

eried, 2022-04-16:
More tests:

1) added 470ohm resistor in serie to the 3 data lines: nothing changed (tried CIO and DSP)
2) if I connect the screen directly and then plug the board, the screen gets powered with the last number the CIO sent, so... 5V and GND are ok (also BEEPER because I hear the music at boot)

I was thinking on the pass-thru code, but it is not as simple... so I am still thinking what could be wrong. I made a second board with everything new, but it behaves the same

visualapproach, 2022-04-16:
Pass through is harder on the 6 wire than the 4 wire because it's bidirectional data on one wire. The display has a memory and shows the last state so you can't tell if it's working by looking at the temp. If it works it will scroll through the ip number and fw_version at startup.
It appears neither the cio nor the dsp connection is working for you. Do you have any equipment to verify the data wires are in the same position as expected?

eried, 2022-04-16:
this are the wires:
5V GND D0 D1 D2 D3

when is idle:
[image]

I suspect that is the temp, which is properly read.

Now if I press TEMP UP for example:
[image]

if I turn the pump/heater:
[image]

the captures:
[temp up from 38 to 39 display says 38.csv](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/8500761/temp.up.from.38.to.39.display.says.38.csv)
[press pump and heater on.csv](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/8500762/press.pump.and.heater.on.csv)
[idle.csv](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/8500763/idle.csv)

I do not know much what else could help debugging

visualapproach, 2022-04-17:
Fantastic! I love people with logic analyzers ;)
Lets first find out if the wires are connected to the correct pins!
Assuming you didn't alter the pinout in the code this is how it shall be connected (CIO port H1). 
Pin1 = 5V
Pin2 = GND
Pin3 = DATA = channel 1 (in your picture) = D7
Pin4 = CLK = channel 0 = D2
Pin5 = CS = channel 2 = D1
Pin6 = Audio = Not used in H1, just in H2

If you need to rearrange the wires, do it on both headers/cables, accordingly. It's possible to edit the code instead, but then you will need to do that every time you update the firmware.

eried, 2022-04-17:
Nice! I think is better if I rearrange the wires than the code. I'll try 🤓

eried, 2022-04-17:
I flipped wires on pin3 and pin4, but now it does not even receive the temperature as before (the temperature and C/F is received correctly). I will continue researching the pinout :/

visualapproach, 2022-04-17:
I can't help you because I don't know what pin you connected to which channel. Maybe if you post another pic where you mention what the channels are connected to. And by the way, can you up the sample rate to ~1 Msamples/sec and it may be possible to see more what's happening. Settings in logic analyzer should be SPI protocol, Least significant bit first, 8 bits, CPOL = 1, CPHA = 1, enable line active low. (For my pump.)

eried, 2022-04-17:
I will try, which format is the best for sharing the captures? Do I capture something particular? Keypresses? What I will do meanwhile is to change pins in the config until I see something interesting. My pump is outside and it is very cold outside 🤣🧊

visualapproach, 2022-04-17:
I think screenshots like this is best for now.
[image]

The high frequency line (yellow) is the clock. The one switching the fewest times is the Chip Select (red). You should be able to check your connections with that info ;-)
Temperature here is quite nice today. And sun is shining.

eried, 2022-04-17:
Hi again, If I dont press anything this is how it looks:

[image]

I have the analyzer connected like this:

[image]

from left is 5V GND D0 D1 D2 D3

and my board:
[image]

1) 5V and GND I think are ok, I get 5V with the tester and if I unplug the display and plug the esp8622 board the screen keeps the last contents

2) I get the temperature OK with 

```
      // int cio_cs_pin     = D1, 
      // int cio_data_pin   = D7, 
      // int cio_clk_pin     = D2, 
```

but no screen and the pump does not really do anything else

visualapproach, 2022-04-17:
If I interpret your pics right you have
Ch0 = green wire. It's the clock signal. It's attached to cio D7 pin.
So cio_clk_pin should be D7.

Ch1 = red wire. It's the data signal. It's attached to cio D2 pin.
So cio_data_pin should be D2.

Ch2 = chip select (enable) so
cio_cs_pin should be D1

As for the display port, the order should be the same. Green = clock. Dsp_clk_pin = D5
Red = data. dsp_data_pin = D4
And dsp_cs_pin = D3

Quite easy to mess things up here with all labels...

It's unclear what you mean by "temperature is working". The display is deceptive so only check web ui. And try airbubbles on off as a test for good connection between esp and the cio. The display is more apparent when it works by scrolling text etc.

I guess you should use the miami2021 model.

eried, 2022-04-17:
just to clarify: by "temperature is working" I mean that web ui gets the celsius and temperature from the pool, the screen is always dead (off) but if I plug the screen to CIO, and then quickly connect it to the esp8622 board, then the screen stays with the last temp.

The web ui actually updates the temperature (I have seen it changing down while testing). If I press any button on the web UI, nothing happens, the UI just goes back to the previous state smoothly (i.e. the slider goes back to original position)

eried, 2022-04-17:
Checking a bit what is actually sending, this is without any keypress

[image]

in between there is this, constantly (just a keepalive?):
[image]

and this is when I press the HEATER:

[image]

I wonder if bestway changed something again. This model is a TAHITI, the pump has a date of 2021-04-12 in a handle and the model is S100101

visualapproach, 2022-04-17:
Signals are consistent with MIAMI2021 (misleading name since that's just describing the lining. The S1001xx is more trustworthy but I didn't know that at the time of coding)
The repeating three bytes is the cio asking the display which button is pressed. 0x1b1b means no button. 0x1012 means heater button is pressed.
The longer messages is the cio telling the dsp which led segments to turn on and at what brightness. From that we deduct all the states.

What do you say about your wiring with regard to my interpretation earlier?

eried, 2022-04-17:
Your interpretation is accurate, if I use:

`bwc.begin(D1, D2, D7, D3, D5, D4, D6);`

the ESP receives the current temp and unit from SIO (web interface), but nothing else happens. As on some old issues I hear some "crackling sounds" in the screen, but nothing else works. I captured some more precise actions in case you can take a look. I am thinking I need to make a simpler sketch for debugging what is happening. Do you have some scripts you used before this project took shape?

[some captures TAHITI 2021.zip](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/files/8501589/some.captures.TAHITI.2021.zip)

visualapproach, 2022-04-17:
Ok, I developed with an arduino + esp01 before going 8266 but I don't think it's necessary to go simpler. There's some debugging features we should explore instead. And if you want to make sure your device is compatible softwarewise you could write down which codes corresponds to each button on the display. Like (0x42) 0x1b 0x1b
Do this without the esp.
I guess it will match the miami2021. The problem is probably different drivers on the hardware which may require some mods. If you can tell what chips are used we might get the answer instead of guessing.

To the debugging:
Look at the console from the web browser. What does it say about 'pressed button'? This is what the display sends to the esp and if it's crazy it will probably screw things up

eried, 2022-04-17:
> And if you want to make sure your device is compatible softwarewise you could write down which codes corresponds to each button on the display. Like (0x42) 0x1b 0x1b

and the longer sequence is to write back to the screen? I will take note of all the buttons, I guess holding them is a different code? (or maybe this is interpreted later). The screen does not even turn on, so I am thinking some init is missing

visualapproach, 2022-04-17:
As soon as you touch a button that value will change. As you release it will turn back to 1B1B.
You said you'd tried resistors to 'both cio and dsp'. Did you try to only have them between esp and dsp? Not on the power pins ofc.

eried, 2022-04-17:
yes, no power pins, only esp and dsp, in series. I made a small extension plug to test against the CIO too, the beep sound changed tho, which is strange because I only added a resistor to the 3 lines after 5V and GND

eried, 2022-04-17:
This is the init sequence, before the first beep, after the pump is powered:
[image]

then there is 4 second delay.

With nothing pressed, and off, is what you said 1B 1B

[image]

the other buttons:

- 0x42 0x02 0x00 temp up
- 0x42 0x13 0x12 temp down
- 0x42 0x01 0x00 lock
- 0x42 0x03 0x00 useless timer
- 0x42 0x10 0x12 heater
- 0x42 0x12 0x12 bubbles
- 0x42 0x11 0x12 pump
- 0x42 0x08 0x09 c/f unit

visualapproach, 2022-04-17:
Yeah the sound wire probably needs a pull-up/down or a bias. And maybe a dc-blocking capacitor. Something something. 

IIRC you had pressed button FFFF (on the phone now so not sure) which would indicate a weak pull down from the dsp. Maybe higher resistance needed on those data lines? Like 1K?
Maybe check the signals between dsp and esp with your analyzer would be good.

visualapproach, 2022-04-17:
> This is the init sequence, before the first beep, after the pump is powered:
>
> [image]
>
>
>
> then there is 4 second delay.
>
>
>
> With nothing pressed, and off, is what you said 1B 1B
>
>
>
> [image]
>
>
>
> the [...]

Yup it's the Miami2021 model you shall have defined!
Now, can you confirm that the browser says 'pressed button Ffff' or 1B1B? Rightclick in the browser and choose inspect. Click console.
I think you know that but just in case.

eried, 2022-04-17:
The browser just gives random values all the time for button presses. I tried putting the logic analyzer between the esp and dsp but strangely the esp does not boot, I tried with 2 units... very strange. 

I tested also with a 15K pulldown on all the lines except PWR and GND, but still nothing.

visualapproach, 2022-04-18:
> The browser just gives random values all the time for button presses. I tried putting the logic analyzer between the esp and dsp but strangely the esp does not boot, I tried with 2 units... very strange.
> 
> I tested also with a 15K pulldown on all the lines except PWR and GND, but still nothing.

Pins D3 and D4: Boot fails if pulled down (during boot)

eried, 2022-04-18:
I see. Still no joy. I see that the noise from the buzzer correspond to D4 blinking wifi status. I am not sure what else to try for the moment, I played with using INPUT_PULLUP but that only changes that the pressedButton is always 0000

visualapproach, 2022-04-18:
D4 has nothing to do with Wi-Fi. It just happens to be the built in led on one of the data pins to the dsp. Why don't you try the low pass filters sketched out on paper in another thread? It may work the same with your LLC.

eried, 2022-04-18:
[image]
 
these? I will give them a try tomorrow then :) I need to get some caps

visualapproach, 2022-04-18:
Yeah!

eried, 2022-04-18:
I noticed something else, if I touch the 5v from the CIO, I can feel somewhat a "tingling" in my finger. Im assuming the transformer to 5v is quite noisy too. If I remove VIN and use an external power, that noise sound from the display related with D4 disappears.

So maybe I also need a big cap in VIN. I tried writing a dumb pass thru, only sending stuff from the cio to display, without any success tho.

eried, 2022-04-19:
[image]

ok, so this should eat any noise over 300 kHz. 

considering that the clock switches maximum 80 times each ms I guess a more addecuate filter would be for 100 kHz or even a bit lower (1.5nF cap instead of 0.5), but I am just following known workarounds for now. 

[image]

I also added a 1000 uF cap to GND/5V in case there is also some noise there. I will test in the night, I hope it works!

visualapproach, 2022-04-19:
Holding thumbs

visualapproach, 2022-04-19:
Boring! Someone had better luck with really short cables. Maybe worth a try? (Don't cut the original wires in the pump)

eried, 2022-04-19:
I will try too. Nothing to lose now 😅 I was thinking on building the board with those 4x level shifter made of individual transistors, they say "i2c" in the description.

Just a small clarification, when the CS line is up, the CIO is talking right?

eried, 2022-04-19:
I think I discovered the reason why it does not work:
[image]

I will try to order another level shifter... And we will see next month depending on AliExpress

eried, 2022-04-28:
OK so I changed everything to this:

[image]

And it works!! :D note that I actually had TXS before, so there is something wrong with those 8 ch level shifter. The beep is also stronger with this ones. The only thing I noticed is that from time to time the esp reboots (I still have not received my pcbs from china tho)

visualapproach, 2022-04-28:
Congrats! Good to know. So what does it say is the reboot reason? You can check it on the bottom of the SPA CONFIG page.
And to answer your question - all lines are idling high (5V), active low

eried, 2022-04-28:
It says: "Last boot: 28/04/2022, 19:21:39 Hardware Watchdog"

Anyhow, this is so awesome 😎 thanks for the help and the firmware. I'll try to port it slowly to esp32, but im very happy that it finally works.

visualapproach, 2022-04-28:
You're welcome! Always glad to here about success. Check out this thread: #150 about websockets. It's on the 4 wire system, but may apply for you as well.

eried, 2022-05-03:
Just to close this issue, my model is a TAHITI, the pump has a date of 2021-04-12 in a handle and the model is S100101. The board with the 8CH level shifter didn't work, not even with resistors, short cables, pull downs or pull up. Tried different boards with all different parts. The option with 2x 4CH level (the simpler looking ones with GND and VIN at the middle pins) works perfectly, but it rebooted from time to time, which now stopped after adding a big 1000uF 16V cap in 5V GND.

closed 2022-05-03 by eried

Brynod, 2022-05-04:
Hi Guys,
I understand this issue is now closed, however, i wanted to chip in. Having built this module with the 8ch Level Shifter, i couldn't get it to work. I put my analyzer on and saw that the Data line was only giving little "Wisps" (unfortunately i didnt take a screen grab) and didnt seem to be switching the logic correctly. I stumbled upon this thread and saw @eried comments regarding the 4ch level shifters so i swapped out the 8ch for 2x 4ch Level Shifters. The result (on the bench, connected to the DISP out) is that the logic signals are now clean and look like they should. It is late here in the UK but i shall be testing tomorrow and will update with the result. Pretty sure it's gonna work now....

Hope this can help others with the same issue!

PS EXCELLENT work on this @visualapproach, thanks so much for sharing this !
[image]

eried, 2022-05-05:
Mine has been working flawless now 😌

visualapproach, 2022-05-05:
Thank you for your support guys! I'll add this finding to the build instructions. Very useful! Wish I understood why the 8ch LLC don't work.

Brynod, 2022-05-05:
Hi,
I can confirm that mine is now working after changing out to 2x 4ch level shifters😊 I don't want to chance it rebooting so am going to put a cap on the 5v line pre-emptively! Thanks for the info chaps and again, excellent work.

PS pump model is a 54154, cant tell you the liner model because I'm using it on an mSpa Alpine 6😂 if anyone needs any help with this conversion then just let me know 👍🏿

svanscho, 2022-05-24:
Ran into the same issue today after the JST connectors finally arrived. I'll try swapping out for a different LLC. I went for a first time right approach and soldered everything onto the PCB sigh...

Is there a PIN compatible LLC with the TXS that works? Or should I just go for one of the 4 spare PCBs I have...

eried, 2022-05-25:
Are you using the version with 8CH? that is exactly what happened to me. It only got resolved switching to the 2x 4CH level shifters... and now I am playing around with that version + ESP32 :D 

[image]

Brynod, 2022-05-25:
+1.....I was using wemos d1 mini, nothing would communicate until I swapped out the 8ch shifter for 2x 4 channel shifter and everything was sorted

Hope this helps

svanscho, 2022-05-25:
@visualapproach I'll leave it up to you but given the experiences from a few people here, wouldn't it be a good idea to either make it the default LLC in the documentation for the project or at least warn people that it does make a difference depending on the pump model?

visualapproach, 2022-05-25:
@svanscho Yes. I mention it in the build instructions and I have a new PCB ordered for testing, that take both kinds of LLC. In dev branch where things happens first before merging to master.
Availability and pricing has changed between the two variants of level shifters so that's why I complemented Eric's PCB. There has been some issues with that one too  so I don't know if there should be a default one. Maybe there will come other PCB contributions as well. But you're probably right

svanscho, 2022-10-03:
Just confirming that in the end I used Eric's PCB with the red LLC (and 560R) and everything is working flawless now!