visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #255: Home Assistant - Target and actual temperature reported as negative numbers?

misterpeee, 2022-04-25:
I'm running 3.2.1 on my 2020 Bali unit - I've enabled the MQTT integration into Home Assistant 2022.4.6 via Mosquitto broker 6.0.1.  The units system is set to Metric the general HA configuration screen.

The target and actual temperature are reported correctly via the Project WebUI and physical display, however, the values in Home Assistant are reported as negative Celsius values.  

Switching to Fahrenheit via Home Assistant MQTT topic, the values switch to the actual/correct Celsius values but the physical display reports Fahrenheit.  Switching back to C will make the values jump to impossible numbers (100c) - switching between F & C a couple of times will in most cases correct the issue - i.e. both HA and physical report the correct temp in C.  Rebooting the ESP8266 will result in the values returning to negative.

[image]

visualapproach, 2022-04-25:
Hi, I think you should try the development branch. I fixed some of those things recently.

visualapproach, 2022-04-25:
Okay. Thing is that after switching unit the esp has to send a new config to HA and possibly an update on the values as well. You could lower the mqtt interval to shorten the max time it takes to correct values but it will always take a small amount of time. I'll see if can be improved.

misterpeee, 2022-04-25:
More info - Lowered the MQTT Telemetry Interval to 300 seconds - Waited over 10mins post reboot and it continued to report the values incorrectly converted to F.

Checking the debug console:

{"CONTENT":"STATES","TIME":1650920546,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":0,"RED":0,"FLT":0,"TGT":38,"TMP":19,"CH1":32,"CH2":49,"CH3":57,"HJT":0,"BRT":7}

HA values:

[image]

So HA is taking the TGT 38 and TMP 19 and reporting target as 3c and actual as -9c - i.e taking the values as F and performing an incorrect conversion to C.

As above, switching to F fixes it!  I assume this triggers the transmission of a new config to HA?

misterpeee, 2022-04-26:
Just built the latest Dev branch version (2020-04-26), configured MQTT with an Telemetry Interval of 300 seconds - Still the same behavior post boot.  Initially reports -18c (for a few seconds so that's fine!), but then it coverts the true C temps to F and reports them as C in Home Assistant.

Example - Reported Temps:
[image]

Home Assistant values:
[image]

So in this example the reported actual 12 C has been converted to -11 F. Target of 38 C has been converted to 3 F - but both reported as C (i hope i've made sense!).

Changing the target temp via the UI results in the value reported to HA following the change (with the F equivalent values)

If i switch to F->C via the UI, Home Assistant instantly reports the correct C values, switching back to C again, it continues to report the correct values.  In addition, if you make a change to the MQTT config and save it - HA reports the correct values.  So this does appear to be an issue with the initial state following a boot.

Thanks

eried, 2022-04-29:
sometimes I see a very low temp, is this the same issue?

[image]

misterpeee, 2022-04-29:
It would appear to be a similar issue - if I'm reading the values correctly on the graph they appear to be the C value converted to F but reported in C in HA.

So 40C is 4.4F and the dip in hot target temp appears to be about 4?

What's interesting is it appears to correct itself? Mine doesn't do that unless I switch units or resave the MQTT config - but then it comes back Post reboot.

visualapproach, 2022-04-29:
I'm not sure it's the same thing. The esp always starts up with target=0 until it gets the first reading. Maybe you or the device restarted when that dip happened?

visualapproach, 2022-04-29:
Looking at a graph here I see a measurement of -7 past night instead of 20C, so probably F/C conversion done in error. Maybe I should always report C over mqtt and let the receiver do the conversion.

eried, 2022-04-29:
> I'm not sure it's the same thing. The esp always starts up with target=0 until it gets the first reading. Maybe you or the device restarted when that dip happened?

it is actually a value around 0 but not 0, some 1C and some -1C, but the target temperature always starts at 4C

visualapproach, 2022-04-29:
4C = 39F. Appears HA thinks 39 is F and converts to C. Meaning when the ESP starts and sends the config to HA it says to expect F. 
You have a different pump model than me, so it may behave differently at startup. Or there is something else that differs.
I have my pump set to C, which is saved in the states.txt file. You can check by browsing to IP/states.txt. UNT:1 is C.
HA general settings = C
Restarting and it shows the correct values in HA.
Tried the same with F on the pump.

I'll monitor the target temp curve and see if I can catch what happens at those jumps.

eried, 2022-04-30:
I uploaded the latest dev and also added a big cap to try to prevent some reboots in case that is the issue. I will report back.

eried, 2022-04-30:
I am wondering if that automatic thermostat from home asistant is actually trying to manage the heater by itself instead of letting the spa cio 😵

visualapproach, 2022-04-30:
You can run mqtt explorer on a computer and catch layzspa/command messages to find out. I don't know, you might be right. My pump is inside the house so can't turn on the pump.

visualapproach, 2022-04-30:
I had one blipp today and there was no command sent to the ESP. The ESP however sent to config changes. First F then C again. Only thing I can think of is a sporadic error in the transmission between cio and esp.

eried, 2022-05-01:
I am pretty sure it is related on how the pump works near the set point when is hot. Now with the capacitor I dont get any random reboots, except when is hot:

[image]

misterpeee, 2022-05-01:
I probably should be clear - the issue I'm having is in testing without the pump heating.  I've only just pulled it out of winter storage and haven't yet it up the spa to test it activity heating etc.

The issue is with the values being reported with it not connected to the spa.

closed 2023-03-13 by visualapproach