visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #257: Home Assistant/MQTT: Hydrojets not available

tympietor, 2022-04-30:
First of all, I want to thank you for this awesome project. I had the entire setup running without a single hickup! Running the Maldives 2021 version (latest Cube model, 6pin).

I got it all running with MQTT on my HA instance, but there's one thing missing; Hydrojets are not passed through with MQTT. I just checked the development branch, but couldn't find it there either.

Now I went a bit through the main.cpp file, and found that I think I could copy the entire MQTT config for the air bubbles, and change the CMD value to 11 and other unique identifiers to identify as hydrojets

e.g. 
```
  doc["payload_on"] = F("{CMD:11,VALUE:true,XTIME:0,INTERVAL:0}");
  doc["payload_off"] = F("{CMD:11,VALUE:false,XTIME:0,INTERVAL:0}");
```

Would that work? 

Either way, would be great to add this as standard MQTT setup :)

visualapproach, 2022-04-30:
Thank you @tympietor you are right. The sensor was not added automatically in HA. Please try dev version 2022-04-30.

tympietor, 2022-04-30:
> Thank you @tympietor you are right. The sensor was not added automatically in HA. Please try dev version 2022-04-30.

Gosh, what an amazing quick response! Thank you so much, it works flawlessly with HA now!

closed 2022-05-02 by tympietor