visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #263: Random flashing display - Lazyspa Helsinki 2021

9khil, 2022-05-08:
I just fitted the pcb to my 2021 unit using miami2021 model. can control using portal and home assistant, but display and button lights on the pump are  flashing randomly.

tried three 510ohm resistors without any luck. 

I can post video and debug log if needed, just wanted to know if anyone had experienced the same issues. 

Board:

[image]

eried, 2022-05-09:
I would suggest trying shorter cables and eventually migrating to the version with this 4CH level shifters -> https://s.click.aliexpress.com/e/_9wDImz

visualapproach, 2022-05-09:
Hi,
What @eried said if the resistors don't do the trick. See [#253 ](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/discussions/253#discussioncomment-2717605) for info where to put them.

9khil, 2022-05-10:
> Hi, What @eried said if the resistors don't do the trick. See [#253 ](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/discussions/253#discussioncomment-2717605) for info where to put them.

waaaaaait a minute! I were pretty sure they would sit between LLC and DISP. I was wrong then. Will try to move them over and shorten cables. Thanks!

emblitz123, 2022-05-10:
Hello,
I suspect there is an error in #253.
was written:
Headers H2:
1----------CIO
2----------CIO
3 resistor CIO
4 resistor CIO
5 resistor CIO
6----------CIO

But H2 is the display!

I have drawn the resistors in the plan.
Is that correct now?
[image]

emblitz

eried, 2022-05-10:
Since the 8CH chip trying to do some automatic detection of the direction (if is in or out) and the resistors help in a experimental way you could use resistors in H2 or H1, or both. It is just part of the experiment.

visualapproach, 2022-05-10:
#253 text corrected now, thanks.

emblitz123, 2022-05-10:
Hello,
I would like to give feedback on my troubleshooting.
I shortened the wires between LLC and Disp and soldered 3x 680Ohm resistors between LLC and Disp.
The display works now and everything is fine :-).
I am very happy.
Thank you for this great project!!!
You can now add the Santorini whirlpool to the list.
It works.

Best regards
emblitz

visualapproach, 2022-05-10:
Thank you @emblitz123 for the feedback! It matters to me and the community to see what solved the issue. Good job!

closed 2022-05-30 by 9khil