visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #270: Hydrojets are displayed although MIAMI2021 was selected

johanneslasnik, 2022-05-11:
I have reflashed several times now, yet the hydrojets keep popping up. 
Also, the button is not in line with the others.

[image]

visualapproach, 2022-05-11:
Alignment should be fixed in development branch. You only need to upload "websocket.js" file.
The hydrojets button will only show immidiately after page loading, and then only when receiving HASJETS:TRUE. Only way it receives HASJETS:TRUE is if you defined a model with hydrojets in "model.h", or if you explicitly change the "BWC_8266_globals.h":
```
#elif defined(MIAMI2021)
const uint16_t ButtonCodes[] =
{
  0x1B1B, 0x0100, 0x0300, 0x1212, 0x0809, 0x1012, 0x1112, 0x1312, 0x0200, 0x0000
};
const bool HASJETS = false;
```

Please check the console in the browser:
[image]

visualapproach, 2022-05-11:
Update to dev branch 2022-05-11. I see that it says Hasjets:'0' instead of Hasjets:false.
It's a bug in your version.

visualapproach, 2022-05-11:
Actually it seems that 2022-04-11 shouldn't send '0' as a string. Maybe you downloaded just before I fixed it and I maybe didn't update the version date? Strange but try compiling/uploading the today version and see.

johanneslasnik, 2022-05-11:
Yep, working now!
Thank you

[image]

visualapproach, 2022-05-11:
No problem 🤝

closed 2022-05-11 by johanneslasnik