visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #272: Hydrojets displayed with "no hydrojet model"

DandeMC, 2022-05-12:
Hello,

I don't understand why the hydrojets are shown in the menu. I selected in model.h "#define PRE2021			//the older one, no hydrojets" and I have the Palm Springs from 2020, but the menu shows the hydrojets.

Is this a bug or a settings misstake from me?

[image]

visualapproach, 2022-05-12:
#270

DandeMC, 2022-05-12:
Oh perfect, thanks. Is there any way to update without lose my settings?

visualapproach, 2022-05-24:
Yes. Uploading the new code does not erase the files. Uploading a new file image though, does. But you can upload every file in the data/ directory through the "upload.html" page from the web interface.

DandeMC, 2022-05-24:
ok and the upload system nows where it must storage the file and overrides it?

visualapproach, 2022-05-24:
Uploading files through web page is like saving a file to hard drive. Uploading the file system again is like formatting the harddrive and adding all files.

closed 2023-03-13 by visualapproach