visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #275: 4 Wire Energy kWh values way off?

chunkysteveo, 2022-05-19:
Howdy all,

I'm a little confused about the reported kWh values from the unit over time. Running a 4-wire pump importing into HA with MQTT. I have also setup a Daily Energy meter helper which resets every day to allow me to see the energy use per day.

In idle mode all seems well, see graph, it uses about 5Wh a day, which seems right, as it's just powering the unit screen and the ESP:
[image]

I've not filled the tub and set the pump running. In the code it states the pump is 40W, but watching the energy values come into Home Assistant, it's using 100'skWh per day - one full day of the pump being 1039kWh??
[image]
(the vertical spikes in energy use is me hitting the blowers to mix the water)

A day with the pump on should only be about 1kWh, right? I reset the times on the web gui and watched the KWH value in the TIMES message rise to KWH:12 in just 18 minutes, I don't think that's right?

I'm not using the latest version of the files, but it's only a couple of months out from the latest release.

Any thoughts on this, am I missing something here?

visualapproach, 2022-05-19:
I'll have a look. Would be one expensive pump if it's correct

visualapproach, 2022-05-19:
should be fixed now :-) (dev branch)

chunkysteveo, 2022-05-21:
Ahhh ok! Cool! Let me get the files downloaded and check that out! Thansk.

chunkysteveo, 2022-05-31:
Updated the files and uploaded to the pump - the kWh values are way lower now, much better! Thanks as always @visualapproach !!!

visualapproach, 2022-05-31:
You're welcome! If the electricity prices goes up further I can lower the reported kWh more 😄