visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #281: scheduled pump off doesn't work

DandeMC, 2022-05-24:
Hello,

I don't know if this is a mistake. I have 2 planned settings active. 8pm pump on, 10pm pump off. it is working. but if I use the pump myself, e.g. turn on at 6pm, it will not turn off automatically at 10pm. is that intentional or a bug?

DandeMC, 2022-05-24:
Firmware version: 2022-04-11

I think its always and I turned it on via web page.

DandeMC, 2022-05-24:
I was wrong. I turned it on via mqtt, not web page.

DandeMC, 2022-05-24:
[image]

visualapproach, 2022-05-24:
thx, seems okay. Can you try again tonight to verify it wasn't just the time server that was wrong or something?

DandeMC, 2022-05-24:
yes I will test it and write you tomorrow.

DandeMC, 2022-05-25:
It's the same big. I turned it manually on at 7:58pm and it doesn't turned off on 10pm.

visualapproach, 2022-05-25:
Assuming you have a home automation system since you use MQTT. Can it be the HAS that triggers the pump on again? You can use MQTT Explorer to log all commands sent to the ESP.

DandeMC, 2022-05-26:
No it can't. I log the energy use from the pool too and I can see that it stays on when I turned it on manually. Yesterday it worked fine with the schedule on and off without manually turned on. 

[image]

visualapproach, 2022-05-26:
I didn't ask whether it was on or off. You already stated that. I asked if something else turned it on again. What's sending mqtt commands and can you check that it's not sending a contradictory command at 22:00?

DandeMC, 2022-05-26:
I checked this already. but chance would also be small.

closed 2023-03-13 by visualapproach