visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #283: PlatformIO *** [upload] Error 2 A fatal error occurred: Invalid head of packet (0xE0)

chrstnmr, 2022-05-27:
Hello everyone, 

I am trying to update my ESP currently. It ran for a year now but unfortunately sometimes it randomly triggered to switch the pump or else on after refreshing to readout the current value via MQTT. So before digging into this I wanted to update the SPA Software. 

I re-downloaded the project and wanted to update it via USB. However, it fails to upload via USB. Unfortunately my OTA is not on yet.

The error I receive:

A fatal error occurred: Invalid head of packet (0xE0)
*** [upload] Error 2

I am using a correct usb cable and the device also shows in the PlatformIO Device list. What am I doing wrong? Is it because the ESP already has the software on it? Does it need a complete reset?

Here are some more information on the monitoring:

Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.usbserial-1120
Uploading .pio/build/nodemcuv2/firmware.bin
esptool.py v3.0
Serial port /dev/cu.usbserial-1120
Connecting....

Thanks for your help everyone.

visualapproach, 2022-05-27:
If it's not hardware issues you could try the 'clean build' option in platformio. It clears the build cache.

chrstnmr, 2022-05-27:
@visualapproach Thanks for your reply. Could you provide me a short info on how this works?

visualapproach, 2022-05-30:
Click the platformio icon, and then clean build button

chrstnmr, 2022-05-31:
Okey, I was able to solve it:
add

```
upload_port = /dev/cu.wch*
monitor_port = /dev/cu.wch*
```

to platformio.ini

closed 2022-05-31 by chrstnmr

mattburns, 2023-09-16:
In case anyone else is seeing this error, it could be dodgy usb cables (especially if you're using a macbook and usb-c adapter like me)

I was able to work around having to find better cables by reducing the `upload_speed` baud rate. You do this in `platformio.ini` , eg:

    ; upload_speed = 921600
    upload_speed = 115200

peetzweg, 2024-05-01:
@mattburns had exactly this problem. Thanks! Happy Spa-ing! 🙌