visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #298: Palm Spring AirJet - S100101 Not work Display and Web interface

lucableve, 2022-06-15:
Hi, I connected as per instructions and soldered the esp and the level shifter but nothing works.
As soon as I switch on the power, a jingle song starts from the display and shows a date and nothing more on the display as well as the buttons no longer work
From the serial web interface I don't see anything pass, not even the temperature.

Thanks

[image]

[image]

https://user-images.githubusercontent.com/28319157/173850228-7f93758d-8c44-422e-9a78-c629b16e72f8.mp4

visualapproach, 2022-06-15:
Hi! From the web interface go to inspect console in the browser. Look for buttoncodes and see what it says when idle and when pressing some buttons.

visualapproach, 2022-06-15:
Okay, you're supposed to choose one of the existing models. Try the miami2021.

lucableve, 2022-06-15:
Nothing to do, I've tried them all, I think I'm at the point that it's the TXS0108E

lucableve, 2022-06-15:
I solved it, took a new board and re-soldered everything by changing the nodemcu and level shifter as well. I noticed that the cables were also the critical part as I was using bad quality idc.
Now it works, so the model you indicated is correct. I thank you

closed 2022-06-15 by lucableve