visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #299: Unable to interface with Lay-Z controller

nos86, 2022-06-16:
Hello,
first of all, my congrats for this amazing project. It is very useful and I hope it will save a lot of time trying to automatize my spa.

My spa is the Lay-Z Bahamas bought in September 2021.

I have soldered this project using Wemos D1 Mini and the TXS0108E and checked the routes 3 times to be sure that nothing wrong is there.
After that, I have uploaded the software and the LittleFS checking that devices goes online regularly.
As last, I connected it to my controller, but the result was unexpected: display is off and the esp is offline.

After several tried in different configurations, I put the ESP in parallel to the CIO-DSP direct connection and this is the results:
- If I connect only the power pins, ESP boot up and controller works regularly
- If add the third pin (DATA), ESP is still working but DSP is off
- If add the forth OR the fifth pins (CLK or CS), even the esp won't to boot (even I disconnect DATA pin), device is still rebooting, see picture attached 
[image]

If I have understood correctly CIO is the controller side and DSP is the display side.

If my understanding is correct, can someone helps to understand what is wrong?

visualapproach, 2022-06-16:
Hello, and Thank you for giving a detailed explanation of your problem and your actions and setup. I'm not aware of anyone reporting this behavior before. Some people with 4-wire pumps have problems with crashes but the 6-wire pumps usually (works?) have problems with flashing displays which can be mitigated with some resistors. And yes, CIO is the PCB controlling the stuff inside the pump housing, and DSP is the PCB controlling the display and touch buttons. There is many people reporting success with different pump models so the first suspect in this case is the hardware, even if you said you've checked. Can you state the model number on your pump, and also a picture of your PCB, both sides please? Hope we can make it work

skyluker, 2022-06-17:
As far as I'm concerned S100101 requires another type of level shifter.
https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/251#issue-1205598202

visualapproach, 2022-06-17:
Thank you for chiming in @skyluker. That's a great suggestion, I didn't have the time to think it through properly but thank you for sharing your experience and supporting the project!

visualapproach, 2022-06-17:
@nos86 if redoing it a third time, may I suggest you use the other type of converter as @skyluker suggested. And while you're at it why not go for pinout as PCBv2 to make the routing easier. You just have to change the PCB define in model.h

nos86, 2022-06-19:
thank you for advice @skyluker.
I ordered them on Amazon and they will arrive on Monday. Then I will replace the LLC and check it again.
In your opinion, new LLC will solve the problem of missing boot as well?

A suggestion for @visualapproach is to create an applicability table where for each model it's defined if resistors are needed or not and which kind of LLC to be used. This will help new makers (like me) to make it working easily.
As soon as my pcb works, I'll make a table for it and submit to you as a pull request

closed 2022-06-23 by nos86