visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #301: Layzspa time to target calculation

strutings, 2022-06-19:
Hi, 

Just wondering how the time to target temp is calculated? Seems like it's stuck here after upgrade to 2022-05-24

visualapproach, 2022-06-19:
It takes two measurements and extrapolates the time taken. So it will show garbage until 2 temp updates has been given.
Subject to changes in the future

chunkysteveo, 2022-06-22:
The 4-wire version for TTT does not appear to work - i've always had that as a "to do" to check code.... 

Heating Est is always the same as Time in the web gui, and TTT value is always 0 (as the times appear matched/copied)?

chunkysteveo, 2022-06-22:
OK - found code to calculate _tttt_calculated in the 6 wire which wasn't present in the 4 wire base code. Hot tub is heating up, so added it to my dev branch and now getting readings (although they are negative values, is that right? Do I have to wait for a couple of degrees to change for the calculations to give a positive time value for _tttt?)

visualapproach, 2022-06-22:
Yes you Will oboy get meaningful values after two updates so it knows the rate of change. Negative values is shown when the target vs real temperature is diverging (the graphed lines would cross each other in the past)

chunkysteveo, 2022-06-22:
Hmm, OK. With the copied code from the 6-wire, I was heating the 'tub, and the TTTT time value in the web GUI always showed a time estimate that was LESS than the current time.

I don't think I fully understand the TTTT calculations, and what it's showing. My TTTT value just gets more and more negative as time and heating progresses. Maybe there is a more suitable method to update the times ONLY when the temperature value changes - so the hour value is only updated on temp change, and if temp = target..... TTTT = 0?

(or is that how it's supposed to work, and my attempt with the 4 wire code is messing it up?!)

Steve

visualapproach, 2022-07-09:
Dev branch updated with a beta test: virtual temp.
It needs to know the ambient temperature to be accurate. HA can provide it via mqtt if you make an automation. Command 14 I think.
It's supposed to be self adjusting so heat up, and let cool to where your lower setting use to be. The pump needs to be running all time when calibrating.
When turning off pump the virtual temperature will show the supposed water temp in the tub, not in the pump.
It definitely over engineered but it's a good challenge...
You are welcome to try it out.

visualapproach, 2022-07-09:
Only for 6 wires for now...

closed 2023-03-13 by visualapproach