visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #304: 4 wire web command schedule and ui fixes for palm springs hydrojet 54144

cobaltfish, 2022-06-22:
I noticed that on the web ui, bubbles / air jets show on a hydrojet only:
[image]
I also noticed that while it can be done by uploading commands in the text file, the UI doesn't support setting God Mode on the config page, which I use to schedule the heating for off peak electricity:
[image]

I've made a fork with some fixes for these:
[image]

[image]

(I've also experimented with the async libs for the webserver, and have been running for about a week without any E13 errors, which is a new best for me!) 

I updated Code/4-wire-version/data/WebSocket.js to use 
   msgobj.HASAIR as logic for the air title, button, totals like used for jets
I also added IDs + divs to Code/4-wire-version/data/index.html so that the WebSocket.js would be able to do the getElementById in the same way
This was a new item required in Code/4-wire-version/src/main.cpp 
  doc["HASAIR"] = HASAIR;
and is set in Code/4-wire-version/lib/BWC4W/BWC_8266_4w_globals.h for a new type NO54144
I've given it a bit mask for the heating to be 2 stage, as I'd noted from the serial logs that when heating was turned on using the buttons it did so

For the changes on the config page, that was just in the Code/4-wire-version/data/config.html, to add additional items to the command list and drop down menu

https://github.com/cobaltfish/WiFi-remote-for-Bestway-Lay-Z-SPA/tree/feature/4w-palm-springs-54144

visualapproach, 2022-06-22:
Cool! I'll check it out. Thx

visualapproach, 2022-06-28:
Really nice, thanks for sharing. I pulled it into development branch and reset some defaults in model.h. Thank you, I really hope this will solve the communication problems for some people with the async library.

closed 2023-03-13 by visualapproach