visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #306: Start behavior is not saved

e-i-k-e, 2022-06-23:
Hello,

I have the problem with the current version that the start behavior is not saved. Default is "Last saved states" but I would like to have it "off".
After I have selected and saved the setting and then go to Settings again, or even if I restart the ESP, "Last saved states" is always displayed here.

Firmware version: 2022-05-24
Model: PRE2021

[image]

visualapproach, 2022-06-24:
Hi. Save off and then check the page layzspa/settings.txt
It should end with 'restore: false'.
Save last state and check for 'restore: true' in the layzspa/settings.txt file.
Does it do that?

e-i-k-e, 2022-06-24:
> Hi. Speichern Sie ab und überprüfen Sie dann die Seite layzspa/settings.txt. Sie sollte mit 'restore: false' enden. Speichern Sie den letzten Zustand und suchen Sie in der Datei „layzspa/settings.txt“ nach „restore: true“. Tut es das?

Excuse me. But how can I call up the files?
[image]

visualapproach, 2022-06-24:
Enter the url in the address field in your browser:
"layzspa.local/settings.txt" or 
"192.168.3.156/settings.txt"

e-i-k-e, 2022-06-24:
> Enter the url in the address field in your browser: "layzspa.local/settings.txt" or "192.168.3.156/settings.txt"

[image]

`{"CLTIME":1655999700,"FTIME":1655999702,"UPTIME":310551,"PUMPTIME":79532,"HEATINGTIME":25983,"AIRTIME":0,"JETTIME":0,"TIMEZONE":0,"PRICE":0.270000011,"FINT":30,"CLINT":14,"AUDIO":false,"SAVETIME":"2022-06-24 11:49:56","RESTORE":true}`

[image]

`{"CLTIME":1655999700,"FTIME":1655999702,"UPTIME":314654,"PUMPTIME":83541,"HEATINGTIME":26002,"AIRTIME":0,"JETTIME":0,"TIMEZONE":0,"PRICE":0.270000011,"FINT":30,"CLINT":14,"AUDIO":false,"SAVETIME":"2022-06-24 12:58:20","RESTORE":true}`

visualapproach, 2022-06-24:
Okay so it does not save properly. Either wait until I can check it out or just upload the newest dev branch. Remember to upload the new data/ files also (the html etc pages). I don't know if latest code solves it but it works for me.

e-i-k-e, 2022-06-24:
> Okay, es wird nicht richtig gespeichert. Entweder warten, bis ich es auschecken kann, oder einfach den neuesten Dev-Zweig hochladen. Denken Sie daran, auch die neuen Daten/Dateien hochzuladen (die HTML-Seiten usw.). Ich weiß nicht, ob der neueste Code es löst, aber es funktioniert für mich.

OK. I would like to fix this. Because a WLAN separation is enough and it starts with the previous settings. I experienced that the heating was then on unintentionally.

Can I upload and replace the files in this way via the web interface via "file upload"?
[image]

visualapproach, 2022-06-24:
Yes you can. It's a good way to save your logs like uptime etc.

e-i-k-e, 2022-06-24:
> Ja, du kannst. Es ist eine gute Möglichkeit, Ihre Protokolle wie Betriebszeit usw. zu speichern.

hmm.. still have the problem.
Can you estimate what a new version you are releasing?

visualapproach, 2022-06-26:
I'll fix it within a day I think

visualapproach, 2022-06-26:
Fixed. you need to upload "config.html" and it should work :-)

e-i-k-e, 2022-06-26:
> Fixed. you need to upload "config.html" and it should work :-)

Thanks! It worked! :)

[image]

closed 2022-06-26 by e-i-k-e