visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #307: FHEM Support ?

xGDNAx, 2022-06-26:
hello has someone already integrated this into FHEM?

visualapproach, 2022-06-26:
You can use MQTT to communicate with FHEM. No idea how though, but it should be doable

xGDNAx, 2022-06-27:
Hello, yes, it should definitely work via MQTT, but how? I don't have the experience there, MQTT commands should certainly all run the same way and depend on the system, right?
GOOGLE translates for me

chunkysteveo, 2022-06-27:
This would be best asking in a FHEM support forum or group chat. MQTT from the Spa controller is universal and can be picked up and managed from any MQTT broker. It would be up to the supporting system (FHEM) to deal with the commands.

Maybe ask the same question in an FHEM online forum, for assistance? We can certainly point you to where you would enter your MQTT broker details, and what the spa sends via MQTT, but I don't think anyone here has used FHEM?

Steve

xGDNAx, 2022-07-03:
Hello, sorry for the late reply.

I don't have all the parts there yet and wanted to look for data, but I haven't found what I was looking for 100%.

where can I find a list of the exact MQTT commands? English isn't really my thing, it's really hard for me.

visualapproach, 2022-07-08:
The available MQTT commands is described in the build instructions but I get that it's not super easy to understand when new to it. But basically you need to publish a message to a topic.
To command the layzspa you need to publish a command to the topic "layzspa/command". The command is a JSON string with four key/value pairs. Example '{CMD:0, VALUE:1, XTIME:0, INTERVAL:100}'
This will send the command 0 (list of commands in the instructions) with the value 1 (On) to the esp and it will be executed immediately and repeated every 100s

xGDNAx, 2022-07-08:
Sorry, I don't understand anything more, the MQTT structure is completely different from what I know.

chunkysteveo, 2022-07-08:
You need to learn and understand MQTT from the ground up, and then apply what you have learned to figure out how to pull in information, and push out commands from and to the spa unit.

MQTT is very daunting to get your head around, but when you learn how it is structured, and how it's all setup, you will soon get the hang of it and be able to interpret and break down the messages and commands.

There is a good program - MQTT Explorer - which has helped me see all my network MQTT traffic, and it helps me visualise what and who is talking back and forth. That helped me to understand MQTT structure better.

[image]

See the image above - this is MQTT Explorer - and all the traffic to and from anything to do with "layzspa". Ignore "layzspaADDA" and "homeassistant".

On the left is where all the MQTT topics are being published, you can see the `Status = Alive`, `reboot_time`, `message`, `times`, etc. There is on there called `command` - that is the topic `layzspa/command`, and it's bee populated with some commands by me pressing the buttons on the ESP8266 web GUI.

On the top right you can see the values of the topics, here I have highlighted the command topic, and you can see the value is a JSON object of:

```
 {
 "CMD":2,
 "VALUE":1,
 "VALUE":0,
 "XTIME":0,
 "INTERVAL":0
 }
```

Ignore that `VALUE` is there twice - MQTT Explorer is tracking a change in that value from 1 to 0, red to green - this is how you can see things happening on your network with MQTT.

Your FHEM setup will need to subscribe to the topic `layzspa/message` to get all the info about the states of the pump, heater etc. When something changes on the spa, the MQTT topic will change, e,g, `AIR` will go from 0 to 1 when the blowers are switched on. In my Home Assistant, I have a switch which watches that value of `AIR`, and when it goes to a 1 - the "Bubbles" switch goes ON.

So bottom right is the "Publish" section. Here is where you can publish things to the topics.... Here is where the commands come in. You are going to publish commands to the Layzspa via the topic `layzspa/command`. The message of this needs to be a JSON object, looking much like:

`{"CMD":2,"VALUE":1,"XTIME":0,"INTERVAL":0}`

If click publish as per the button in the image above - my blower bubbles will come on! Command = 2, Value = 2 - which in the documentation states the ESP8266 will instruct the spa to turn on the blower fan. Likewise you will lots change in the window on the left as the spa reports back the things have updated - the `AIR` in the topic `layzspa/message` will change to 1, as it's saying - "hey - my blowers are on!". That will make my switch in Home Assistant turn on too.

`{"CMD":2,"VALUE":0,"XTIME":0,"INTERVAL":0}` publish that, and the blowers will turn off.

So your FHEM will now need to be able to publish MQTT messages on the topic `layzspa/command` in the JSON format to instruct the spa to do things.

That's your back and forth from MQTT to and from the Spa ESP8266 unit. 

It will make more sense when you have it all hooked up - if you publish a commend to turn on the blowers, and they actually come on, it's magic, and you will smile!!! Doing it as a test with nothing to see change or happen won't encourage you to keep learning.

Install/download MQTT Explorer, sign into your MQTT broker server, and watch and interact with the MQTT topics, and enjoy learning!!

Steve

xGDNAx, 2022-07-08:
THX for Help, its work for Me.
[image]

but i cant set Temp 
command = {CMD:0,VALUE:{{ value|int }},XTIME:0,INTERVAL:0} 30 is this right ??

chunkysteveo, 2022-07-08:
To set a temperature you would send... `{"CMD":0,"VALUE":30,"XTIME":0,"INTERVAL":0}` for 30°C. The value has to be an integer i.e. no decimal point, whole numbers (from 20 - 40)

xGDNAx, 2022-07-08:
Hy yes, via FHEM I get the slider so I have to find the way how he puts the number in the code for me

xGDNAx, 2022-07-10:
Hy its work for me, not perfekt but its work

closed 2022-07-10 by xGDNAx