visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #316: How does a firmware update work ?

emblitz123, 2022-07-01:
Hello,
I use the firmware version 2022-04-11.
[image]
Where can I see whether there is a more up-to-date firmware and how can I make an update?
On Github the firmware is called differently (3.xx).

My circuit board is often not accessible.
In IoBroker, "Dead" is then displayed via MQTT.
If I pull the 230V mains plug and plug it in, everything is fine again for a few hours until the circuit board hangs up again.

I would like to try to do an update to fix the bugs. 
In the menu of the web interface there is an item "File Uploader". Can I update it?
[image]

Best regards
emblitz

visualapproach, 2022-07-01:
Hi

There is different branches on github. Presently MASTER and DEVELOPMENT. All new updates goes into development branch first. Then after a while I merge it to master.
To check code date, go to Code/x-wire/config.h and look for FW_VERSION=xxxxxx

Updating is exactly the same as first install, except you can upload over the air. See build instructions. LittleFS files can be uploaded as first time or one by one from the upload page.

Unfortunately I don't know a way to sync github version tags with the code date.

emblitz123, 2022-07-01:
OK thank you.
I thought the update would be possible via the website.
What did you add the File Upload menu item for?

Best regards
emblitz

visualapproach, 2022-07-01:
To upload files to the file system without wiping the whole drive. Nice to save the stats and settings

closed 2023-03-13 by visualapproach