visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #317: Hawaii Square S100102 6 Wire

GoblinRules, 2022-07-01:
Hello,

Hoping someone can possibly help.

I have a 2021 Hawaii with a Square Pump and a Model N# S100102 on the side, so have used the #define MIAMI2021 model.

(On a side note I struggled to get my wifi to connect, not sure if it was down to the starting/end characters, so created a guest network with a simple password, connected then in the web interface changed the ssid & password there, reason I tried this was when doing it through the AP stage I saw the additional characters being added in the url.)

On the basic setup without resistors on CLK/Data/CS I can get the display to work but none of the buttons on the panel work nor does any of the actions from the web interface or any from Home Assistant.

I then Tried disconnecting the 10K Resistor but still had the issues.

I then readded the 10K and the 560 Ohms on the CLK/Data/CS but now the display does not work it only has 3 random characters stuck statically, the buttons do not work and the web interface/Home Assistant also does not work.

Are there any other suggestions?

hjkoster400d, 2022-07-02:
I have the Helsinki 2021 model. I think the pump version is S100103. I used the 10k resistor but not the 560ohm resistors. Sometimes the display works, but most of the time it's flickering all kinds of colors/on/off. 
Does 470 ohm also work (that the value i've laying around) Otherwise i've 680, what would be the best in this case? 
I used the Miami2021 model in the firmware  and board v1 (from jlcpcb). Thanks in advance

visualapproach, 2022-07-02:
@hjkoster400d 680 has worked for people so try that first

segapus, 2022-07-02:
I too have the Helsinki 2021 model. My pump is an S100101. I just now tried to attach the wifi module and the controls in the web interface work, but the display and buttons do not. I get a weird sound coming from the display that is in sync with the small LED blinking (a weak sound goes on each time the LED lights up). Anyone stumbled upon this issue?

visualapproach, 2022-07-02:
> To confirm I am using the TXS1008E, I have the Board set to V1.

See this post #312

there707, 2022-07-03:
Similar to @GoblinRules and @segapus I too can't get the display/buttons to work. I'm using the TXS1008E and used MIAMI2021(does this matter if my actual model is Palm Springs?) & PCB_V1 set in model.h. 
My pump is the square model S100104 6wire (does the slight variations in these models matter?). I have used the 10k ohm per instructions, but not any other resistors (680ohm) on clk/data/cs as I seen earlier in this thread. 

Lots of questions...appreciate any thoughts?

there707, 2022-07-09:
Ok @GoblinRules it's my turn to apologize. This is what I think you did based on [#312](https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/discussions/312)
1. Update pins
[image]
2.Wire per PCB_V2 (means cross overs from PCB_V1)
[image]
3. Leave TXS1008E in and don't jumper out

These questions for me are unanswered or more likely just not understood from https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/discussions/312
- did you set model.h to PCB_V2 or PCB_V1? To me it has to V1 so the new pins take effect. 
- I'm questioning the D0 in the new pin definition vs D1 that existed before.

GoblinRules, 2022-07-10:
Hi

I didn’t change change model.h from PCB_V1 nor did I change BWC config. 

So to clarify my process. 

- I had a V1 board
- I removed the TXS1008E
- I bridged A1-A8 to B1-B8 (1-1/2-2/3-3 and so on)
- Do not bridge the power and ground
- Only had the 10k resistor 
- No resistors on the display

I posted pictures of the setup I had in the other post but let me know if you need more.

there707, 2022-07-10:
Great, I think I got most of this. You say that you do not have the resistors on the DSP but the picture must be an older one? Without these resistors, I get sound for startup and then the display flashes through some characters and then shows temperature. Brightness slider works but that's it. Can't start pump via DSP or GUI. 
[image]

GoblinRules, 2022-07-10:
Yeah that's an older pic, this was my final setup. https://imgur.com/a/vhmQ4kr

I had the sounds originally but after the bridge setup I no longer get that weird noise.

closed 2023-03-13 by visualapproach