visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #321: FRS: Auto unlock

raomin, 2022-07-04:
I find it cumbersome to always have to unlock the display for pressing a new command...

What about having an Auto unlock command that would, when a button is pressed in a locked state, send 
```c++
      _qButton(LOCK, LOCKEDSTATE, 0, 5000); //press LOCK button until states[LOCKEDSTATE] is 0
```
so that the display is automatically unlocked and the command sent.

Maybe a different beep could indicate the user to wait slightly longer than usual...?

visualapproach, 2022-07-10:
Thanks for your suggestion. Agreed, but some people with kids may still want the lock. I might arrange something some day.

raomin, 2022-07-10:
It would be a config based feature that only interested people would activate.... I can prepare a PR for this.

visualapproach, 2022-07-10:
Please do! Make it to the dev branch please.

closed 2023-03-13 by visualapproach