visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #323: Upload failed since PlatformIO last update

Metarko, 2022-07-07:
Hello and at first many Thanks to visualapproach for the very great project. 
I will send you more then one coffee ;)

Does anybody else have problems with uploading since yesterday ( new update of platfomIO). ?
I´m not an expert but it seems as visual studio have problems to communicate with the ESP.

Anybody else how have such a problem or an solution?

visualapproach, 2022-07-08:
add this to platformio.ini, and next time please post screenshots so we can see what's wrong.
`upload_port = COM4`
replace COM4 with whatever number or mac versions of the COM port you got.

Metarko, 2022-07-09:
Thank you for your quick response. I have tried it bevore i have written you. Heres the Error Message while uploding:
[](url
[image]
)

visualapproach, 2022-07-09:
Yeah, you need to "clean all" (clean the cache on the PC) in the pio menu. And pio no longer recognizes ESP with ch340 UART. Maybe they will change it in future versions. So you need the upload_port = xxx until further

Metarko, 2022-07-09:
Thank you very much , now it´s working again. Sorry for this nope question :) .

closed 2023-03-13 by visualapproach