visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #330: Display works, nothing else

menime8, 2022-07-12:
Hi there, 
i have a Dreamstream [SPA](https://m.bestwaycorp.at/Product/Detail/1060151NOX22), which has exact the same electronics like Lay Z Spa. Both selled from Bestway. 
My Pump looks like the newer 2021 model.

[image]
[image]
[image]
[image]

I started with PCB V2, NodeMCUV3 and the TXS0108E. First try, nothing happend. 
Second try with 10K resistor, it Starts, DSP starts flashing.
Third try: 560 ohm resistors between DSP and ESP. Display works fine, melody and IP shown. 

But i can't get any connection to the CIO. Wires and connections are all checked with my meter.

I hope anyone can give me a hint 🤔

Thanks

menime8, 2022-07-12:
> Maybe try resistors btwn LLC and cio also? Or switch to the MOSFET version of LLC. Or try without LLC which is probably fine in practice but may blow the esp in theory according to some people. For the price I'd take the chance...

Tried both versions. Resistors everywhere, then only ESP direct to CIO and DSP. 
[image]

SigmaPic, 2022-07-12:
> I have 6 wires, +5v, 0V, RX, TX and one for Sound. What does the last wire?

It's a 3 wire SPI line. Isn't it ?
So, data, clk and chip select.

If you still have access to the pump PCB.
May you tell me if you can read the reference of that chip 

[image]

menime8, 2022-07-12:
I thought it was RX TX.... 
I will have a look tomorrow for chipnumbers.

visualapproach, 2022-07-12:
> Tried both versions. Resistors everywhere, then only ESP direct to CIO and DSP.
>
> [image]

Oh I see now that you connected 5V to the 3V3 input. Remove that blue wire and check that the ESP is not toasted

menime8, 2022-07-12:
> Oh I see now that you connected 5V to the 3V3 input. Remove that blue wire and check that the ESP is not toasted

It isn´t.

I used the plan from here [https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/discussions/312#discussioncomment-3056766](url)

I know it could be toasted, this was my last try on sunday.

menime8, 2022-07-13:
> > I have 6 wires, +5v, 0V, RX, TX and one for Sound. What does the last wire?
> 
> It's a 3 wire SPI line. Isn't it ? So, data, clk and chip select instead of RX and TX.
> 
> If you still have access to the pump PCB. May you tell me if you can read the reference of that chip
> 
>
[image]
I read Nuvoton MS51FB9AE

https://www.nuvoton.com/products/microcontrollers/8bit-8051-mcus/industrial-8051-series/ms51fb9ae/

menime8, 2022-07-13:
Today i tried many different options, with and without resistors, with 560 and 680. It wouldn't work. Checked again all connections.
The other mosfet LLC arrives, too.
I see my ESP has resistors between D3 and 3v and D4 and 3v. I ordered a other ESP. More Info i have on the next few days.

visualapproach, 2022-07-13:
Did you get rid of the blue wire?

menime8, 2022-07-13:
Yes
[image]

visualapproach, 2022-07-13:
Good, I would also try some caps on CIO side and use the B port to get a better LPF. And/or overclock the ESP if there is something speed related.

visualapproach, 2022-07-13:
Ok so both ends has 1K resistors and unknown caps. So we can calculate a range of cap values with the RC formula. Cut off freq should be higher than the SPI freq. The SPI on my pump is 22 kHz IIRC. I reckon 200-1000 pF would do it.
I need to think a bit on this but what you can check with the scope is the clock frequency on the clock line. And measuring the highs and lows voltages.
I'm not an EE so don't expect miracles here but let's try to find this out

SigmaPic, 2022-07-13:
> I read Nuvoton MS51FB9AE
> 
> https://www.nuvoton.com/products/microcontrollers/8bit-8051-mcus/industrial-8051-series/ms51fb9ae/
> 

Thanks. It's a microcontroller. I will check the pin config capabilities (push pull, open drain, pull up...)

Looking at the signals with a scope may allow you to detect the root cause of the issue. You can look at signal on CIO side but also on esp side and see if something goes wrong.

> I'm pretty sure, it works with that older Nodemcu.

Which one do you talk about ?

@visualapproach Did you implement a sort of test mode ? We can imagine plugging both connectors from the BWC  together. Then you generate some signal in one side and you check that you can read them on other side.
Only the 5V should be added. By doing this, it would allow to check the BWC board.

visualapproach, 2022-07-13:
That's a great idea @SigmaPic 
I'm on it

SigmaPic, 2022-07-13:
> That's a great idea @SigmaPic 
> I'm on it

I'm very busy this time. Otherwise I would have propose a PR.

Maybe you can try to generate 101010 at the rate of the SPI and check if you read the correct bits. Then you change the polarity of the inputs and outputs.

SigmaPic, 2022-07-14:
> I'm pretty sure, it works with that older Nodemcu.

Ok. I see the resistors: 12k pull up on CS and CLK on DSP side.
On CS, it should not be an issue because this pin is set at high level at startup because it's an active low => the pull up is a good think
For the CLK, it may not be an issue because if CS is high, CLK should no be taken into account.

menime8, 2022-07-14:
> > I'm pretty sure, it works with that older Nodemcu.
> 
> Which one do you talk about ?

I have the version 3.2. i ordered the 3.0 version with the CH340

SigmaPic, 2022-07-14:
> Yes
> [image]
> 
> 

On this photo you don't have the resistors on CIO side. Do you confirm that you tried with them ?

menime8, 2022-07-14:
> > Yes
> > [image]
> 
> On this photo you don't have the resistors on CIO side. Do you confirm that you tried with them ?

Yes, tried 560 and 680 ohms

visualapproach, 2022-07-14:
@SigmaPic I wonder if we can/should set the gpio:s to open drain or with pull up resistors. Don't remember what it is now and what can be done, but could be worth a look. When I have time

menime8, 2022-07-14:
One question, isn't it possible to use the Miso/Mosi/CLK from the ESP?

[image]

visualapproach, 2022-07-14:
Espressif states mode 3 is not supported.

SigmaPic, 2022-07-14:
> I'm pretty sure, it works with that older Nodemcu.

I think it won't change anything because there are some pull up on your mosfet LLC. See my shematic below.
So the pull up on esp side won't change anything.
But, you can try to be sure. Maybe I'm wrong.

> @SigmaPic I wonder if we can/should set the gpio:s to open drain or with pull up resistors. Don't remember what it is now and what can be done, but could be worth a look. When I have time

I take a look at the full schematic and I found something strange that may explain some troubles.
The issue concerns the mosfet LLC when data are sent from DSP/CIO to ESP.
The other way is ok (from ESP to DSP/CIO).

As you can see on  my simulation, the voltage on ESP input stay at 1.5V when DSP/CIO drive the line to 0.
This comes from the serial 1k resistor.
On this simulation, I put one 1k which is the one in the DSP/CIO and one 1k which is the one we put on our PCB.
If I remove one 1k, it's better but not perfect.
To get perfect results, I need to remove both 1k resistors which is not possible because we have at least one.

[image]

Why it happens ?
Because these two 1k resistors act as a voltage divider with 10k pull up on the LLC (5V side).
This means that the input of the LLC is at 0.83V instead of 0V due to this voltage divider.
Because of that, the mosfet is not driven and the output of the LLC is 0.83V + mosfet diode forward volatge which is about 0.8V.
So we get about 1.5/1.6V in output of the LLC which is greater that the input low voltage of the ESP which is 0.25x3.3 = 0.8V.

To confirm that issue, we could developp a dumy test SW that just put all pins in input and print the state in the console.
Then we can manually put 5V on each pin and check if it's correctly detected.

As a workaround, we can eventually modify  the resistors mounted on the LLC.
Put for example 50k instead of 10k.
An removed all pull up resistors on ESP side, the ones we talked previously that are externally mounted on some ESP.
By doing this, we are theoretically in the spec of the chip.
But it requires some soldering and the PCB of the LLC is very cheap => difficult to replace the resistor without breaking the PCB.
[image]

To "perfectly" solve that issue the easiest way is to use two pin of the ESP to handle the data line.
For all other lines (CS, CLK, audio), we can continue to the use the mosfet LLC or replace it by simple mosfet as shown on this schematic. But these other lines work in a single direction, so only one mosfet and one pin is needed.
[image]

@visualapproach What do you think about that ?

But now you may wonder why it work for some people ?
It may work for some people because, we reach 1.5/1.6V on the input of the ESP.
According to ESP datasheet, everything under 0.8V is seen as 0 and everything over 2.5V is seen a 1.
It means that everything between 0.8V and  2.5V is undertermined. Depending on some factors, it may work or not.

And why not using the recommended TXS0108E ?
Finally, it may be better to use TXS0108E instead of mosfet LLC. What I don't like is that Texas recommend to shorten the output wires and not use capacitive load. In out case, we have long wires and filter with capacitors. So, I'm not sure that the chip works correctly.

My old school design with discrete mosfets takes two more pins (one for the CIO and one for the DSP) but looks more robust.

SigmaPic, 2022-07-14:
> [image]

About the pinout of the CIO.

Where is the 5V and the GND ?
Pin 5 = GND
Pin 6 = 5V 

??

SigmaPic, 2022-07-14:
> First I think you've made a fantastic job with the schematics and explanations. Very good! Regarding the double pins per data wire, I abandon it for two reasons; 1 it requires more components and therefore more difficult for most people. Nowadays I think you can get the PCB fabs to populate the PCB but I haven't tried that and it seemed difficult to find the right components. 2 there isn't enough GPIOs on the ESP8266. Not entirely true, but D0 cannot have interrupts and D8 is pulled to GND and boot fails if pulled high. Now, there is RX and TX pins also which I don't want to mess with because it will also make installation harder. However we might pull it off if we use D0 (no irq), D8 (pulled down), D3, D4 (pulled up) and D5 to the DSP since we don't need interrupts there. Leaving D1, D2, D6 and D7 for the CIO. So much for parallel traces :-)
> 

Yes it requires to solder SOT23 chips. That's the price to pay :)
D0 and D8 are perfect, isn't it ?
No need of interrupt on data line, isn't it ?
and D0 can be used as an output for data line with an external pull down resistor if needed.

> There is also the possibility to switch to ESP32 with lots more GPIOs and horsepower to serve web pages quicker. But it consumes more power i guess, which in some cases seem to be a weak spot. And not to mention - I haven't got it to work yet. (Haven't put too much effort in yet but it would take a lot of time recoding, and making new PCB.)

Yes, it will take a lot of effort.

> So, any other options? 

Direct connection.
Maybe we can add a diode from ESP input to VCC so that the ESP is protected.
This resistor will act has a voltage divider with the serial resistors.
We can increase a bit the serial resistor value to reduce the current coming from the CIO.

> Are there LLCs with weaker pullups to buy? For your information, the reason I introduced the TX0108E is because for a while the other one was difficult to source and much more expensive. Now I think there is no real difference.

I don't think you can buy LLC with 50K resistors.

@visualapproach  What is your opinion about @menime8 issue ?
He tried with a direct connection.
It means it's not an issue related to voltgae shifting.

Do you think it's a good idea to print in the console the command received from CIO ?
Every time the CS goes high, you coul print in the console (debug.html) th value of _receivedByte.
This would allow to see if he get some command or not.

visualapproach, 2022-07-14:
> @visualapproach What is your opinion about @menime8 issue ?
He tried with a direct connection.
It means it's not an issue related to voltgae shifting.

I don't know, there may be something more to it in that case. Don't know if he blew the ESP by shorting 5V and 3V3 or if he has a new one now.

> Do you think it's a good idea to print in the console the command received from CIO ?
Every time the CS goes high, you coul print in the console (debug.html) th value of _receivedByte.
This would allow to see if he get some command or not.

As a one off in this case maybe. Otherwise its a fulltime job to try to decode what's wrong, and it is usually hardware fault caused by bad soldering, bad wires or overheated components during soldering. I think first off is to reduce the chance of this by doing a board self test as you suggested. Then if it does not work for several people maybe go that route. But didn't he have an oscilloscope? Or was it someone else? Easiest would be to check out the signals at the ESP with the scope I think.
You should read something like this:
CS.  `----__________________________________------`
CLK:`-----_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-------`
DAT:`-----___-_---_--____--____--_-____---------`

SigmaPic, 2022-07-14:
> I don't know, there may be something more to it in that case. Don't know if he blew the ESP by shorting 5V and 3V3 or if he has a new one now.
>
>
> As a one off in this case maybe. Otherwise its a [...]

Your right, the shortcut may jae broken something.

@visualapproach Do you know if the GND is connected to the ground of the main power ?
In other word, is it possible to connect the GND of the CIO to the mass of the scope without any electrical issue ?
Sometimes, the GND can be referenced to the main power (230V) and connecting the mass of the scope to the GND can provoke a short circuit. No issue here ?

visualapproach, 2022-07-14:
@SigmaPic You are right, I don't know where the GND goes to so maybe a bad idea. Please do not destroy your scopes.

visualapproach, 2022-07-14:
And also @air101 found out that his problem was due to a particular type of NodeMCU. #325

menime8, 2022-07-14:
Yes, it was me with the osci. Tomorrow the other ESP arrives. I another issue the ESP was faulty. I think there are a few fakes on the market. I'm very approved with the soldering iron, also with hot air for SMD. I repair since 2015 car amplifiers. Since then i don't have one failed solderpoint. Also i double checked every connection with the meter. 

Before the bridge 3 to 5 Volt i also tried ever possible constellation. 

I use a USB scope with a laptop! Hantek 6022BE.
So i isolated. 
If the 0V is grounded, the RCD should trigger, i touched it few times while measuring.

Lets see what i can find at the next weekend.

visualapproach, 2022-07-14:
@menime8 Sorry for talking about you in third person. The statement about bad solders was a general experience, not directed at you. I just described my situation where a lot of time searching for code errors often ends with success after mentioned problems had been fixed. I think you are great trying to help people in the forums!

menime8, 2022-07-14:
@visualapproach 

No problem. Your code works, i think this is approved more than 100 times. My connections are fine, too. So i see the issue at the nodeMCU. My Version is 3.2 with the ESP8266 12F, they say its the better WiFi Version. Now i ordered the older 12E.

visualapproach, 2022-07-15:
@SigmaPic I just uploaded a new version to test the setup by connecting the cables together. It runs 1000 edges on each data line, both directions and gives back a report. You need to visit the page "/hwtestinfo.html" to run the test.

SigmaPic, 2022-07-15:
> @SigmaPic I just uploaded a new version to test the setup by connecting the cables together. It runs 1000 edges on each data line, both directions and gives back a report. You need to visit the page "/hwtestinfo.html" to run the test.

Nice !!!
Thanks

@menime8 Do you want to test this software. It may allow you to confirm that's there's is an hardware problem on BWC side

SigmaPic, 2022-07-15:
> So, any other options? Are there LLCs with weaker pullups to buy? For your information, the reason I introduced the TX0108E is because for a while the other one was difficult to source and much more expensive. Now I think there is no real difference.

After thinking a bit more about that topic, I think  I have found "the" easiest solution that fully matches all specifications. I have to check the datasheet again and simulate it but on the paper it works.
And the the most amazing is that it can be implemented ok the PCB V2.
The solution is so easy that I wonder if I missed something.
Details for tomorrow. ;)

visualapproach, 2022-07-15:
Curious:)

SigmaPic, 2022-07-16:
The idea is to use a simple voltage divider to convert the 5V coming from CIO/DSP into 3.3V.
In the other way, no need of conversion since 3.3V is seen as a logical 1 by the CIO/DSP.

Signals look perfect and it could almost be implemented on PCB v2.

[image]

The only issue is the pull up resistor on D3/D4 that make the mess.

These pins shall be replaced by D0/D8 or something else as you suggested it.
That's really annoying, there are not so much pins and 3 of them shall be pulled up or down.

> Not entirely true, but D0 cannot have interrupts and D8 is pulled to GND and boot fails if pulled high.

We could also use D11 and D12, isn't it ?

@visualapproach What do you think ?

It's easier and cheaper. No need of LLC.

Do you have an idea why the DSP is flashing at boot and why adding serial resistor helps ?

visualapproach, 2022-07-16:
> The only issue is the pull up resistor on D3/D4 that make the mess.

I think one of them is for the on board LED. How much do they affect the signal?

> We could also use D11 and D12, isn't it ?

What is that? I'm only aware of D0-D8

> Do you have an idea why the DSP is flashing at boot and why adding serial resistor helps ?

Not really. Obviously random but errors would case things to behave weird. There is 'enable-bits' that would flash the display if toggled, but the strange thing is that it works during start up sequence.

I think it's a nice idea. Would need some mods on the pcb to make it pretty and easy. I wonder if it would work on all other models as well. Possibly the audio will be more silent but it's no good on the 2021 models anyway from what I have heard. What was your setup again @SigmaPic ? Is it working for you?

menime8, 2022-07-16:
> > The only issue is the pull up resistor on D3/D4 that make the mess.
> 
> I think one of them is for the on board LED. How much do they affect the signal?

I deleted the 2 resistors, another idiot idea :-) LED is working. 

Where i can find the new software? I think i´m blind...

My new ESP arrived 2 hours ago.

menime8, 2022-07-16:
[image]

There missing files i need?

visualapproach, 2022-07-16:
@menime8 Newest version is found in the development branch:
[image]

Sometimes one need to clean build files:
[image]

menime8, 2022-07-16:
> /hwtestinfo.html

DSP input pin 2 0 errors of 1000
DSP input pin 3 0 errors of 1000
DSP input pin 4 0 errors of 1000
CIO input pin 2 0 errors of 1000
CIO input pin 3 0 errors of 1000
CIO input pin 4 0 errors of 1000

If i start the test with disconnected plugs i get:

DSP input pin 2 500 errors of 1000
DSP input pin 3 500 errors of 1000
DSP input pin 4 500 errors of 1000
CIO input pin 2 500 errors of 1000
CIO input pin 3 500 errors of 1000
CIO input pin 4 500 errors of 1000

visualapproach, 2022-07-16:
Nice. 500 is = no contact.

menime8, 2022-07-16:
> Nice. 500 is = no contact.

Also test with single broken contacts, works fine. Thank you. 

Lets see what my CIO says

SigmaPic, 2022-07-16:
> I think one of them is for the on board LED. How much do they affect the signal?

You're right, the LED is on D0.
Maybe D11 and D12 is better.

> > We could also use D11 and D12, isn't it ?
> 
> What is that? I'm only aware of D0-D8

D11 and D12 is located on the other side of the esp. Look on the schematic [https://github.com/nodemcu/nodemcu-devkit-v1.0/blob/master/NODEMCU_DEVKIT_V1.0.PDF]()

> > Do you have an idea why the DSP is flashing at boot and why adding serial resistor helps ?
> 
> Not really. Obviously random but errors would case things to behave weird. There is 'enable-bits' that would flash the display if toggled, but the strange thing is that it works during start up sequence.

Sorry, I didn't catch you. What are the 'enable-bits' you talked about?
Can you explain ?

> I think it's a nice idea. Would need some mods on the pcb to make it pretty and easy. I wonder if it would work on all other models as well. 

Yep it depends on the microcontroller of the other models. But it should work I think. I will check again that point.

> Possibly the audio will be more silent but it's no good on the 2021 models anyway from what I have heard. 

What is no good on 2021 model ?
I am interested because I have a 2021.

> What was your setup again @SigmaPic ? Is it working for you?

I have ordered a PCB V2 customized by me with SMD resistors and a jumper to root the audio directly from CIO to DSP. I can share it once it will be tested. I have a model 2021.

SigmaPic, 2022-07-16:
> > /hwtestinfo.html
> 
> DSP input pin 2 0 errors of 1000
> DSP input pin 3 0 errors of 1000
> DSP input pin 4 0 errors of 1000
> CIO input pin 2 0 errors of 1000
> CIO input pin 3 0 errors of 1000
> CIO input pin 4 0 errors of 1000
> 
> If i start the test with disconnected plugs i get:
> 
> DSP input pin 2 500 errors of 1000
> DSP input pin 3 500 errors of 1000
> DSP input pin 4 500 errors of 1000
> CIO input pin 2 500 errors of 1000
> CIO input pin 3 500 errors of 1000
> CIO input pin 4 500 errors of 1000
> 
> AND: with PCBV2 you don´t need 5V, because you´ll get it from your USB-Port. It is easier for the most people.

Dis you try with the old esp ?

SigmaPic, 2022-07-16:
> I deleted the 2 resistors, another idiot idea :-) LED is working. 

If you use a MOSFET LLC, you can remove it because there are some pull ups on the LLC.

menime8, 2022-07-16:
> Dis you try with the old esp ?

With old and new, same the same results.
So now the new one also won't work.

Here i have a few shots with the osci. Attention, different time dividers.
[image]
[image]
[image]

RatAttack83, 2022-07-16:
Interesting news in this topic... So I have disconnectet my spa and tried to get it to work again... with Powerbank and with laptop... it wont work... :( 
I ordered some othe NodeMCU and tried it... same result...
so i can measure with a mobile scope, but my contact is on holiday for the next three weeks and my 230V scope is not the best idea.
But i have read a bit in this topic and think you are on the right way. :) i hope you will get the Model 2021/22 to work :)

If someone give me shematics to test, i can etch it at work and check it at home :) so perhaps i can contribute a little.

SigmaPic, 2022-07-16:
> > Dis you try with the old esp ?
> 
> With old and new, same the same results.
> So now the new one also won't work.
> 
> Here i have a few shots with the osci. Attention, different time dividers.
> [image]
> [image]
> [
> [image]
> ](url)

I think it's chip select, clock and last one data. Isn't it ?
Clearly the last signal is not correct. Edges are too smooth. Too much filtered.
What was your setup? Which LLC ? Resistors ?

menime8, 2022-07-17:
> I think it's chip select, clock and last one data. Isn't it ? Clearly the last signal is not correct. Edges are too smooth. Too much filtered. What was your setup? Which LLC ? Resistors ?

Scoped with original DSP on it, nothing else.

SigmaPic, 2022-07-17:
>  > I think it's chip select, clock and last one data. Isn't it ? Clearly the last signal is not correct. Edges are too smooth. Too much filtered. What was your setup? Which LLC ? Resistors ?
> 
> Scoped with original DSP on it, nothing else.
> 

😳 That's really strange.
It's the first time you look at the signal ?

menime8, 2022-07-17:
> 
Yes. I will now disassemble the display again, to see what R or RL there is.

SigmaPic, 2022-07-17:
I decoded the signal,  it's a command 0x42 0x1B 0x1B

Maybe other people have already looked at the signal and know if the waveform is normal.

It's strange because some edges are Ok and some other edges are smooth.

menime8, 2022-07-17:
[image]
[image]

Shorten the wires, thanks for your hint with that Strange signal. I think the wire acts like a capacitance.

Now i can read the actual temp and thats it. No switch is working.

menime8, 2022-07-17:
> [image] [image]
> 
> Shorten the wires, thanks for your hint with that Strange signal. I think the wire acts like a capacitance.
> 
> Now i can read the actual temp and thats it. No switch is working.

I use now the pre2021. I get data from CIO, but can't send any back.

menime8, 2022-07-17:
> > [image] ![S
> > Shorten the wires, thanks for your hint with that Strange signal. I think the wire acts like a capacitance.
> > Now i can read the actual temp and thats it. No switch is working.
> 
> I use now the pre2021. I get data from CIO, but can't send any back.

SigmaPic, 2022-07-17:
@menime8  Can you monitor the signal on the input of the esp.

If you can do it with the LLC as well. I think we will understand why it doesn't work.

menime8, 2022-07-17:
> @menime8 Can you monitor the signal on the input of the esp.
> 
> If you can do it with the LLC as well. I think we will understand why it doesn't work.

Lets see if i would do this next few days. Everytime disassemble everything and bring the pump inside,  sometimes i would use the pool 😉
And it needs filtered every day.

SigmaPic, 2022-07-17:
What change with pre2021 ?

It's the pinout ?

Can you tell us the correspondence between the scope screenshot and the pin number on the connector ?

visualapproach, 2022-07-17:
Have you tried MIAMI2021? It's for the newer pumps. It uses different codes for the buttons etc. Or maybe the special version 54149E that uses two data lines and a clock.

[116 more replies]

closed 2022-07-21 by menime8