visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #334: MQTT not auto-added to Home Assistant

olskar, 2022-07-16:
Hello! 
I have recently installed this in my pump. I have enabled MQTT but it is not autoadded in home assistant. I have enabled discovery for MQTT and other devices get autoadded, what could be the problem here? 

I can see it sends messages as below:
{"CONTENT":"TIMES","TIME":1657984235,"CLTIME":1657979616,"FTIME":1657979616,"UPTIME":4851,"PUMPTIME":4614,"HEATINGTIME":5,"AIRTIME":25,"JETTIME":0,"KWH":0.059999999,"COST":0.059999999,"FINT":30,"CLINT":14,"TTTT":0}

Thank you!

visualapproach, 2022-07-16:
Config is only sent at startup, so try restart ESP

olskar, 2022-07-16:
How is the config sent supposed to look like?

I have restarted the ESP but can only see this being sent on mqtt and nothing in Home Assistant

{"CONTENT":"STATES","TIME":1657990557,"LCK":0,"PWR":1,"UNT":1,"AIR":0,"GRN":2,"RED":0,"FLT":1,"TGT":20,"TMP":33,"CH1":32,"CH2":32,"CH3":32,"JET":0,"ERR":0,"GOD":false}

or

{"CONTENT":"TIMES","TIME":1657990563,"CLTIME":1657979616,"FTIME":1657979616,"UPTIME":9644,"PUMPTIME":9348,"HEATINGTIME":4687,"AIRTIME":37,"JETTIME":0,"KWH":2.588999987,"COST":2.588999987,"FINT":30,"CLINT":14,"TTTT":0}

visualapproach, 2022-07-16:
It sends once on topics like 'homeassistant/sensor/layzspa...'
Are you sure you don't have a device called layzspa in HA?

olskar, 2022-07-16:
Yes, I'm sure and I dont see anything like that being sent out in the MQTT broker either :(

All it sends are in these topics:
layzspa/Status
layzspa/MAC_Address
layzspa/MQTT_Connect_Count
layzspa/message
layzspa/times

Nothing is sent on homeassistant/#

visualapproach, 2022-07-16:
Post a pic on the web gui page

olskar, 2022-07-16:
[image]

[image]

visualapproach, 2022-07-16:
Oh you have a 4 wire. Auto-discovery is not added yet. And mqtt says undefined. That's either my bad or you haven't connected to mqtt broker in HA.
You can add sensors manually until further. There are some examples in the HA folder.

chunkysteveo, 2022-07-16:
I have auto discovery working on a test version for my 4 wire, but the climate module doesn't load in... it's not quite ready to push yet. The manual sensors work well though!!

olskar, 2022-07-17:
Ah, that explains it, thanks :)

closed 2022-07-17 by olskar