visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #337: Hawaii HydroJet Pro / Square Pump S200102 - No DSP, no control via web

Bischof-Mak, 2022-07-18:
Hello everyone, I'm desperate and hope we can move forward together. I have tested many different constellations and try to list it here.

The following components (currently):

- Lay-Z-Spa Hawaii Hydrojet PRO
- Square Pump with 6-cable (6-wire)
- BOARD BWC V2
- new Nodemcu V3
- TXS0108E 8 Channel Logic Level Bi-directional Converter
- Resistors between pin 3-5 DISP (currently 680 Ohm)
- 10K resistor

Software:

- 6-wire version
- Model defined: MALDIVES2021
- Board defined: PCB_V2

Problem:
When plugging in the pump, the Melody sounds wonderfully in the display. After a short time the display beeps again. The web interface can also be called up wonderfully. Unfortunately, I get no indication on the display. Also in the web interface is no display (DSP). Control is not possible, neither in the web interface nor directly on the display.

What have I already tested everything:

- Short cables at H1/2A and H1/2A.
- With R2-8 resistors with 560 Ohm with cable to H1/2B
- All cables measured for connection.  (8Channel C - DSP and 8Channel C - PUMP)

When connecting without resistors at H1A, H2B you hear a slight click in the display.

I hope someone can help me, so that I can at least make some progress.
I also have various spare components lying around and can also gladly exchange, soldering, etc..
If more information is needed, just ask.
PS: written in german and translated ;-)
[image]
[image]
[image]
[image]

visualapproach, 2022-07-18:
Welcome @Bischof-Mak and thank you for being precise in your description. There is an ongoing discussion about this. No definite conclusions yet, but I hope it will get sorted out, and you are welcome to help with ideas or testing if needed.

SigmaPic, 2022-07-18:
If you take some photos, please take the whole PCB but also some zoom on components so we can read the values and see the routes.

Bischof-Mak, 2022-07-18:
> Do you you have a scope meter ? It could be interesting to monitor the signal firstly between BWC and DSP.
> 
> If you have photos of your CIO and your DSP, it can help as well.
> 
> I recently discovered that DSP chip is open drain and need a pull up. I am thinking about that.
> 
> Photos and scope screenshots can help.

Unfortunately I do not have a scopemeter
I could get hold of one, but then I still have to be able to operate it.

I will take detailed pictures right now and report back soon

Bischof-Mak, 2022-07-18:
so, i hope it will help :-)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SigmaPic, 2022-07-18:
Perfect !!!!!

Your hardware seems similar to #330.

At list display should work. Maybe try with 580 ohms resistors between DSP and ESP. It works for @menime8. Look at #330.

For the communication with CIO, also look at #330. We are discussing about that.

I hope we will progress in next days.

Bischof-Mak, 2022-07-19:
> At list display should work. Maybe try with 580 ohms resistors between DSP and ESP. 

I already tested 560Ohm resistors at 5,6,7 (display 3-5) without success. Do you think I should test 580 ohms? Then I will get some.

SigmaPic, 2022-07-19:
> > At list display should work. Maybe try with 580 ohms resistors between DSP and ESP. 
> 
> I already tested 560Ohm resistors at 5,6,7 (display 3-5) without success. Do you think I should test 580 ohms? Then I will get some.
> 
> And what I do not understand at all, why my display shows nothing at all?

No need to try 580 then.

A couple of other options:
- try to remove the LLC and keep the resistor. There is a small risk to break the esp but according to other people.
- Try with the LLC based on MOSFET (red board)
- Try with the other version of the esp (cp2102) but I don't think it will help.

The best thing to do is to monitor the signals with a scope to see what happens

visualapproach, 2022-07-19:
And also try overclocking the ESP to 160 MHz in case the clock speed is too high for the bit banging to cope. I don't think so but since it appears to be faster from the scope pics posted it's worth trying. Just edit the platformio.ini to release the extra horsepowers

Bischof-Mak, 2022-07-19:
> And also try overclocking the ESP to 160 MHz in case the clock speed is too high for the bit banging to cope. I don't think so but since it appears to be faster from the scope pics posted it's worth trying. Just edit the platformio.ini to release the extra horsepowers

I tested it but there was no change.

Next Step, i try the LLC based on MOSFET....
I will report....

SigmaPic, 2022-07-19:
> I tested it but there was no change.
>
> Next Step, i try the LLC based on MOSFET....
> I will report....

Yes try with MOSFET. It works for your pump according to #340

Bischof-Mak, 2022-07-19:
sorry für die doofe nachfrage:

kannst du mir genauer erklären was ich machen soll ??

visualapproach, 2022-07-19:
You are not connected to the esp. Did you enter the Wi-Fi credentials? Is your ESP connected to Wi-Fi? Are you on the same network?

Bischof-Mak, 2022-07-19:
I am with the browser on the web interface.
you can see it on the left picture....

visualapproach, 2022-07-19:
Yes and you can also see 'web socket connection failed' and the red bar in the title. Websockets don't work on different networks and usually not on guest networks either.

Bischof-Mak, 2022-07-19:
hmmm, then I have a problem of a different kind?

I come in any case via WLAN on the ESP and this also gets an address from my network......

Bischof-Mak, 2022-07-19:
I will now try to reprogram the ESP and get back to you!

Thanks for your patience and help

visualapproach, 2022-07-19:
No worries, now you know what the problem is and can investigate it

Bischof-Mak, 2022-07-20:
Shortly again a deviate from the actual problem:

I would need help again please, because I despair of the software....

I used Visual Studio Code.
Set up a new installation, installed VSC, installed platformio, opened folder 6-wire and uploaded with the arrow....

After that my module was sparking, I connected and entered my SSID and password. After that the ESP connected to my network. I then wanted to call me the web interface, unfortunately it shows me nothing.

My ESP is connected via COM and in the serial monitor it shows the following:
 IP: "192.168.200.24"
OTA > ready
HTTP > server started
WebSocket > server started
Failed to read mqtt.json. Using defaults.
MQTT > buffer size successfully increased
End of setup()
HTTP > request: /
HTTP > file not found: /index.html

It seems that it is not uploading the DATA folder for me?!
How can I do this or how must this be?
Can someone please help me...

[image]
[image]

SigmaPic, 2022-07-20:
You missed that 

[image]

Bischof-Mak, 2022-07-20:
> You missed that
> 
> [image]

thank you I read that 100 times and didn't understand....
But now found and fixed....
here a link for others like me :-)
https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/199

Back to other problem.... i will report...

SigmaPic, 2022-07-20:
Regarding the startup issue you seems to have as well (esp need to be reboot to start). Check last posts of #330.

There is a new software on dev branch with a new pinout that is easy to solder.

Bischof-Mak, 2022-07-20:
I was able to install everything perfectly. Everything could be controlled and read via display and web interface.
(Only the problem with the startup was present but as described).

After that I went to tackle the last problem. I noticed the post https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/330 with the new solution.

1. i soldered as described the ESP --> MOSFET --> CIO / DSP. additionally 25v10uF at 5V and GND.
2. i downloaded under "development" the new code. 
3. have adapted the code under model.h: PCB_V2B
4. uploaded new code.
5. 
[image]

Installed and gave power to the pump. ESP comes up and display melody plays.
Unfortunately, the display shows nothing and control via web interface is also not possible.

Nevertheless, I am extremely excited about what you do and can.

Bischof-Mak, 2022-07-20:
Another question:
After all, this discussion can be closed? The actual problem (as described in the headline) is fixed.
Only the "start - problem" is now the case.

Gladly we can treat this in the post (https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/330).

Do we want to do it that way?

visualapproach, 2022-07-20:
You may close this and continue in #330 pls

SigmaPic, 2022-07-20:
> I was able to install everything perfectly. Everything could be controlled and read via display and web interface. (Only the problem with the startup was present but as described).
> 
> After that I went to tackle the last problem. I noticed the post #330 with the new solution.
> 
> 1. i soldered as described the ESP --> MOSFET --> CIO / DSP. additionally 25v10uF at 5V and GND.
> 2. i downloaded under "development" the new code.
> 3. have adapted the code under model.h: PCB_V2B
> 4. uploaded new code.
> 
> [image]
> 
> Installed and gave power to the pump. ESP comes up and display melody plays. Unfortunately, the display shows nothing and control via web interface is also not possible.
> 
> Nevertheless, I am extremely excited about what you do and can.

Several questions:

When you say : everything could be controlled and read via display and web interface.
What solved the issue, the capacitor or the mosfet LLC ?
Can you discriminate, so we know what was the problem.

With the new software (PCB_V2B), you have to solder the  wire differently.
I don't see it correcty on your photo.
Can you take a photo where we see the connectors ?

You should have something like that. The swap can be done between the LLC and ESP as well:
[image]

Bischof-Mak, 2022-07-20:
> When you say : everything could be controlled and read via display and web interface.
> What solved the issue, the capacitor or the mosfet LLC ?
> Can you discriminate, so we know what was the problem.

1. the problem was solved by the MOSFET without resistors (everything could be controlled. Only the start problem of the ESP was).

> 
> With the new software (PCB_V2B), you have to solder the wire differently.
> I don't see it correcty on your photo.
> Can you take a photo where we see the connectors ?

2. now I just checked again exactly how the wiring is and it looks exactly like in your picture (PCB_V2B). I have made a photo, but it is simply difficult to recognize the wiring on it.
But i think and hope it is correct ;-)
[image]

visualapproach, 2022-07-20:
I only have limited time for now and starting to loose track of all posts. We should make a list in #330 like:

- [ ] user | model | setup LLC/MCU etc | problem | 
- [ ] user | model | setup LLC/MCU etc | problem | 
- [ ] user | model | setup LLC/MCU etc | problem | 

...
and see if we can spot something in common.

menime8, 2022-07-20:
> I only have limited time for now and starting to loose track of all posts. We should make a list in #330 like:
>
> * [ ]  user | model | setup LLC/MCU etc | problem |
> * [ ]  user | model | setup [...]

Good Idea.
Think we should start a fresh topic and Close the others.

SigmaPic, 2022-07-20:
Yes good idea. I think a new post is the best. Maybe we can wait that @Bischof-Mak check its setup. And also that @menime8 test the new pinout tomorrow.

Then we can start this new post

SigmaPic, 2022-07-20:
@Bischof-Mak another question, you have built and flashed the software from the development branch, not from master.

Do you confirm?

Bischof-Mak, 2022-07-21:
> It looks correct. Can you check that you have no short-circuit with a multi-meter ?
> 
> It's really strange because we just changed the pinout. The only difference is that pull up present on D3 and D4 are now on DSP signals.
> 
> We missed something. It should work. @visualapproach an idea ?
> 
> It's strange, neither display work, nor the CIO.

I just measured every connection with the multimeter and checked if there is a short circuit.

Everything fits.

I should get today again PINs to plug. Then I build a second ESP with MOSFET on my BWV v2 and solder according to the new plan (PCB_V2B).

I will get back to you as soon as I can.

Bischof-Mak, 2022-07-21:
> @Bischof-Mak another question, you have built and flashed the software from the development branch, not from master.
> 
> Do you confirm?

Yes, DEV BRACH

That what is new on it, I have seen, because I have rebuilt me in this course the HTML files in German.

But I do not know any way that I can look after programming, on the ESP directly what is on it.

visualapproach, 2022-07-21:
[image]

Bischof-Mak, 2022-07-21:
Unfortunately, I was shocked to discover that I did not have the correct software on the ESP.

Now try again to upload the DEV Brach....

Am now but also fail, will report here. https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/330
i have the same error like @menime8

      |  ^~~~~~~
In file included from lib\BWC/BWC_common.h:22,
                 from src\main.h:5,
                 from src\main.cpp:1:
lib\BWC/BWC_6w_type1.h:11:14: error: #elif with no expression
   11 | #elif PCB_V2B
      |              ^

Bischof-Mak, 2022-07-21:
You professionals have made it !!!!!

now I managed to load the DEV_Branch.

Module connected, power plugged in, everything boots up, everything can be operated (display and web interface).

Well done.