visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #340: S200102 Pump - Maledives with Hydrojet / No reaction after power supply

MK89OH, 2022-07-19:
Hello,
 
This is really a great project and it's really fun to work with it.
I have to buy a coffee here too.
 
So now I come to my problem and I hope you can help me.
 
Lay-Z-Spa - Maledives with Hydrojet (S200102)

Hardware:
Board V2
ESP 8266 V2 / ESP 8266 V3 12F / ESP 8266 V3 12E (all have been tested, same symptoms)
TXS0108E 8 Channel Logic Level Bi-directional Converter (didn't work at all) / LLC – 8 Kanal – the red one (it works)
Without resistors
 
Software:
6-wire version
Model defined: MALDIVES2021
Board defined: PCB_V2
 
My problem:
The module itself works, but only if you disconnect and reconnect the 6-pin connector between the mainboard and the module with the pump switched on, then the ESP wakes up, so to speak.
However, if the pump is now completely disconnected from the power supply and the pump is connected to the power supply again, the module will not start again. The module only starts again by briefly disconnecting the 6-pin plug between the mainboard and the module, then everything works again.
 
A bit strange.
Has anyone here had the same effect and is there a solution for this?
Many thanks.

SigmaPic, 2022-07-19:
Maybe it's linked to the boot pin of the esp.
Some pins need to be pulled up at boot to boot correctly.

[image]

menime8, 2022-07-19:
Tried with the 10k resistor?

You have a small capacitor? Connect it between +5V and GND. Like a 10V10uf.

SigmaPic, 2022-07-19:
> Tried with the 10k resistor?
> 
> You have a small capacitor? Connect it between +5V and GND. Like a 10V10uf.

10k is useless with MOSFET LLC.

Maybe the cap can helps.

Also you can try with serial resistors as well.

Do you have a scope ?

MK89OH, 2022-07-19:
Thank you for the suggestions.
In your opinion, would a 25V1000uF capacitor work as well?
Otherwise I would have to get a 10V10uF for testing.

menime8, 2022-07-19:
> Thank you for the suggestions. In your opinion, would a 25V1000uF capacitor work as well? Otherwise I would have to get a 10V10uF for testing.

1000uf is very much. I don't know what the powersource make, its like a shortcircuit. Maybe it makes your Problem bigger

Bischof-Mak, 2022-07-19:
ahh ok, but you doesn`t have hydros right?

and the startproblems i have too. Wenn i connect the pump the ESP won`t AND the Display comes not up. When i press the reset on ESP, then starts everything. Yours too?

I have a 25v10uF can i use it and when yes where i have it to install?

visualapproach, 2022-07-19:
Across 5V and GND. Make sure minus goes to GND

MK89OH, 2022-07-19:
Thank you for your ideas 😃 .
I have order some 10v10uF capacitor, then i can test it...

SigmaPic, 2022-07-20:
Check last posts in #330.
There is a new software on the dev branch with a new pinout that may solve the startup issue.
It's easy to try

SigmaPic, 2022-07-21:
@MK89OH have you tried the new pinout?

MK89OH, 2022-07-21:
> @MK89OH have you tried the new pinout?

Yes yes, it is perfect 😃 .
Now all my ESP's works with this pinout. Some ESP's had by the old pinout problems with the start process...
I have testet it now by my Pump #54123 (PRE2021) and by my S100101 (MIAMI2021) and it works fine.
Only by the S200102 Pump i must test it in the next days.

Very good WORK 🤩

SigmaPic, 2022-07-21:
> Yes yes, it is perfect 😃 .
> Now all my ESP's works with this pinout. Some ESP's had by the old pinout problems with the start process...
> I have testet it now by my Pump #54123 (PRE2021) and by my [...]

Ok perfect!

Let us know if it works with S200102 as well.

Bischof-Mak, 2022-07-22:
> Ok perfect!
>
> Let us know if it works with S200102 as well.

My Pump is a S200102 too and it work perfekt ;-)

MK89OH, 2022-07-22:
> My Pump is a S200102 too and it work perfekt ;-)

Hello, thank you for th information. With or without cap?

Bischof-Mak, 2022-07-22:
> Hello, thank you for th information. With or without cap?

With AND without !! i tried both variations

MK89OH, 2022-07-22:
> With AND without !! i tried both variations

DANKE. I test it too and give feedback...

MK89OH, 2022-08-06:
So, i test it without cap, and it works perfekt.
I think i can close this issue... 😃

closed 2022-08-06 by MK89OH