visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #342: Home Assistant and MQTT

SchallertDK, 2022-07-20:
Hi,
I cant get my MQTT Broker to automatic find the Spa module.
Everything's seems to work. The spa modul is connected to MQTT:
[image]

And the Home Assistant receive data from the spa module:
[image]

But the broker wont find it? And i cant get the lovelace as showed in your "Build-instructions-Bestway-WiFi-remote.pdf":
[image]

[image]

What am i missing? 2 peoples has tried to help me, but we cant find the "error"?

chunkysteveo, 2022-07-20:
are you on a 4-wire pump? Auto discovery isn't implemented yet on the 4-wire firmware...

SchallertDK, 2022-07-20:
Yes i have.
How do i so do? :)

chunkysteveo, 2022-07-20:
You have to manually add in the sensors at the moment - using the code and help in the repo https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/tree/master/Code/Home%20assistant%20config

SchallertDK, 2022-07-20:
Thank you,
I will try to set it up. But i'm not an expert at this HA yet :)

chunkysteveo, 2022-07-20:
Ohh i may test! My efforts worked, but the climate wouldn't import for me.

Will try the new dev branch......

visualapproach, 2022-07-20:
> Ohh i may test! My efforts worked, but the climate wouldn't import for me.
> 
> 
> 
> Will try the new dev branch......

I had some trouble too. Turned out the message was too long for the ESP to handle so don't use long base names.

chunkysteveo, 2022-07-20:
Ahhh - i was getting there.... cutting down the packaged worked. Good to know I was on the right track!

visualapproach, 2022-07-20:
Alright, will handle it when !AFK

chunkysteveo, 2022-07-20:
Auto discovery worked really well - but the God Mode switch didn't import into HA?

SchallertDK, 2022-07-21:
My HA went in "Safe Mode" after trying adding new "text"

I made a file named "layzspa.yaml" into the folder "package" and pasted everything from the layzspa.yaml file at github + removed the # at the 2 areas where it only was for the 4-wired pump.

In the configuration.yaml i added:
[image]

In the automations.yaml i added what was in the "automations.yaml" file at github

About the two helpers i need to make, i cant find any helpers who can take the infos from the github file?
[image]

[image]

I to get out of "safe mode" at the HA i had to delete 
"homeassistant:
  packages: !include_dir_named packages" from the configuration.yaml

So i'm back to start again and dont know what i have done wrong

chunkysteveo, 2022-07-21:
If you can download the "Development" branch of this project - HA Autodiscover is now in place and does work for me - adds in all the required sensors (apart from god mode switch?). Maybe try the development branch - treat it like a new project - so you will need to set your model number etc and upload again. But then at least HA will/should work.

Alternatively you can just add in the sensors and switches and climate in the configuration.yaml file in their appropriate places.

I've not used packages in HA, so can't advise. But the helpers - one is a Number helper, and the other is a "Toggle" helper.

SchallertDK, 2022-07-21:
I had now downloaded the development zip file, opened like a new project, change the parameters so it is matching my pump and ip. Build image file, uploade to the spa module. everything with "succes".
But i stil dont se any activity at the MQTT in HA.
The Spa module is connected with MQTT an no fails there.
About this GOD MODE you write about, where do i manually enable that? I cant find it in the PlatformIO file?

chunkysteveo, 2022-07-21:
Hmm, not sure if it's not working for you? If you go to Integrations, MQTT it should be there as a new device with entities?

The god mode switch is the "Take Control" switch which allows us to control the 4-wire pumps. It was/is called god mode in the code in parts(!).

SchallertDK, 2022-07-21:
Ok, the Take Control bottom is at the Spa "website" - So i think it is ok?
The MQTT is still not reacting for any new device?
[image]

I have restartet the HA and the Spa ESP 3 times, with no respons.

When i'm typing # at the configuration in the MQTT it shows the communication to the spa module:
[image]

chunkysteveo, 2022-07-21:
Is discovery enabled in Mosquito? If you click on re-configure MQTT and check it is? and the topic is homeassistant/status ?
[image]

SchallertDK, 2022-07-21:
Yes it is:
[image]

chunkysteveo, 2022-07-21:
hmm, i'm not 100% sure what the issue is, sorry?

You can still try to add the switches and sensors manually into HA using configuration.yaml

chunkysteveo, 2022-07-21:
Try adding the below in `configuration.yaml `where you have `switch:` (you can't have more than one `switch:` declaration, so look for it and add below it, or copy and paste all of the below if there is no `switch:`)
```
switch:
  - platform: mqtt
    name: "Hot Tub Bubbles"
    icon: "mdi:chart-bubble"
    state_topic: "layzspa/message"
    command_topic: "layzspa/command"
    value_template: "{{ value_json.AIR }}"
    payload_on: '{"CMD":2,"VALUE":1,"XTIME":0,"INTERVAL":0}'
    payload_off: '{"CMD":2,"VALUE":0,"XTIME":0,"INTERVAL":0}'
    state_on: 1
    state_off: 0
    availability_topic: "layzspa/Status"
    payload_available: "Alive"
    payload_not_available: "Dead"      
```

SchallertDK, 2022-07-21:
This text i have to replace in the configuration.yaml, where the "switch" for air bubbles is right? In yoour switch it is named hot tub bubbles

chunkysteveo, 2022-07-21:
Yeah, mine is called Hot Tub Bubbles, you can call it what you like.

SchallertDK, 2022-07-21:
Right now my HA is down, and i cant reach it. I tried to make those 2 helpers, and then it went down.

Next question is: the 2 areas with "#" , i have to delete the # at both areas right?

When i try to make that "layzspa.yaml" file in the "package" folder, then i have to add this in the configuration.yaml, and remove the "#", right? :
#homeassistant:
#  packages: !include_dir_named packages

But when i do this, the HA says it is wrong and, when i delete it, the HA is happy again.
Right now i have it all in the configuration.yaml - But it is right?

chunkysteveo, 2022-07-21:
I'm not sure on using the packages way - mine was all just added to the main configuration.yaml file in one go. Editing yaml is tricky to get the indents and format right.

SchallertDK, 2022-07-21:
Okay, then i will do the same.
What about this: ? Did you add this in the configuration.yaml in the bottom after all the other stuff?
[image]

chunkysteveo, 2022-07-21:
No, I have them added individually - my HA is a large setup that's been worked on for years now. I would recommend reading up or watching some tutorials on how to add custom switches and sensors to yaml.

SchallertDK, 2022-07-21:
Thank you for your help today, i will watch som tutorials :)

visualapproach, 2022-07-21:
I fixed a minor thing in HA config for the 4w now. Best result is when you do it in this order:
1 - Update ESP
2 - delete the old device from HA:
[image]
3 - restart ESP
4 - look at the device in HA again. Refresh page with F5 if it's not populated yet.

visualapproach, 2022-07-21:
And for now, use max 14 chars in base topic.

SchallertDK, 2022-07-21:
I couldn't remove the old device as you shows in the screendump, because i never got the chance to find it.
I have uploaded the last version to the ESP
I have manually removed all text regarding the spa in the configuration.yaml
Restartet the ESP and refreshing (F5) the HA multiply times, but nothing is popping up?

What are the "max 14 chars in base topic"?

visualapproach, 2022-07-22:
[image]

SchallertDK, 2022-07-22:
is the Client ID and Base Topic: layzspa2 - in the instructions is the name "layzspa"
I have try setting the Telemetry Interval from 600 to 5, but no changes

visualapproach, 2022-07-22:
I don't understand what you're asking. Base topic is whatever you type in the box. And you should not type more than 14 characters for now.
Changing the telemetry interval will, well, change the interval between mqtt messages

SchallertDK, 2022-07-22:
My spa module says it is connected with MQTT
My HA gets data from the spa module
But i cant make lovelace or get the lovelace as shown in the instructions pages.
When i manually import node-red flow, and set up af lovelace with that, there is no communication between lovelace and spa module. But it shows in node-red that it is connected and running:
[image]

visualapproach, 2022-07-22:
You don't need node red to get a dashboard. Disable that flow, I think it just messes things up unless you have a good reason to use it. Then restart HA then restart ESP. Go to devices in the configure page.

SchallertDK, 2022-07-22:
Still no sign of new device in the MQTT :(
[image]

visualapproach, 2022-07-22:
Judging from your picture you are looking at the wrong place. Click on DEVICES in the top menu

SchallertDK, 2022-07-22:
There is nothing called layzspa, nothing at al even with the word "spa" (?)

visualapproach, 2022-07-22:
You need v3.4.0 of the firmware

SchallertDK, 2022-07-22:
Now you lost me again :)
Firmware to the esp (spa module?)

SchallertDK, 2022-07-22:
But the firmware at the spa module says:
[image]

chunkysteveo, 2022-07-22:
It will be very similar, if not the same as a dev branch. I am just testing the new release now.

SchallertDK, 2022-07-22:
Is there any where else i can check the it is the rigt version? I tried to search in the Visual Studio Code for "firmware" but couldn't find any "version" number

visualapproach, 2022-07-22:
> But the firmware at the spa module says:
> 
> [image]
> 
> 

Meaning you didn't upload it properly because then it would say 2022-07-21

chunkysteveo, 2022-07-22:
The latest release is the latest, which includes HA Auto discovery. I have just downloaded it, set it up on my pump and it's connection to MQTT, HA, and pulled in as a device with all sensors. The firmware/code works, but there is an issue with your setup somewhere?

Have you set the model number and PCB version in model.h file. Then upload and set MQTT etc. IT should then work with autodisovery...?

SchallertDK, 2022-07-22:
I thought i did everything after the book. But maybe my PCB is wrong?
I bought the one descripted in the instruction
[image]

The pump model number is right. #54123

SchallertDK, 2022-07-22:
[image]

SchallertDK, 2022-07-22:
Ohhh... I can see it maybe is the wrong PCB i have chosen. I dont have round corners at mine :/

chunkysteveo, 2022-07-22:
#define PCB_V1, comment out #define PCB_V2

SchallertDK, 2022-07-22:
I really thought we had it now - But still no device to found

My setup now:
[image]

Spa "display":
[image]

chunkysteveo, 2022-07-22:
anything in your Mos logs?? - /hassio/addon/core_mosquitto/logs

SchallertDK, 2022-07-22:
What about the purple text in the bottom at the coding's?
Should it be undefined with "//" in front?
[image]

SchallertDK, 2022-07-22:
How/where do i define that model number?

chunkysteveo, 2022-07-22:
have you uploaded a new file system too? I see you have no AIR/Bubbles in the GUI, is that right??

Ignore the text you have circled, it won't trigger.

chunkysteveo, 2022-07-22:
Build filesystem image, and then Upload filesystem image?

SchallertDK, 2022-07-22:
a new file system?
I downloaded the code - opened with visual studio code - change the pump model and the PC - set up the ip adress - build file system image (with succes) - upload Filesystem image (with succes)

At the spa module i set up again the MQTT and restart the ESP.

After that i restart the HA

SchallertDK, 2022-07-22:
Build filesystem image:
 *  Executing task: C:\Users\admin\.platformio\penv\Scripts\platformio.exe run --target buildfs --environment nodemcuv2 

Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (3.2.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 1.200.0 (2.0)
 - toolchain-xtensa @ 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.19.4
|-- ESPDateTime @ 1.0.4
|-- WebSockets @ 2.3.7
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|   |-- ESPAsyncTCP @ 1.2.2
|   |-- Ethernet @ 2.0.0
|   |   |-- SPI @ 1.0
|   |-- SPI @ 1.0
|-- PubSubClient @ 2.8.0
|-- WiFiManager @ 0.16.0
|   |-- DNSServer @ 1.1.1
|   |   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266WebServer @ 1.0
|   |   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- EspSoftwareSerial @ 6.16.1
|-- ESPAsyncTCP @ 1.2.2
|-- ArduinoOTA @ 1.0
|   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266mDNS @ 1.2
|   |   |-- ESP8266WiFi @ 1.0
|-- BWC4W
|   |-- ArduinoJson @ 6.19.4
|   |-- ESPDateTime @ 1.0.4
|   |-- LittleFS @ 0.1.0
|   |-- EspSoftwareSerial @ 6.16.1
|   |-- Ticker @ 1.0
|-- DNSServer @ 1.1.1
|   |-- ESP8266WiFi @ 1.0
|-- ESP8266WebServer @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- ESP8266WiFi @ 1.0
|-- LittleFS @ 0.1.0
|-- Ticker @ 1.0
Building in release mode
Building file system image from 'data' directory to .pio\build\nodemcuv2\littlefs.bin
/config.html
/debug.html
/favicon.png
/function.js
/index.html
/main.css
/manifest.json
/menu.png
/mqtt.html
/remove.html
/success.html
/upload.html
/visualapproach.png
/WebSocket.js
/wifi.html
================================================================================= [SUCCESS] Took 4.15 seconds ================================================================================= *  Terminal will be reused by tasks, press any key to close it.

chunkysteveo, 2022-07-22:
have you uploaded the new image? Clicked the "Upload" button in Platform.io? Right arrow in the small blue bar at the bottom?

SchallertDK, 2022-07-22:
No :/
I thought i was uploading by the "upload filesystem image"
But damn. Now it works
Thank you so much

chunkysteveo, 2022-07-22:
RTFM - as some would say!! lol

So it's now pulling into HA as a device and entities are present for you to add as sensors in the lovelace UI?

(Building and uploading the filesystem is only updating the HTML website part of the device. The actual running code is uploaded separately using the Build/Upload command in Platform.io).

SchallertDK, 2022-07-22:
Yes, everything seems to communicate :)
Is it possible to make the pump planning in HA, just like you can do in the "spa config" at the browser?

chunkysteveo, 2022-07-22:
Awesome news!!

It could be possible with some automations - I suspect your knowledge in Nodered may be better for scheduling things? I know in HA Automations there are date and time triggers, so you could set things up - it would be quite complex I suspect?! See if anyone has done any "Climate/Thermostat Scheduling" and use that as inspiration to set the Spa climate...?

[1 more reply]

closed 2022-07-22 by SchallertDK