visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #352: E02 leads to wonky temperatures

Skeletorjus, 2022-07-22:
After receiving an E02 I diacovered that the temperature was reporters as zero. Ideally this should be "unavailable" since the real value is unavailable for real, and in addition it won't mess up those pretty graphs... :)

Also, after restarting the pump the virtual temperature fell to around -100 before stabilizing at the correct value after about 20 minutes.

I've tried to capture how it went down in this screenshot:

[image]

visualapproach, 2022-07-23:
Okay. It sets the virtual temp to -99 at startup to indicate that it needs to be updated. It shouldn't show through mqtt though. I saw the same after the update but not the following reboots.
Can you confirm that it is like this when you make a soft reboot?

visualapproach, 2022-07-23:
Disregard, I have the same behavior now so that will be fixed in a coming update

Skeletorjus, 2022-07-24:
Thanks 👍
Forgot to mention that target temperature, both C and F, is reported as 100 after the restart.

[image]

visualapproach, 2022-07-24:
Thank you! The virtual temp should be fixed in latest dev branch. Need to take a look at target temp. Temperatures are reported both in C and F. In HA they will look the same because HA is converting everything to your preferred unit. Redundant sensors but it may be useful for other home automation systems.

air101, 2022-07-25:
Is it possible to reset the pump from e02 via command line? or does it need a power supply restart?

visualapproach, 2022-07-25:
It needs power cycle. Better to mitigate the cause of the error. Either your pipes or filter is clogged, or your impeller axle is worn. There is replacement kits to buy.

visualapproach, 2022-07-25:
@Skeletorjus Problem should be solved in newest dev code

Skeletorjus, 2022-07-25:
> @Skeletorjus Problem should be solved in newest dev code

Thanks a lot! Installed and will see how it goes. Seems promising so far.
A thing I just noticed is that when the automation to update the ambient temperature ran after the restart, I got -7 back from the sensor:
The value sent was 17.
Also saw the same yesterday after an E02.
[image]

visualapproach, 2022-07-25:
Oh yes, I had to make two commands for set ambient: 14: set ambient F, 15 set ambient C

Skeletorjus, 2022-07-25:
Oh, right, that explains that. Fixed my automation and it's correct again.

closed 2022-07-29 by visualapproach