visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #353: No wifi AP

SigmaPic, 2022-07-23:
Hello,

This time I need some help.
My ESP was working wonderfully.
I tried the OTA update by changing the config in platformio.ini and uploading the binary via OTA.

[image]

The upload succeed but then it was not possible any more to connect to the webserver.

I clean all, erased the flash, rebuild everything on the last release without any modification, reupload thorugh USB but I don't have the AP to configure the wifi.

I tried to upload another demo project that create an acess point and I can see it.

Do I miss something ?

visualapproach, 2022-07-23:
Hi! The ESP Access Point should really only be needed once. The Wi-Fi settings persists OTA updates.

SigmaPic, 2022-07-23:
Even after a full chip erase ?

In the console I have that in loop

[image]

SigmaPic, 2022-07-23:
OK. I success to recover it.

I tried many things. I even tried to upload a specific binary that perform a full erase of the flash. But that didn't solve the issue.

The last thing I tried was:
1. copy the code of my demo project that works well inside BWC projet
2. build and upload
3. the demo code didn't work
4. tried to clean all and rebuild
5. works

Finally a clean all works.

I will retry OTA

visualapproach, 2022-07-23:
I wrote a lot here but you beat me to it. Clean all is a good action when things misbehave! And a computer restart.
If wiping flash on the ESP you probably wipe the Wi-Fi creds also. But otherwise you need to actively delete that data once it in there.

visualapproach, 2022-07-23:
Try to reboot the ESP after a good install when still connected to USB. 8266 forums mentions that as a good thing...

visualapproach, 2022-07-23:
and maybe also after first OTA

SigmaPic, 2022-07-23:
> I wrote a lot here but you beat me to it. Clean all is a good action when things misbehave! And a computer restart. If wiping flash on the ESP you probably wipe the Wi-Fi creds also. But otherwise you need to actively delete that data once it in there.

What do you mean by "you need to actively delete that data once it in there." ?

> Try to reboot the ESP after a good install when still connected to USB. 8266 forums mentions that as a good thing...
> and maybe also after first OTA

I tried

I reproduce the issue.
This is the console output when I upload in OTA.
It success but then it's no more possible to access the webserver.

```
Uploading .pio\build\nodemcuv2\firmware.bin
12:10:58 [DEBUG]: Options: {'esp_ip': 'layzspa.local', 'host_ip': '0.0.0.0', 'esp_port': 8266, 'host_port': 55674, 'auth': 'esp8266', 'image': '.pio\\build\\nodemcuv2\\firmware.bin', 'spiffs': False, 'debug': True, 'progress': True}
12:10:58 [INFO]: Starting on 0.0.0.0:55674
12:10:58 [INFO]: Upload size: 498016
12:10:58 [INFO]: Sending invitation to: layzspa.local
Authenticating...OK
12:11:03 [INFO]: Waiting for device...

Uploading: [                                                            ] 0% 
...
Uploading: [============================================================] 100% Done...

12:11:11 [INFO]: Waiting for result...
12:11:12 [INFO]: Result: OK
========================================================== [SUCCESS] Took 50.83 seconds ==========================================================
```

I think that my ESP reboot in loop when this is printed in loop in the console, right ?

[image]

Do I need to do something else that changing that ?

[image]

SigmaPic, 2022-07-23:
I tried a few more OTA and now even the "clean all" doesn't recover the ESP...

visualapproach, 2022-07-23:
You need to get it working with USB first. And after I commented out this line I didn't get this behavior:
`;upload_speed = 921600`
Did you OTA upload with the upload speed set?

SigmaPic, 2022-07-23:
> You need to get it working with USB first. 
For the moment I don't get it work with USB. Other projects work but not this one.

> And after I commented out this line I didn't get this behavior: `;upload_speed = 921600` 
What behavior do you mean ?

> Did you OTA upload with the upload speed set?
No

SigmaPic, 2022-07-23:
By uncommenting this line, I work now through USB:
`upload_speed = 115200`

visualapproach, 2022-07-23:
I mean the behavior that the ESP reboots over and over after a OTA.

visualapproach, 2022-07-23:
Good, now try a OTA with ESP connected to USB. After success and you see it works in the terminal- press the reboot button

visualapproach, 2022-07-23:
Super long wifi password or with strange characters may cause problems from what I've read.

SigmaPic, 2022-07-23:
> Good, now try a OTA with ESP connected to USB. After success and you see it works in the terminal- press the reboot button

That's exactly what I did.

Does the "upload_speed" may explain why it reboot in loop even whith USB update ?

visualapproach, 2022-07-23:
Don't know. Strange behavior. Do you have another MCU to test on? Or another USB cable?

SigmaPic, 2022-07-23:
Can try with an other USB but it doesn't explain why OTA doesn't work.
I just retried and get sale result, after OTA, the blue flash every 500ms (boot in loop)

visualapproach, 2022-07-23:
In pump or on desk?

visualapproach, 2022-07-23:
Sometimes it helps to edit the code a bit. Just something trivial

visualapproach, 2022-07-23:
If it's only when in pump the error occurs: Is any pins pulled down so it resets or try to boot in programming mode?

SigmaPic, 2022-07-23:
It's on the desk.
I have nothing connected to the esp.
I will try to modify a bit the SW.

I will also try to make it work again through USB and once it works, I will backup the binary. Then I will try the OTA and then upload the backup directly with esp tools to check if it's an issue of build.

visualapproach, 2022-07-23:
Yes, or you can upload the dev branch that I just updated a little to use less memory.

visualapproach, 2022-07-23:
Just throwing up suggestions here, without any reasons

Update vscode and platformio.
Try from another computer (way back someone had trouble and it turned out to be his computer that was causing it)
Upload an OTA sketch other than this. Upload some arbitrary sketch via OTA. Works? Then the esp is probably alright. But if there is an issue with flash mem it's a bit random. 
There may also be memory issues caused by code, like (but not just) writing beyond an array's last element. Random stuff will happen. Or not, depending on luck.

If the problem persists we can see if we can trace it in the code with some tools but since it works for me and presumably others, I think there is something in your setup.

Can you post all of platformio.ini

You don't deserve this after all your help 🥲

SigmaPic, 2022-07-24:
I progressed.

I feel like that the produced binary change between two builds. And sometimes the produced binary work, sometimes not. An even on USB it's the same behavior because after an OTA that fail, the USB upload doesn't work every time.

I tried to upload through OTA with a working binary produced from an USB build and it works fine.
`"C:\Users\SigmaPic\.platformio\penv\Scripts\python.exe" "C:\Users\SigmaPic\.platformio\packages\framework-arduinoespressif8266\tools\espota.py" --auth=esp8266 --debug --progress -i layzspa.local -f firmware.bin`

Every time I flash this binary from USB or OTA, it works.

Conclusion, I think my ESP works correctly.

> Just throwing up suggestions here, without any reasons
> 
> Update vscode and platformio. 

Done.

> Try from another computer (way back someone had trouble and it turned out to be his computer that was causing it) 

I'm runing on Win 7 which is maybe not recommended.
I can try on a VM with Win 10.

> Upload an OTA sketch other than this. Upload some arbitrary sketch via OTA. Works? Then the esp is probably alright. But if there is an issue with flash mem it's a bit random. There may also be [...]

Clearly the produced binary with OTA config is the cause.

> Can you post all of platformio.ini

Here is my config for OTA:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
src_dir = src
data_dir = data

[env:nodemcuv2]
platform = espressif8266@^3
board = nodemcuv2
framework = arduino
lib_deps = 
  bblanchon/ArduinoJson
  mcxiaoke/ESPDateTime
  links2004/WebSockets@^2.3.3
  knolleary/PubSubClient@^2.8
  tzapu/WiFiManager@^0.16.0
	me-no-dev/ESPAsyncTCP
board_build.filesystem = littlefs
monitor_speed = 115200
; Set upload speed to 115200 if you get transfer errors
upload_speed = 115200
; board_build.f_cpu = 160000000L
build_flags =
    -DWEBSOCKETS_NETWORK_TYPE="NETWORK_ESP8266_ASYNC"
; build_type = debug
; monitor_filters = esp8266_exception_decoder, default

; Uncomment the lines below by removing semicolons and edit IP for OTA upload.
; You have to upload via USB cable the first time. (upload_protocol = esptool)
; Make sure to use a data-USB cable. There is power only cables that wont work.
;upload_protocol = esptool
upload_protocol = espota
upload_port = layzspa.local
upload_flags = 
  --auth=esp8266

; edit com port if not found automatically (if you have CH340 UART)
; upload_port = COM4
```

SigmaPic, 2022-07-24:
I'm not so happy to say it, but it works with Win 10.

I don't know if it's Windows 10 or Python or another dep but it seems to work.

SigmaPic, 2022-07-24:
Maybe the issue comes from  python. I will try to update my python on Win 7.

visualapproach, 2022-07-24:
Good work on finding that out! How annoying that it produces different binaries each time.

SigmaPic, 2022-07-24:
This afternoon, I will try to analyze the build logs to find why it happens

SigmaPic, 2022-07-24:
I progressed

When pressing the "Build" button in Visual Studio Code the following line is executed:
`xtensa-lx106-elf-g++ -o .pio\build\nodemcuv2\lib241\BWC\BWC_common.cpp.o -c -fno-rtti -std=gnu++17 -fno-exceptions -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -D_GNU_SOURCE -ffunction-sections -fdata-sections -Wall -Werror=return-type -free -fipa-pta -DPLATFORMIO=60103 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU_ESP12E -DWEBSOCKETS_NETWORK_TYPE=NETWORK_ESP8266_ASYNC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_NODEMCUV2\" -DFLASHMODE_DIO -DLWIP_OPEN_SRC -DNONOSDK22x_190703=1 -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DVTABLES_IN_FLASH -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -Ilib\BWC -IC:\Users\SigmaPic\.platformio\packages\framework-arduinoespressif8266\libraries\Ticker\src -IC:\Users\SigmaPic\.platformio\packages\framework-arduinoespressif8266\libraries\LittleFS\src -I.pio\libdeps\nodemcuv2\ArduinoJson\src -I.pio\libdeps\nodemcuv2\ESPDateTime\include -I.pio\libdeps\nodemcuv2\ESPDateTime\src -IC:\Users\SigmaPic\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include -IC:\Users\SigmaPic\.platformio\packages\framework-arduinoespressif8266\cores\esp8266 -IC:\Users\SigmaPic\.platformio\packages\toolchain-xtensa\include -IC:\Users\SigmaPic\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip2\include -IC:\Users\SigmaPic\.platformio\packages\framework-arduinoespressif8266\variants\nodemcu lib\BWC\BWC_common.cpp`

When I execute this line from another cmd (not in visual) I get a different output.

SigmaPic, 2022-07-24:
Same behavior if I execute the build from Visual or from cmd with following command:
`C:\Users\SigmaPic\.platformio\penv\Scripts\platformio.exe run --environment nodemcuv2 `

SigmaPic, 2022-07-24:
I poste on PlatformIO : [https://community.platformio.org/t/build-is-not-reproductible-different-binary-between-build-execution/28924](https://community.platformio.org/t/build-is-not-reproductible-different-binary-between-build-execution/28924)

I think that my job msut stop here :)

If someone else face this issue, two solutions:

- Call manually the build command from an external cmd
- Use Windows 10

visualapproach, 2022-07-24:
Thanks for sharing. I think the report to platformio is the best way to go. Hope you will enjoy the device now.

SigmaPic, 2022-07-24:
> Thanks for sharing. I think the report to platformio is the best way to go. Hope you will enjoy the device now.

Still waiting some hardware parts :)

closed 2023-11-23 by visualapproach