visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #356: Miami Pump 54123 PCBv1 with 6 wire

dennixone, 2022-08-06:
Hi Guys i have pump 54123 compiled with PCB v1 and NO54123 in Model.h and 6 wire version.

Pump Turn on but Display Buttons dont work.
ESP is connected to WiFi but i cant Turn on over the Webinterface maybe my pump Miami 54123 is not supported?

Can anyone Help? I am from Germany 

[image]
[image]

visualapproach, 2022-08-06:
Hi there! Can't help you with your nationality, but NO54123 is not an option in the 6-wire version. You need to choose PRE2021 :)

dennixone, 2022-08-07:
Thanks you i will try later ✌️👍

dennixone, 2022-08-07:
Amazing it works very nice i modified index.html to disable airjet :)

[image]

closed 2023-03-13 by visualapproach