visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #367: Cancun 2021 Square Pump and PCB_V2 Not Working

Jinsin456, 2022-08-24:
Hi all, tried this mod on my parents hot tub last night and it didn't seem to work.

They have the Cancun Airjet model with the square 6 wire pump. I tried using PCB_V2 but went belt and braces and installed all resistors and caps. Connected the SMD connecters per the standard setup to the side of the PCB after the filter. Also using the TXS0108E LLC and ESP8266 V3.

I changed the software to MIAMI2021 and uncommented both mentions of #define PCB_V2 in the model.h file. Uploaded all files too.

I can connect to the AP and when I put in Wifi credentials, everything works as it should and I can access the web interface using the IP address, it shows Wifi connected and the current date and time on the web interface.

When I install the module and turn it on, the tone plays and I can sometimes hear a relay / solenoid click coming from the bottom where the pump and heater are but I get nothing else. The display does not turn on and none of the physical buttons on the display work.

The web interface shows [ ] in red where I assume the temperature should be and I can't control anything, when I try to adjust anything, it moves then after about 2 seconds reverts back to where it was. It doesn't seem the be controlling anything as nothing happened when I press a button on the web interface.

I can reset the ESP from the web interface and it plays the tone then a short beep about 1 second after the tone but nothing changes, everything is still unresponsive.

The ESP boots up and I can access it while it's connected to the hot tub, the blue LED seems to be constantly blinking quickly too, don't know if that's meant to happen.

I figured I would make another board without the resistors and caps and go from there and then possibly try the crossover I've seen in the instructions file. If there are any other suggestions I would be most grateful. Thank you all :)

visualapproach, 2022-08-24:
Hi, Thanks for the Detailed explanation. You must only have one pcb defined. You said 'both'. Choose one and leave the rest commented out.

Jinsin456, 2022-08-24:
So this is the only changes I made to the model.h file, what I meant was that I uncommented both mentions of the V2 board. This is the bit I struggle with lol

Thank you :)

[image]

visualapproach, 2022-08-24:
Yeah, now you have two pcb's defined. Best chance of success is if you comment out 
'//PCB_V2'
And leave
'PCB_V2B' as is.

Then resolder the wires according to the comments. I'd suggest not using the LPfilter unless it's needed.

If you don't want to resolder anything yet, go for PCB_V2 and comment out PCB_V2B by putting // in front.
The PCB definitions tells the program what wire is on which pin so you need to match that.

Jinsin456, 2022-08-27:
Made a tiny bit of progress on this today but still not there yet.

So I uncommented PCB_V2 and left PCB_V2B

I made a new board and only connected the ESP, LLC and wires, still exactly the same, plays the tone, after a out 10 seconds it makes a small chirp. Still no display and web interface loads but shows nothing

I connected the wires according to the alternative connection in the wiring diagram and using MIAMI2021 and PCB_V2B, the hot tub does exactly the same, tone and chirp but no display but this time on the web interface is shows the current temperature so getting somewhere

I've attached pics of how it's connected but I'm really stuck with this one now

Also tried PRE2021 with updated wiring and it did nothing

Please excuse dodgy soldering, will make it neater when it works lol
[image]
[image]
[image]

visualapproach, 2022-09-13:
Did you try with a red LLC?

eried, 2022-10-04:
> I made a new board and only connected the ESP, LLC and wires, still exactly the same, plays the tone, after a out 10 seconds it makes a small chirp.

just as a comment, I tried for days to add resistors, change the wiring, etc and for my pump the only thing that worked was the more "primitive" version of the level shifters https://github.com/visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA/issues/251#issuecomment-1112516802

dont know if is the case, but the tone and chirp is a dejavu :)