visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #371: [Feature Request] Add command to update cost

danteali, 2022-09-05:
Hi, thanks for this awesome project, much appreciated.

Unfortunately my coding skills are lacking and was wondering how difficult it would be to add a command to allow the cost to be updated via MQTT.

I have a sensor in Home Assistant which regularly polls my current electricity cost from my provider. I would love to be able to then push any changes to the Spa via a Home Assistant automation (mqtt.publish).

wyliecoyoteuk, 2022-09-05:
This would definitely be a useful feature, as I for example have one rate during the day and another for 00:30AM to 4:30AM

visualapproach, 2022-09-05:
Hi,

Thanks for your suggestion. IMO It would be more efficient to do that in HA rather than to push prices to all devices and return to HA. The device publishes both energy consumption and power which can be integrated in HA and multiplied by the price in each moment in time.
Maybe someone here already made such a sensor they want to share?

eried, 2022-10-04:
I think it can be already done in HA, in the energy dashboard no? (I have tiber as electricity provider but since kWh has been so cheap lately I havent tested this)

danteali, 2022-10-04:
Lol, don't know why I didn't just think of ignoring the cost from the Spa and using the kWh + cost from supplier.

Would still like to be able to push the kWh price for consistency buy totally not essential. 

Thanks

closed 2022-10-04 by danteali

danteali, 2022-10-04:
Damn that's cheap. I'm on Octopus too but not the Go tariff. Need to have a look and see what my options are for moving.

wyliecoyoteuk, 2022-10-04:
Octopus Go is for people with an EV, 35.6p/KWh in the day (think it is now 42p) 7.5p/KWh in the early hours.
We were lucky enough to fix it in for 12 months in August before the latest rises.

chunkysteveo, 2022-10-04:
> And house is in the wrong orientation for solar panels which is a massive frustration. Thinking of getting them anyway even at the lower efficiency.

You'd be surprised how well roof solar works in the UK when it's setup East/West in parallel. One side (East) works for half the day, the other essentially "takes over" on the West side as the Sun shifts over.

... or so I have heard and read - I have no solar yet, but my house is not South facing, but has a perfectly lined up East roof slope and West. One day.....!

(I'm assuming all this on you not having a huge South facing roof?!!)

wyliecoyoteuk, 2022-10-04:
Our last house faced south, but it was in a conservation area, so no solar on the roof.
Our current property faces almost due South, so should be good. I wish I had gone for it last year before the prices started going up.

chunkysteveo, 2022-10-04:
Yeah, I'm predicting that solar will be more and more popular and the price of fitting will skyrocket... surpassing the reduction in panel costs.

wyliecoyoteuk, 2022-10-04:
I had problems even getting a quote from local firms.
Some  have suspended  enquiries, some are only taking jobs in a 15 mile radius, some are taking up to 8 weeks to quote.
I had one quote that said they couldn't install until June or July 2023.
After 3 weeks, I finally got a reasonable quote, and the install is scheduled for November.