visualapproach/WiFi-remote-for-Bestway-Lay-Z-SPA #382: LEDshow and brightness

eried, 2022-09-30:
hi! thanks for fixing the issues with mqtt, I noticed in the last version also brightness = 0 is total off, which is awesome, I was wondering how could I switch between 0 and X set value after there is some activity in the buttons? what would be the easiest hack in the code. I like to have it dark but also it is cool to not use the phone to operate it

Also, what is the LEDshow? I see it is just a sequence for the 7-segment display, but why it was commented out?

visualapproach, 2022-10-04:
> hi! thanks for fixing the issues with mqtt, I noticed in the last version also brightness = 0 is total off, which is awesome, I was wondering how could I switch between 0 and X set value after there [...]

Yes that would be nice. I have to look into it when I have time.

> Also, what is the LEDshow? I see it is just a sequence for the 7-segment display, but why it was commented out? 

I had it there to check all LEDs and for fun but now with the textout it's just taking time at startup. Left it there as a possible debug tool.

eried, 2022-10-04:
Cool! I could try to code the auto-brightness, if you give me some hints on where it would be ok i.e. "change x value in line 5 to apply it" (I am just lost on how the whole code works).

BTW the virtual temp is looking nice! it is still a bit weird but looks promising :D

visualapproach, 2022-10-04:
> Cool! I could try to code the auto-brightness, if you give me some hints on where it would be ok i.e. "change x value in line 5 to apply it" (I am just lost on how the whole code works).

Thx, I was thinking in terms of adding a "set brightness 8 immediately" command followed by a "set brightness x in 5 secs" where x is previous value. Can't remember exactly where in the code it is now but in a function "handleButtonQ()" there is a section that deals with physical button presses. If someone changes brightness in between these commands it will be overwritten by x unless you make some extra code for that.
I'm also thinking about a separate command queue for these types of things but it requires more work and time... so first solution would be better to get it done fast.

> 
> 
> BTW the virtual temp is looking nice! it is still a bit weird but looks promising :D

Weird how? It needs a "learning" period with changing temperatures with pump on and without heater on. Water temp should be more than 1 degree from ambient.

Cheers

eried, 2022-10-04:
it just drops to -98 or -97 from time to time

[image]

eried, 2022-10-04:
I would actually leave that virtual value instead of the current one in the lcd :D but ofc that will change with the lid on or off...

visualapproach, 2022-10-04:
-98 is a bit cold LOL

rosetown-kelly, 2022-10-04:
Any chance the spikes to -98 were times where the unit was temporarily toggled from Celsius to Fahrenheit? Either just playing with the buttons or testing it out. 

-98/-97 is what my virtual temperature shows, whenever I have it set to F, as mentioned in the issue I just opened. #384

eried, 2022-10-04:
nah, not toggled by me at least. Also happens from 4:32 AM to 5:35 AM... but I guess is something with C and F.

visualapproach, 2022-10-05:
@eried did the ESP reboot when that -98 happened? (Check last reboot time and compare with your graphs)

visualapproach, 2022-10-05:
@eried I made the "light the display"-feature you requested. You are welcome to try it.

eried, 2022-10-05:
Ohh yes it is my daily reboot in the command queue 😂 thanks. I will test the other function soon

eried, 2022-10-05:
> @eried I made the "light the display"-feature you requested. You are welcome to try it.

It works perfectly. I just changed it to 10 secs and raise the brightness by 1 only:

`  if(_dspBrightness <8 && override_dsp_brt_timer > 0) brt = _dspBrightness+1;`

I was thinking maybe on brightness 0 it would be nice to just blink 1 segment on the LCD every few secs (or a decimal?), just as an indication of life, but right now this is fantastic for checking aurora borealis while in total darkness :D

thanks

closed 2022-10-05 by eried